arendst/Tasmota #2111: Domoticz Multiple IDX for PWM Dimmers

jimpetre, 2018-03-08:
Query more than an issue;

I'm using a Magichome/Arilux controller. Is there any way to assign individual IDX's to the relevant GPIO's as per board pinouts for PWM  to allow separate control in domoticz?

Using a 'generic' module type IDX can be assigned to pin if selecting 'relay' as output type but there is no dimming control of this. at present i have the device configured as a Sonoff LED and this allows an IDX to be assigned but control is only possible for one GPIO  

to be clear i don't want to control RGBW with the magic home but to drive separate Single color led strips as individual devices.  

Thanks

gcannar, 2018-03-09:
Same problem here! Flashed MagicHome with Tasmote, all ok!
Created a virtual RGB in domoticz but i can control only the on off function , no colors no dimmer...how to set the IDX? Domoticz give me "32" for the RBG switch...and i set in Tasmota IDX to 32...
But how to set other mqtt command sush color and dim? 
Domoticz have just one IDX for all the thing...

Some one can help me? :)

Thanks.

arendst, 2018-03-09:
Does Domoticz support RGB with MQTT interface now? It didn't a few month ago...

jimpetre, 2018-03-09:
I’m. It sure, I just want to be able to control all each gpio with a different idx... ie set a different dummy switch per pin assigned to an idx defined in the web interface / user config.h

jimpetre, 2018-03-10:
Sorry that reply wasn’t very clear. I have no desire to control RGB, (searching through previous issues this doesn’t seem possible with Domoticz) but... I’d like to control dimming for each mosfet channel of the magic Home as an individual device.

 If you select ‘Relay’ as ‘type’ in the web config it gives you an idx to configure with Domoticz but selecting PWM it doesn’t. Is there any way to do this?

jimpetre, 2018-03-20:
> Dimming for each channel will be possible from release 5.12.0h using
> command channel 0..100

You da man!

ascillato2, 2018-08-23:
Closing this issue as it is fixed.

closed 2018-08-23 by ascillato2

mVik85, 2020-12-10:
> Still no way to assign different IDX's to PWM1,2,3,4,5? as you would with relay 1,2,3,4,5?

If anyone would wonder about this (like me until 5 minutes ago), here the answer:
https://github.com/emontnemery/domoticz_mqtt_discovery