arendst/Tasmota #2114: Sonoff Dual R2 keeps crashing after succesful flash

hattemp, 2018-03-08:
My Sonoff Dual R2 keeps crashing (blinking Led 32 times) after a succesful upload.
i am using arduino IDE 1.8.5 with the upload settings provided on the wiki.
i am using a ftdi adapter on 3.3v
i am using the latest tasmota version.
i have used these setting succesfull with 2 basic and 1 th module. they are working fine.

tryed playing with the setting butt no luck.
tryed eps8285 butt no luck, it wil not startup then.
 below is a screen from the serial monitor.
it keeps repeating this:

[image]

these are the esp8266 upload setting i used:

[image]

these are the esp8285 setting i used:

[image]

can anyone help me or kan i throw it in the bin?
i can give u more screens if it is helpfull.

Jason2866, 2018-03-10:
Dont use IwIP 1.4. It is instable. Use older version.

hattemp, 2018-03-10:
Thanks wil try that.