arendst/Tasmota #2116: Request: Send / Receive RAW IR data for any device

cyberelectronics, 2018-03-09:
Hi, 
Using IRrecvDumpV2 example sketch from IRremoteESP8266 library, recommended by David Conran (Thanks!), I was able to find the RAW codes for my pellet stove IR remote control and with IRMQTTServer example sketch I played back succesfuly the RAW data to the pellet stove.

The question is it possible to implement this RAW data Receive/Transmit function over MQTT in Tasmota (no need for decode, just receiving/sending RAW data info over MQTT)?

**Here is the RAW data:**
`Extraflame Ecologica Idro Pellet Stove - IR commands

**Button 1+2 = Power ON / OFF**
Encoding  : UNKNOWN
Code      : 9EB58962 (11 bits)
Library   : v2.3.3

Raw Timing[21]:
   +  6744, -  2462,    +  3396, -  1624,    +   884, -   788,    +   912, -  1610, 
   +  1722, -   814,    +   858, -  1628,    +   884, -   788,    +   884, -  1626, 
   +  1716, -   794,    +  1722, -  1626,    +  1726

uint16_t rawData[21] = {6744, 2462,  3396, 1624,  884, 788,  912, 1610,  1722, 814,  858, 1628,  884, 788,  884, 1626,  1716, 794,  1722, 1626,  1726};  // UNKNOWN 9EB58962

**Button1 = Set Temperature Higher**
Encoding  : UNKNOWN
Code      : 4B3AB9D1 (12 bits)
Library   : v2.3.3

Raw Timing[23]:
   +  6742, -  2464,    +  3420, -  1626,    +   858, -   790,    +   904, -  1642, 
   +  1694, -   814,    +  1696, -   842,    +   832, -   814,    +   860, -  1652, 
   +  1694, -   814,    +   858, -   814,    +   858, -   814,    +  1696

uint16_t rawData[23] = {6742, 2464,  3420, 1626,  858, 790,  904, 1642,  1694, 814,  1696, 842,  832, 814,  860, 1652,  1694, 814,  858, 814,  858, 814,  1696};  // UNKNOWN 4B3AB9D1

**Button2 = Set Power Higher**
Encoding  : UNKNOWN
Code      : 2B890461 (10 bits)
Library   : v2.3.3

Raw Timing[19]:
   +  6796, -  2462,    +  3394, -  1626,    +   910, -   762,    +   886, -  1634, 
   +  4232, -   788,    +   884, -   790,    +   886, -  1624,    +  3396, -   788, 
   +   884, -   790,    +  1722

uint16_t rawData[19] = {6796, 2462,  3394, 1626,  910, 762,  886, 1634,  4232, 788,  884, 790,  886, 1624,  3396, 788,  884, 790,  1722};  // UNKNOWN 2B890461 

**Button3 = Set Temperature Lower**
Encoding  : UNKNOWN
Code      : 3BEE891B (9 bits)
Library   : v2.3.3

Raw Timing[17]:
   +  6218, -  2436,    +  3396, -  1624,    +   912, -   760,    +   912, -  1610, 
   +  3398, -  1624,    +   912, -   762,    +   912, -  1598,    +  4236, -  1624, 
   +  1746

uint16_t rawData[17] = {6218, 2436,  3396, 1624,  912, 760,  912, 1610,  3398, 1624,  912, 762,  912, 1598,  4236, 1624,  1746};  // UNKNOWN 3BEE891B

**Button4 = Set Power Lower** 
Encoding  : UNKNOWN
Code      : BD1A8437 (11 bits)
Library   : v2.3.3

Raw Timing[21]:
   +  6800, -  2462,    +  3396, -  1626,    +   884, -   788,    +   888, -  1634, 
   +  2558, -   790,    +   884, -   786,    +   886, -   788,    +   884, -  1626, 
   +  2560, -  1624,    +   884, -   788,    +  1722

uint16_t rawData[21] = {6800, 2462,  3396, 1626,  884, 788,  888, 1634,  2558, 790,  884, 786,  886, 788,  884, 1626,  2560, 1624,  884, 788,  1722};  // UNKNOWN BD1A8437
`
Thank you!

arendst, 2018-03-10:
It would have to be a special compile option as it uses an extra buffer during ir receive.

Considering....

OTrumpf, 2018-03-17:
+1 .... for an old iRobot Roomba cleaner which could be started via Alexa + OpenHAB + Tasmota-IR-Device.

cyberelectronics, 2018-03-19:
We found a temporary solution :) .  
My friend implemented the IRMQTTServer sketch protocols in the webapp, so now we can create/execute IR RAW codes. 
The only problem is we need to use a separate device with IRrecvDumpV2 sketch on it (serial), to be able to read/learn the IR RAW codes. (example at 7:40)
Both, reading and sending RAW codes using Tasmota, would be the best solution! 
https://youtu.be/Q5JDlyIvATc

jnt2007, 2018-07-24:
+1 it would be useful for my Samsung AC. When I get IR from remote control I can see only encoded info
`00:00:14 IRR: RawLen 74, Bits 37, Value 3FFF4B54, Decode -292978525
00:00:15 IRR: RawLen 74, Bits 37, Value 3FFF4B54, Decode 961051959`

So I haven't any idea how to send this command via Tasmota IRsend...

jsmath, 2018-09-22:
I would simply like the ability to send raw data using the IRSend command.  It is possible to get the raw receive data loading the "IRrecvDumpV2.ino" Arduino project (part of same IRremoteESP8266 lib that Tasmota uses).  See my example serial output below.  I can then copy the "rawData[87]" array data that is output into the "IRsendDemo.ino" Arduino project (I also changed the "irsend.sendRaw(rawData, 67, 38);" line in the example so 67 is replaced with my array size.. 87 in my case).   Then, running the IRSendDemo, I am able to send my IR command correctly.

The question is, how do I convert this raw data to a command that I can send with Tasmota?  The IRsend command requires a protocol to be specified, but in my case I would like to just send raw data since the protocol is unknown.  Also, I believe I need to send 44 bits, whereas the IRSend command seems to limit at 32 bits, at least according to the documentation. (I was also able to send two 22 bit commands back to back which worked since it was a repeating command sequence).

Hopefully at least sending raw IR data can be added as an officially supported feature!

**Serial output from "IRrecvDumpV2.ino":**
Timestamp : 000131.050
Encoding  : UNKNOWN
Code      : FFBBD42F (44 bits)
Not Mitsu AC.
Library   : v2.5.0

Raw Timing[87]:
   +  4440, -  4524,    +   456, -  1558,    +   454, -   550,    +   456, -   550, 
   +   454, -   552,    +   454, -   550,    +   454, -   554,    +   454, -   550, 
   +   454, -   552,    +   454, -   552,    +   454, -  1560,    +   456, -   550, 
   +   454, -   552,    +   452, -  1560,    +   454, -  1558,    +   456, -  1558, 
   +   454, -  1558,    +   456, -  1558,    +   454, -  1558,    +   410, -   596, 
   +   454, -   552,    +   454, - 22964,    +  4440, -  4524,    +   454, -  1560, 
   +   454, -   550,    +   454, -   552,    +   452, -   552,    +   452, -   552, 
   +   454, -   550,    +   454, -   550,    +   454, -   552,    +   454, -   552, 
   +   408, -  1602,    +   454, -   552,    +   454, -   552,    +   454, -  1558, 
   +   456, -  1560,    +   454, -  1558,    +   456, -  1560,    +   454, -  1558, 
   +   454, -  1558,    +   454, -   552,    +   452, -   552,    +   452

uint16_t rawData[87] = {4440, 4524,  456, 1558,  454, 550,  456, 550,  454, 552,  454, 550,  454, 554,  454, 550,  454, 552,  454, 552,  454, 1560,  456, 550,  454, 552,  452, 1560,  454, 1558,  456, 1558,  454, 1558,  456, 1558,  454, 1558,  410, 596,  454, 552,  454, 22964,  4440, 4524,  454, 1560,  454, 550,  454, 552,  452, 552,  452, 552,  454, 550,  454, 550,  454, 552,  454, 552,  408, 1602,  454, 552,  454, 552,  454, 1558,  456, 1560,  454, 1558,  456, 1560,  454, 1558,  454, 1558,  454, 552,  452, 552,  452};  // UNKNOWN FFBBD42F

jsmath, 2018-09-22:
For anyone interested in hacking the code to send their own raw IR commands, it is pretty easy (once you have the rawData array (see example array in my previous post).  I just added the following "case 0:" statement to the IRSendCommand function (in xdrv_05_irremote.ino) after the line "switch (protocol_code) {":
            case 0: //UNKNOWN
                  irsend->sendRaw(rawData, 87, 38); break; 

You also need to define the rawData array elsewhere in the code and set "87" to the correct size.   Here is an example IRSend call (the values for "bits" and "data" aren't used, but these values could be used in the code to select a different rawData array and send different IR commands):

IRSend {"protocol": "UNKNOWN", "bits": 44, "data": "0"}

jnt2007, 2018-09-23:
@jsmath this it pretty simple! Open a pull request for adding this to the official build, I think that is really useful feature!

In case of using RAW data for AC remote I have complex solution on ESP-01. I complete IRMQTTServer with IRrecvDumpV2. Easy and simple :)

OTrumpf, 2018-10-10:
Perhaps sending the RAW IR data as JSON is a great possibility to keep the code free of individual RAW commands!
See: [Github: ESP8266-HTTP-IR-Blaster](https://github.com/mdhiggins/ESP8266-HTTP-IR-Blaster)  --> JSON Scripting

nosaj66au, 2018-10-31:
This could also be used for unknown RF Transmissions not just IR. Would be great in a send as a JSON packet.

Jason

dr-apple, 2018-11-08:
uint16_t hz=38;

 case 0: //UNKNOWN
           irsend->sendRaw(data, sizeof(data), hz);break;

will not work 👎 

xdrv_05_irremote:324:53: error: invalid conversion from 'uint32_t {aka unsigned int}' to 'uint16_t* {aka short unsigned int*}' [-fpermissive]
               irsend->sendRaw(data, sizeof(data), hz);

Gompka, 2018-11-13:
This feature would also help me greatly, both my dyson fan and electric fireplace require raw data to work.

Jason2866, 2018-11-16:
This feature would be nice, i have the need for too.

arendst, 2018-11-21:
How to use IR Raw data receive info: 
Control output of IR raw data to JSON message with command ``SetOption58 1``.
The received JSON message could look like this:
```
{"IrReceived":{"Protocol":"RC5","Bits":12,"Data":"803","RawData":[932,838,932,856,1772,892,908,842,906,864,906,846,904,922,910,840,908,900,902,850,910,1774,928,842,812],"RawDataInfo":[25,25,0]}}
```
where the raw data is ready to be send but WITHOUT the IR frequency as noted below.
RawDataInfo consists of three values. The first value provides the number of packets in the libraries receive buffer limited by define IR_RCV_BUFFER_SIZE in file my_user_config.h. The second number contains the number of packets in the JSON message limited by the allowable number of characters in the JSON message. The third number is either a 0 if the libraries buffer is not overflowed or 1 if the buffer is overflowed. So a correct JSON message would return two equal numbers followed by a 0 as can be seen in the message above.

How to send IR Raw data:
The use a JSON parameter option would take too much resources while sending raw data so I implemented basic IRSend raw data parameters only.
To send IR Raw data the program needs to know the IR frequency and the raw data to send. So the parameters to send raw data ALWAYS needs to start with either a 0, meaning 38kHz IR signal or the wanted IR frequency. 
A complete IRSend Raw data command line would look like this:
```
irsend 0,932,838,932,856,1772,892,908,842,906,864,906,846,904,922,910,840,908,900,902,850,910,1774,928,842,812
```
where the program will replace the first 0 by 38000Hz IR frequency.

Gompka, 2018-11-21:
Thank you very much for implementing this, is there a limit to the length of the raw data? I have a raw data string that is 140 numbers long.

arendst, 2018-11-21:
Yes there is a default limit of 100. This can be changed by define IR_RCV_BUFFER_SIZE in my_user_config.h and when you recompile you are using more memory with a larger buffer.

dr-apple, 2018-11-22:
Hi unfortunately this does not work for Gree air conditioners :-(

When I read the remote control Power On it comes out:

uint16_t poweron [139] = {8956, 4560, 546, 1710, 588, 616, 584, 618, 584, 1712, 586, 1712, 588, 614, 586, 614, 586, 618, 586, 1706, 590, 616 , 586, 1708, 590, 616, 588, 610, 590, 612, 590, 612, 590, 618, 584, 618, 584, 616, 586, 614, 588, 612, 590, 612, 590, 1712, 586 , 616, 588, 612, 590, 614, 588, 614, 586, 618, 586, 616, 586, 1710, 588, 614, 586, 1708, 590, 610, 592, 612, 590, 1708, 588, 614 , 588, 20084, 548, 612, 590, 614, 588, 614, 588, 616, 586, 654, 546, 616, 586, 616, 586, 614, 588, 612, 588, 610, 592, 612, 590 , 612, 590, 614, 586, 1712, 586, 616, 586, 612, 588, 614, 588, 614, 586, 616, 586, 612, 590, 614, 588, 610, 590, 616, 586, 614 , 588, 612, 590, 612, 590, 614, 588, 614, 588, 614, 588, 1708, 590, 612, 590, 1708, 590}; // UNKNOWN FFFE852

with tasmota:

{ "IrReceived": { "Protocol": "UNKNOWN", "bits": 37, "Data", "F179E0E5", "RawData": [8988,4546,584,1676,622,1710,588,614,586,1712,586 , 1674,624,616,588,580,622,616,586,616,586,1710,588,1712,584,1710,588,580,622,618,584,614,586,618,584,580,622,616,586,616,586,612,590,614,588,1708,588,614,590,616,586,612,590,580,622,612,590,612,588,1714,584,580,620,1710,590,616,586,618,586,1712,586,612,590], "RawDataInfo" [73,73,0]}}

But no matter if I send the first code or the IR code of tasmota, in both cases the device rules zero :-(

Why is that?

fostiras, 2018-11-25:
Hello,
sending examples  
 IRsend {"Protocol":"NEC","Bits":32,"Data":2170978686}   and  
 IRsend {"Protocol":"NEC","Bits":32,"Data":0x8166817E}  both through the console and as MQTT topic work OK returning 
stat/sonoff/IRsend/RESULT = {"IRSend":"Done"} and I get IR emission.

When trying to send RAW data with  
 irsend 0,932,838,932,856,1772,892,908,842,906,864,906,846,904,922,910,840,908,900,902,850,910,1774,928,842,812     or  
 IRsend 0,926,844,958,832,1798,868,902,848,900,870,900,852,908,918,958,794,934,874,928,1738,934,856,1764  examples, returns
stat/sonoff/IRsend/RESULT = {"IRSend":"Invalid JSON"}  . 
Is there something else needed for RAW data sending ?
Thank you.

arendst, 2018-11-26:
It works:
```
15:16:02 CMD: irsend 0,932,838,932,856,1772,892,908,842,906,864,906,846,904,922,910,840,908,900,902,850,910,1774,928,842,812
15:16:02 SRC: WebConsole from 192.168.2.1
15:16:02 RSL: Group 0, Index 1, Command IRSEND, Data 0,932,838,932,856,1772,892,908,842,906,864,906,846,904,922,910,840,908,900,902,850,910,1774,928,842,812
15:16:02 MQT: stat/ledunia2/RESULT = {"IRSend":"Done"}
```
as long as you use the latest versions. You did not provide info for further analysis.

fostiras, 2018-11-26:
Thank you for your great work!!
I am using a NodeMcu v2 with pin D4 (GPIO16) connected to an IR emitter configured as IRsend. NodeMcu is loaded with your latest 6.3.0 precompiled firmware.

Jason2866, 2018-11-26:
@fostiras 
You have to use latest development version

fostiras, 2018-11-26:
I 'll do that and test in a few hours! Thank you.

Jason2866, 2018-11-28:
Dev is stable. Last commits are most fixes and improvements
And yes will be in next release.
There is no reason for not using the dev version 6.3.0.14

sheppy99, 2018-12-01:
I'm trying to get a long Daikin String to send over MQTT using the latest Tasmota Development version, and I think I'm running out of buffer space. A capture with a WemosD1 board flashed with IRrecvDumpV2 gives the captures in the attached files which I'd like to resend with a SOnOff Dual V2, Tasmota and a transistor assisted IR LED. Is this possible? Publishing the long string at the bottom of the files as the payload to cmnd/mydevice/irsend or via a web command prompt doesn't seem to work which I guess is down to the length of the command.
What do I need to do to get this working? I'm happy to use I stand alone ESP8266 or an ESP32 if it makes more sense.
[ESPcaptureRAWfanonly.txt](https://github.com/arendst/Sonoff-Tasmota/files/2636305/ESPcaptureRAWfanonly.txt)
[ESPcaptureRAWoff.txt](https://github.com/arendst/Sonoff-Tasmota/files/2636306/ESPcaptureRAWoff.txt)

Gompka, 2018-12-01:
> I'm trying to get a long Daikin String to send over MQTT using the latest Tasmota Development version (6.3.0.15), and I think I'm running out of buffer space. A capture with a WemosD1 board flashed with IRrecvDumpV2 gives the captures in the attached files which I'd like to resend with a SOnOff Dual V2, Tasmota and a transistor assisted IR LED. Is this possible? Publishing the long string at the bottom of the files as the payload to cmnd/mydevice/irsend or via a web command prompt doesn't seem to work which I guess is down to the length of the command.
> What do I need to do to get this working? I'm happy to use I stand alone ESP8266 or an ESP32 if it makes more sense.
> [ESPcaptureRAWfanonly.txt](https://github.com/arendst/Sonoff-Tasmota/files/2636305/ESPcaptureRAWfanonly.txt)
> [ESPcaptureRAWoff.txt](https://github.com/arendst/Sonoff-Tasmota/files/2636306/ESPcaptureRAWoff.txt)

I had to use a really long mqtt buffer in the pub-sub client, i used 2048 to get my long string of 150 values to get sent over mqtt.

sheppy99, 2018-12-01:
Thanks for the reply, where is this set? Also would this mean it won't send from a command window either?

Gompka, 2018-12-01:
> Thanks for the reply, where is this set? Also would this mean it won't send from a command window either?
> Maybe I have 2 problems to solve

Mine was working fine from the console window on the built in webserver, but wouldn't work with mqtt until I made the change. You have to go to the the arduino libraries folder then find PubSubClient, then src folder, and open PubSubClient.h and change #define MQTT_MAX_PACKET_SIZE

sheppy99, 2018-12-01:
This string won't even send from a command prompt so I think I need to change whatever buffer that uses too. Thanks for the location of #define MQTT_MAX_PACKET_SIZE

kimnnguu204, 2018-12-02:
I'm using version 6.3.0..14 to decode my AC Daikin remote :
this is the log i saw on console:

14:49:59 MQT: tele/sonoff/RESULT = {"IrReceived":{"Protocol":"UNKNOWN","Bits":6,"Data":"B3E2CB06","RawData":[436,458,410,434,434,430,438,456,412,456,412],"RawDataInfo":[11,11,0]}}
14:49:59 IRR: RawLen 253, Overflow 0, Bits 126, Value A9E81078, Decode 0
14:50:00 IRR: RawLen 12, Overflow 0, Bits 6, Value B3E2CB06, Decode 0
14:50:01 MQT: tele/sonoff/RESULT = {"IrReceived":{"Protocol":"UNKNOWN","Bits":6,"Data":"B3E2CB06","RawData":[438,456,436,432,436,432,412,456,412,456,412],"RawDataInfo":[11,11,0]}}
14:50:01 IRR: RawLen 296, Overflow 0, Bits 148, Value 150A6533, Decode 0
14:50:03 IRR: RawLen 12, Overflow 0, Bits 6, Value B3E2CB06, Decode 0
14:50:03 MQT: tele/sonoff/RESULT = {"IrReceived":{"Protocol":"UNKNOWN","Bits":6,"Data":"B3E2CB06","RawData":[434,434,436,434,434,434,436,434,434,434,434],"RawDataInfo":[11,11,0]}}
14:50:03 IRR: RawLen 268, Overflow 0, Bits 134, Value A5A8FA9D, Decode 0
14:50:04 IRR: RawLen 12, Overflow 0, Bits 6, Value B3E2CB06, Decode 0
14:50:04 MQT: tele/sonoff/RESULT = {"IrReceived":{"Protocol":"UNKNOWN","Bits":6,"Data":"B3E2CB06","RawData":[456,414,438,430,438,430,438,366,502,430,438],"RawDataInfo":[11,11,0]}}
14:50:04 IRR: RawLen 154, Overflow 0, Bits 77, Value 375B5577, Decode 0
14:50:20 IRR: RawLen 12, Overflow 0, Bits 6, Value B3E2CB06, Decode 0
14:50:20 MQT: tele/sonoff/RESULT = {"IrReceived":{"Protocol":"UNKNOWN","Bits":6,"Data":"B3E2CB06","RawData":[462,432,412,456,412,456,460,408,412,458,412],"RawDataInfo":[11,11,0]}}
14:50:20 IRR: RawLen 158, Overflow 0, Bits 79, Value A101D201, Decode 0
14:50:22 IRR: RawLen 12, Overflow 0, Bits 6, Value B3E2CB06, Decode 0
14:50:22 MQT: tele/sonoff/RESULT = {"IrReceived":{"Protocol":"UNKNOWN","Bits":6,"Data":"B3E2CB06","RawData":[438,456,412,458,410,458,412,456,412,460,410],"RawDataInfo":[11,11,0]}}
14:50:22 IRR: RawLen 282, Overflow 0, Bits 141, Value 07140A96, Decode 0

i tried to send it on console but it doesn't work. 

this is my configure:
#define USE_IR_REMOTE                            // Send IR remote commands using library IRremoteESP8266 and ArduinoJson (+4k3 code, 0k3 mem, 48 iram)
  #define USE_IR_HVAC                            // Support for HVAC (Toshiba, Mitsubishi and LG) system using IR (+3k5 code)
  #define USE_IR_RECEIVE                         // Support for IR receiver (+7k2 code, 264 iram)
    #define IR_RCV_BUFFER_SIZE      700          // Max number of packets allowed in capture buffer (default 100 (*2 bytes ram))
    #define IR_RCV_TIMEOUT          15           // Number of milli-Seconds of no-more-data before we consider a message ended (default 15)
    #define IR_RCV_MIN_UNKNOWN_SIZE 6 

how to make it work :)

ceaswaran, 2018-12-22:
> Yes there is a default limit of 100. This can be changed by define IR_RCV_BUFFER_SIZE in my_user_config.h and when you recompile you are using more memory with a larger buffer.

can we have the IR_RCV_BUFFER_SIZE increased ,bin file

Jason2866, 2018-12-22:
No, because it needs to much space from the rare flash space.
For this needs you need to self compile.

ilya89vet, 2018-12-23:
Is there any way to turn it off ? I have the entire log covered with these messages!
Also inexplicably turns on the fourth relay which I have hanging on GPIO04
[image]
`

ceaswaran, 2018-12-24:
Even after changing IR_RCV_BUFFER_SIZE to 1000 (from 100) & compiling Tasmota. I am not able to get the RAW Data fully.

`19:07:25 MQT: tele/sonoffir/RESULT = {"IrReceived":{"Protocol":"UNKNOWN","Bits":114,"Data":2896510163,"RawData":[3788,1448,622,1136,616,1138,592,442,590,442,614,418,590,1160,618,418,612,418,622,1134,592,1054,726,416,614,1090,664,452,580,414,616,1138,616,1136,620,416,616,1136,618,1136,618,416,616,454,580,1134,618,414,618,414,620,1136,616,454,580,414,616,416,618,414,618,414,618,414,616,414,620,452,580,416,614,416,616,418,614,416,616,452,580,452,578,418,620,416,616,414,616,416,616,416,616,418,616,1134,618,416,616,418,618, ...`

Dot dot Dot at the end. 

Is there a work aroud to get the remaning Raw data fully???

ilya89vet, 2018-12-24:
> Есть ли способ отключить его ? У меня весь журнал покрыт этими сообщениями!
> Также необъяснимо включается четвертое реле, которое у меня висит на GPIO04
> [image]
> `

I do not know correctly or not, but I increased value of parameter IR_RCV_MIN_UNKNOWN_SIZE to 24 and these messages ceased to be published in the weblog!

tiandeyu, 2018-12-28:
> * The first value provides the number of packets in the libraries receive buffer limited by define IR_RCV_BUFFER_SIZE in file my_user_config.h.
> * The second number contains the number of packets in [...]

Here is my data received from my Panasonic heater/ventilator remote, first number of RawDataInfo is 391 while the second number is 187
tele/sonoff/RESULT {"IrReceived":{"Protocol":"UNKNOWN","Bits":196,"Data":"18A28917","RawData":[3484,1774,414,466,416,1336,414,466,416,466,416,466,416,466,414,466,416,466,416,466,414,466,414,466,466,414,414,466,416,1338,416,466,414,466,414,466,416,464,414,466,414,466,414,1336,416,464,414,1338,416,1336,416,466,414,466,414,1338,414,468,414,466,414,1338,414,464,416,466,414,466,416,1338,414,466,414,466,414,1338,416,1336,414,466,414,1338,416,1338,416,464,414,466,414,444,436,1338,466,414,416,466,416,466,416,1338,416,466,414,1336,418,464,414,466,416,1338,416,466,414,464,416,444,436,466,416,464,416,464,416,466,416,464,416,1336,416,464,416,466,414,1338,416,464,414,466,414,1338,416,442,438,1338,416,466,416,466,414,1338,416,466,414,464,468,388,438,1338,416,1338,416,466,416,464,414,466,414,464,416,466,414,464,416,466,414,466,414,1338,416,464,416,1336,416,464,414,466,416,466],"RawDataInfo":[391,187,0]}}

can I increase the limit of JSON message?

Jason2866, 2018-12-28:
@tiandeyu yes, as explained here already
https://github.com/arendst/Sonoff-Tasmota/issues/2116#issuecomment-440720126

tiandeyu, 2018-12-28:
Hi Jason,
I do read this comment, so I changed IR_RCV_BUFFER_SIZE from 100 to 500 then I got the result "RawDataInfo":[391,187,0]. The instructions explained that the first value of RawDataInfo limited by IR_RCV_BUFFER_SIZE, and second value limited by the allowable number of characters in the JSON message.

Below is the received data when IR_RCV_BUFFER_SIZE was 100.
tele/sonoff/RESULT {"IrReceived":{"Protocol":"UNKNOWN","Bits":94,"Data":"BC36F8BE","RawData":[3454,1800,422,464,444,1314,422,466,444,442,444,442,446,444,422,464,444,444,422,464,498,390,422,464,422,492,420,440,444,1314,444,442,444,444,446,440,446,440,422,466,446,418,444,1338,446,440,446,1312,462,1298,444,442,422,462,422,1336,420,466,498,388,498,1262,444,442,446,444,498,392,446,1310,422,464,498,368,524,1258,446,1316,446,440,498,1264,446,1310,446,416,498,440,472,390,498,1262,436,454,498,390,446,444,494,1268,498,390,472,1288,498,390,498,392,498,1262,448,438,472,418,472,416,498,390,498,390,498,414,474,388,498,388,472,1284,498,368,524,390,472,1288,472,418,498,390,472,1290,472,418,498,1260,498,388,472,444,446,414,472,418,472,1288,498,390,472,1290,498,1260,446,444,446,442,446,440,422,442,422,486,446,442,446,442,420,466,446,1314,422,466,422,464,420,464,422,1336,446],"RawDataInfo":[187,187,0]}}

tiandeyu, 2018-12-29:
Problem solved by increasing MQTT_MAX_PACKET_SIZE in both PubSubClient.h and sonoff_post.h,
it seems MQTT_MAX_PACKET_SIZE in sonoff_post.h overwritten the PubSubClient.h one.

Jason2866, 2018-12-29:
@tiandeyu 
which values are you using now?

ceaswaran, 2018-12-29:
> Problem solved by increasing MQTT_MAX_PACKET_SIZE in both PubSubClient.h and sonoff_post.h,
> it seems MQTT_MAX_PACKET_SIZE in sonoff_post.h overrides the PubSubClient.h one.
@tiandeyu Can we have the complined BIn file of yours with MQTT_MAX_PACKET_SIZE & IR_RCV_BUFFER_SIZE to 1000
So that i can test it 
Thanks in advance

tiandeyu, 2018-12-30:
Hi, ceaswaran
There are 4 values I've changed.
my_user_config.h
IR_RCV_BUFFER_SIZE      1000
IR_RCV_TIMEOUT          50
PubSubClient.h and sonoff_post.h
MQTT_MAX_PACKET_SIZE 2000

tiandeyu, 2018-12-30:
There is another issue here if your packet larger than 240
[modify byte to uint16_t](https://github.com/arendst/Sonoff-Tasmota/blob/aa4c58a26661c102da54596d6f7e2cba0104999e/sonoff/xdrv_05_irremote.ino#L534)

sblantipodi, 2019-01-09:
> Yes there is a default limit of 100. This can be changed by define IR_RCV_BUFFER_SIZE in my_user_config.h and when you recompile you are using more memory with a larger buffer.

is it possible to use larger buffer without recompiling tasmota?
a command via web console to increase the size of the buffer?

sblantipodi, 2019-01-10:
@arendst
is it possible to use larger buffer, mqtt packet size and timeout without recompiling tasmota?

a command via web console to increase the size of the buffer/packet size/timeout?

arendst, 2019-01-10:
@sblantipodi You can change the buffers using the appropriate defines. Other than that it's going to be a nightmare as there just is not enough memory to serve all features. It's not only the receive buffer but also the mqtt buffers so it will decrease memory rapidly.

I suggest you make your changes needed and live with it.

sblantipodi, 2019-01-10:
@arendst thanks for the reply. I have changed those settings:
my_user_config.h
IR_RCV_BUFFER_SIZE 1000
IR_RCV_TIMEOUT 50
PubSubClient.h and sonoff_post.h
MQTT_MAX_PACKET_SIZE 2000

now it seems that the data received from the same button is consistent.
I always receive something like this:
{"Protocol":"UNKNOWN","Bits":175,"Data":"7561C5F6", 
with the same data (7561C5F6)
but the protocol remains UNKNOWN so I cannot IRSend it

how can I solve it? thanks.

Jason2866, 2019-01-10:
You have to use `SetOption58 1`

sblantipodi, 2019-01-10:
@Jason2866 I have tried that but don't understand if I reached the "webgui console buffer" limit
this is the output whenusing option58

19:40:32 MQT: tele/tasmotadev/RESULT = {"IrReceived":{"Protocol":"UNKNOWN","Bits":175,"Data":"D6C798FC","RawData":[588,17848,3052,8928,518,476,540,1454,562,430,544,452,542,452,544,452,564,432,542,454,542,456,540,1472,542,454,520,480,526,1466,526,468,574,422,524,1492,494,1472,544,1448,566,1426,542,1452,538,456,562,434,542,456,538,476,518,478,518,480,524,472,546,450,548,448,550,448,522,472,518,478,542,452,542,454,542,452,542,454,542,454,542,454,544,452,562,436,540,456,562,434,538,476,518,478,518,480,526,468,526,470,548,4 ...

sblantipodi, 2019-01-10:
just for referene this is the complete rawdata get from mosquitto
{"IrReceived":{"Protocol":"UNKNOWN","Bits":175,"Data":"7561C5F6","RawData":[632,17826,3054,8928,540,454,548,1444,548,448,548,448,550,446,548,446,550,446,550,446,550,446,548,1444,548,448,548,474,520,1472,508,1484,542,454,544,1448,548,1444,548,1442,548,1444,550,1442,548,448,548,448,548,446,548,448,548,448,546,474,480,514,508,488,534,460,542,454,546,450,548,448,548,446,550,446,550,446,548,446,548,448,550,446,550,446,548,448,548,448,548,448,548,448,548,472,522,474,508,488,538,458,542,452,544,452,548,446,548,446,550,446,548,446,548,446,550,1442,548,1444,508,2898,3120,8928,550,1444,548,474,480,514,510,486,536,460,542,454,544,452,550,446,550,446,550,1442,550,446,548,448,548,1444,548,448,548,1444,548,1446,546,1470,480,1510,538,1454,542,1448,548,448,548,446,548,448,548,446,548,446,548,448,548,448,550,446,548,448,548,448,548,448,548,474,520,474,506,490,510,486,540,456,544,452,546,450,548,448,550,446,548,446,550,446,550,446,550,446,548,448,548,446,548,448,548,448,548,448,548,472,520,474,482,512,534,462,538,458,542,454,546,450,550,2928,3048,8910,568,1444,548,446,548,446,550,446,548,448,550,446,548,446,548,450,546,450,546,1470,508,488,512,484,538,456,544,452,546,448,548,446,550,1442,550,1442,550,1442,548,1444,548,1444,548,1444,548,1470,482,1510,538,1454,544,452,546,448,548,446,550,1444,548,1442,548,1444,550,446,548,448,550,446,550,446,548,448,546,474,520,474,506,490,520,1472,544,1448,546,450,550,444,550,446,548,1444,550,444,550,446,550,446,548,446,548,448,548,446,548,450,546,474,522,474,506,1486,540,1452,548],"RawDataInfo":[349,349,0]}}

sblantipodi, 2019-01-10:
irsend 0,632,17826,3054,8928,540,454,548,1444,548,448,548,448,550,446,548,446,550,446,550,446,550,446,548,1444,548,448,548,474,520,1472,508,1484,542,454,544,1448,548,1444,548,1442,548,1444,550,1442,548,448,548,448,548,446,548,448,548,448,546,474,480,514,508,488,534,460,542,454,546,450,548,448,548,446,550,446,550,446,548,446,548,448,550,446,550,446,548,448,548,448,548,448,548,448,548,472,522,474,508,488,538,458,542,452,544,452,548,446,548,446,550,446,548,446,548,446,550,1442,548,1444,508,2898,3120,8928,550,1444,548,474,480,514,510,486,536,460,542,454,544,452,550,446,550,446,550,1442,550,446,548,448,548,1444,548,448,548,1444,548,1446,546,1470,480,1510,538,1454,542,1448,548,448,548,446,548,448,548,446,548,446,548,448,548,448,550,446,548,448,548,448,548,448,548,474,520,474,506,490,510,486,540,456,544,452,546,450,548,448,550,446,548,446,550,446,550,446,550,446,548,448,548,446,548,448,548,448,548,448,548,472,520,474,482,512,534,462,538,458,542,454,546,450,550,2928,3048,8910,568,1444,548,446,548,446,550,446,548,448,550,446,548,446,548,450,546,450,546,1470,508,488,512,484,538,456,544,452,546,448,548,446,550,1442,550,1442,550,1442,548,1444,548,1444,548,1444,548,1470,482,1510,538,1454,544,452,546,448,548,446,550,1444,548,1442,548,1444,550,446,548,448,550,446,550,446,548,448,546,474,520,474,506,490,520,1472,544,1448,546,450,550,444,550,446,548,1444,550,444,550,446,550,446,548,446,548,448,548,446,548,450,546,474,522,474,506,1486,540,1452,548 
does not poweron my AC

Jason2866, 2019-01-10:
@sblantipodi  Please do not post here your personal findings.
GitHub issues is for general problems with Tasmota
Thank you

sblantipodi, 2019-01-10:
this is from arduino decode

`
IRrecvDumpV2 is now running and waiting for IR input on Pin 12
Timestamp : 000015.322
Encoding  : SAMSUNG_AC
Code      : 02920F000000F001D20F0000000001B2FE71B015F0 (168 bits)
Mesg Desc.: Power: On, Mode: 0 (AUTO), Temp: 16C, Fan: 0 (AUTO), Swing: Off, Beep: Off, Clean: Off, Quiet: Off
Library   : v2.5.4

Raw Timing[349]:
   +   624, - 17854,    +  2998, -  8962,    +   516, -   480,    +   516, -  1476, 
   +   516, -   502,    +   494, -   502,    +   548, -   448,    +   544, -   450, 
   +   546, -   450,    +   544, -   452,    +   522, -   474,    +   518, -  1474, 
   +   520, -   476,    +   520, -   474,    +   520, -  1472,    +   520, -   476, 
   +   520, -   476,    +   520, -  1472,    +   518, -  1478,    +   514, -  1496, 
   +   496, -  1498,    +   550, -  1446,    +   542, -   452,    +   522, -   474, 
   +   520, -   476,    +   518, -   476,    +   520, -   476,    +   520, -   476, 
   +   520, -   474,    +   520, -   476,    +   520, -   476,    +   520, -   476, 
   +   520, -   476,    +   518, -   478,    +   516, -   480,    +   516, -   500, 
   +   494, -   500,    +   494, -   502,    +   548, -   448,    +   546, -   448, 
   +   544, -   452,    +   522, -   474,    +   518, -   476,    +   518, -   478, 
   +   520, -   476,    +   520, -   476,    +   520, -   474,    +   520, -   474, 
   +   522, -   474,    +   520, -   476,    +   520, -   478,    +   518, -   478, 
   +   518, -   480,    +   516, -   502,    +   494, -  1500,    +   548, -  1444, 
   +   544, -  1446,    +   524, -  1468,    +   522, -  2974,    +  3058, -  8902, 
   +   520, -  1472,    +   520, -   476,    +   518, -   476,    +   518, -   476, 
   +   520, -   476,    +   518, -   478,    +   518, -   478,    +   516, -   500, 
   +   496, -   500,    +   494, -  1498,    +   550, -   446,    +   548, -   446, 
   +   546, -  1448,    +   518, -   476,    +   518, -  1472,    +   522, -  1472, 
   +   520, -  1472,    +   520, -  1470,    +   520, -  1474,    +   518, -  1478, 
   +   540, -   456,    +   516, -   500,    +   494, -   502,    +   548, -   448, 
   +   546, -   450,    +   544, -   452,    +   522, -   470,    +   520, -   476, 
   +   518, -   478,    +   520, -   476,    +   520, -   476,    +   518, -   476, 
   +   520, -   476,    +   520, -   474,    +   522, -   476,    +   520, -   476, 
   +   518, -   478,    +   518, -   480,    +   516, -   500,    +   494, -   500, 
   +   496, -   502,    +   548, -   448,    +   546, -   450,    +   544, -   450, 
   +   544, -   452,    +   518, -   478,    +   518, -   478,    +   518, -   476, 
   +   520, -   476,    +   520, -   476,    +   520, -   476,    +   520, -   476, 
   +   518, -   476,    +   520, -   476,    +   518, -   478,    +   516, -   482, 
   +   514, -  2982,    +  2996, -  8980,    +   498, -  1498,    +   546, -   448, 
   +   546, -   450,    +   544, -   452,    +   520, -   474,    +   518, -   478, 
   +   518, -   478,    +   518, -   476,    +   520, -   476,    +   518, -  1474, 
   +   520, -   474,    +   520, -   478,    +   518, -  1476,    +   518, -  1494, 
   +   496, -   500,    +   496, -  1500,    +   548, -   448,    +   544, -  1448, 
   +   542, -  1448,    +   516, -  1474,    +   520, -  1472,    +   520, -  1474, 
   +   518, -  1474,    +   520, -  1474,    +   516, -  1496,    +   498, -   498, 
   +   496, -   500,    +   494, -   500,    +   494, -  1500,    +   542, -  1450, 
   +   520, -  1472,    +   516, -   478,    +   516, -   478,    +   520, -   476, 
   +   520, -   476,    +   520, -   476,    +   518, -  1472,    +   520, -  1474, 
   +   520, -   476,    +   518, -  1494,    +   496, -  1494,    +   498, -   502, 
   +   548, -  1444,    +   544, -   452,    +   544, -  1450,    +   516, -   478, 
   +   518, -   478,    +   520, -   476,    +   520, -   476,    +   520, -   474, 
   +   520, -   476,    +   518, -   476,    +   520, -  1474,    +   516, -  1478, 
   +   544, -  1466,    +   498, -  1500,    +   516

uint16_t rawData[349] = {624, 17854,  2998, 8962,  516, 480,  516, 1476,  516, 502,  494, 502,  548, 448,  544, 450,  546, 450,  544, 452,  522, 474,  518, 1474,  520, 476,  520, 474,  520, 1472,  520, 476,  520, 476,  520, 1472,  518, 1478,  514, 1496,  496, 1498,  550, 1446,  542, 452,  522, 474,  520, 476,  518, 476,  520, 476,  520, 476,  520, 474,  520, 476,  520, 476,  520, 476,  520, 476,  518, 478,  516, 480,  516, 500,  494, 500,  494, 502,  548, 448,  546, 448,  544, 452,  522, 474,  518, 476,  518, 478,  520, 476,  520, 476,  520, 474,  520, 474,  522, 474,  520, 476,  520, 478,  518, 478,  518, 480,  516, 502,  494, 1500,  548, 1444,  544, 1446,  524, 1468,  522, 2974,  3058, 8902,  520, 1472,  520, 476,  518, 476,  518, 476,  520, 476,  518, 478,  518, 478,  516, 500,  496, 500,  494, 1498,  550, 446,  548, 446,  546, 1448,  518, 476,  518, 1472,  522, 1472,  520, 1472,  520, 1470,  520, 1474,  518, 1478,  540, 456,  516, 500,  494, 502,  548, 448,  546, 450,  544, 452,  522, 470,  520, 476,  518, 478,  520, 476,  520, 476,  518, 476,  520, 476,  520, 474,  522, 476,  520, 476,  518, 478,  518, 480,  516, 500,  494, 500,  496, 502,  548, 448,  546, 450,  544, 450,  544, 452,  518, 478,  518, 478,  518, 476,  520, 476,  520, 476,  520, 476,  520, 476,  518, 476,  520, 476,  518, 478,  516, 482,  514, 2982,  2996, 8980,  498, 1498,  546, 448,  546, 450,  544, 452,  520, 474,  518, 478,  518, 478,  518,  [... cut ...] , 500,  494, 1500,  542, 1450,  520, 1472,  516, 478,  516, 478,  520, 476,  520, 476,  520, 476,  518, 1472,  520, 1474,  520, 476,  518, 1494,  496, 1494,  498, 502,  548, 1444,  544, 452,  544, 1450,  516, 478,  518, 478,  520, 476,  520, 476,  520, 474,  520, 476,  518, 476,  520, 1474,  516, 1478,  544, 1466,  498, 1500,  516};  // SAMSUNG_AC
uint8_t state[21] = {0x02, 0x92, 0x0F, 0x00, 0x00, 0x00, 0xF0, 0x01, 0xD2, 0x0F, 0x00, 0x00, 0x00, 0x00, 0x01, 0xB2, 0xFE, 0x71, 0xB0, 0x15, 0xF0};

`