arendst/Tasmota #2117: Sunix RGBWWCW doesn't work correct

ich777, 2018-03-09:
Hello,

i've got the Sunix RGBWWCW Led Controller (looks similar to the AriLux AL-LC11 PWM LED) and flashed it sucessfull with your Firmware, configured it first as this led controller in the WebUI and executed the command "SetOption15 1" in the console, all colors except the warm white work correct (the warm white doesn't work and it mixes the colors of the RGB LED's to look white).

Then i tried to configure it as H801 PWM LED and all colors work but they a switched (warm white = blue & warm white; blue = green; red = warm white; green = blue).

Can i change this in some way?

Also i can give you the GPIOS for the colors:
5 = cold white
12 = green
13 = blue
14 = red
15 = warm white

Also i've attached pictures from the product itself and the PCB (it says ZJ-ESPM-5CH-B V1.3 on the back).

[image]
[image]
[image]

zewelor, 2018-03-09:
Hi

I think thats the same as here: https://github.com/arendst/Sonoff-Tasmota/pull/1414 . It should work correclty when you choose arilux lc06 as board type.

Jason2866, 2018-03-09:
Use search in issue. You will find the solution

ich777, 2018-03-09:
@Jason2866 Thank you for the not helping answer... 👎 
And what should i search for?

timota, 2018-03-09:
Use generic module instead of defined and assign GPIO yourself. 
You can check GPIO in this file https://github.com/arendst/Sonoff-Tasmota/blob/development/sonoff/sonoff_template.h

As an example:

  { "Arilux LC01",     
                       //  (PwmFrequency 1111Hz)
     GPIO_KEY1,        // GPIO00 Optional Button
     GPIO_USER,        // GPIO01 Serial RXD and Optional sensor
     GPIO_LED2_INV,    // GPIO02 RF receiver control
     GPIO_USER,        // GPIO03 Serial TXD and Optional sensor
     GPIO_ARIRFRCV,    // GPIO04 IR or RF receiver (optional)
     GPIO_PWM1,        // GPIO05 RGB LED Red
     0, 0, 0, 0, 0, 0, // Flash connection
     GPIO_PWM2,        // GPIO12 RGB LED Green
     GPIO_PWM3,        // GPIO13 RGB LED Blue
     GPIO_USER,        // GPIO14 RGBW LED White (optional - set to PWM4 for Cold White or Warm White)
     0, 0, 0
  },

where your GPIOs are:
GPIO_PWM1,        // GPIO05 RGB LED Red
GPIO_PWM2,        // GPIO12 RGB LED Green
GPIO_PWM3,        // GPIO13 RGB LED Blue
GPIO_USER,        // GPIO14 RGBW LED White

in Generic module configuration choose GPIO pin and choose correspond PWM. You can find correct mapping Color <-> GPIO. Thats it.

ElwayDenmark, 2019-02-10:
> Hello,
> 
> i've got the Sunix RGBWWCW Led Controller (looks similar to the AriLux AL-LC06 PWM LED) and flashed it sucessfull with your Firmware, configured it first as this led controller in the WebUI and executed the command "SetOption15 1" in the console, all colors except the warm white work correct (the warm white doesn't work and it mixes the colors of the RGB LED's to look white).
> 
> Then i tried to configure it as H801 PWM LED and all colors work but they a switched (warm white = blue & warm white; blue = green; red = warm white; green = blue).
> 
> Can i change this in some way?
> 
> Also i can give you the GPIOS for the colors:
> 5 = cold white
> 12 = green
> 13 = blue
> 14 = red
> 15 = warm white
> 
> Also i've attached pictures from the product itself and the PCB (it says ZJ-ESPM-5CH-B V1.3 on the back).
> 
> Edit: Forget to say that i am using it with Home-Assistant and followed the guide at https://github.com/arendst/Sonoff-Tasmota/wiki/Home-Assistant (MagicHome-Led Controller).
> 
> [image]
> [image]
> [image]

how did you flash it? I have flashed a few devices with Tasmota - but this one, does not have GND. It has IO, RT, TX, -V, 3.3.... When I flash other devices, I have RT, TX, GND, 3.3v that I connect to my FTDI board for flashing. I cannot find info anywhere on how to hook up this for flashing.

Jason2866, 2019-02-10:
-V = GND

ElwayDenmark, 2019-02-10:
> -V = GND

ohh... I feel a bit stupid now. Will try - Thanks!

ElwayDenmark, 2019-02-10:
> > -V = GND
> 
> ohh... I feel a bit stupid now. Will try - Thanks!

That did it - many thanks! Now running on Tasmota. Now I only have to figure out how to control the light. Getting there... :-)

ich777, 2019-02-11:
> That did it - many thanks! Now running on Tasmota. Now I only have to figure out how to control the light. Getting there... :-)

Have you managed to get all the channels working as intended? If yes, please share it.

ElwayDenmark, 2019-02-12:
> > That did it - many thanks! Now running on Tasmota. Now I only have to figure out how to control the light. Getting there... :-)
> 
> @ElwayDenmark Have you managed to get all the channels working as intended? If yes, please share it.

I run it with only one color channel - warm white channel (WW). My led strip is also just a white-only one - and this was always my plan. It runs under the kitchen cabinets to light up/reflect on the floor - so I never wanted full color palette - just white.

When I got into the tasmota interface, I changed the type to Arilux 06. Now, I can toggle on/off and set brightness (via slider) in Tasmota. I played a bit with the GPIO settings in tasmota to get it working. I think I set GPIO 15 to "PVM4", but I have to re-check when I get home (at work now). I never managed to find any documentation (at least none that I could understand) on these GPIO mappings - so I just played with a bunch of combinations - and eventually found on that works.

I did not yet include it into Home Assistant, but will try later. I think there is a "Flux Led" plugin that should work with LED controllers that out-of-the-box come with MagicHome app (which mine did).

ich777, 2019-02-12:
@ElwayDenmark Thanks for the feedback, i've never got it to run the right way with tasmota so i've changed to "espruna" and everything worked perfectly OOB and integrated it into HomeAssistant, there is a well written documentation on the espruna github for integrating it into HomeAssistant.

kellertobias, 2019-07-20:
With the new Generic Board type, where you can setup the GPIOs as you want, it works, when you set the colors to PWM1...PWM5

MrAlfabet, 2019-08-13:
Did you just flash the sonoff.bin? I've tried flashing sonoff, sonoff basic, erasing then flashing, but I'm not seeing a sonoff SSID (also not after power cycle). Do I have a dud?

fremut, 2019-09-11:
GPIO05 ... CW, GPIO15 ... WW, GPIO13 ... B, GPIO12 ... G, GPIO14 ... R, GPIO02 ... onboard green status LED

SilverFire, 2020-04-05:
In order to use AriLux AL-LC06 in CW/WW mode set

GPIO05 > PWM1
GPIO15 > PWM2

Or use the following config:

`{"NAME":"Arilux LC06 in WW/CW mode","GPIO":[17,0,0,0,0,37,0,0,0,0,0,38,0],"FLAG":0,"BASE":18}`

peterdey, 2020-05-31:
How did you manage to flash these boards?

Using an external 3.3V supply connected to 3.3 and V-; FTDI board crossed over to TX and RX.
I can see the bootloader messages at 74880 baud, and it "seems" to go into program mode holding down the button at R25.

But when I attempt to flash (tried both esptool.py and Tasmotizer - on both Windows and Ubuntu), it just times out attempting to connect:

```
Connecting........_____....._____....._____....._____....._____....._____.....____
.
.
.

esptool.FatalError: Timed out waiting for packet header
```

Am I missing something here?

markianwallace, 2020-06-09:
> In order to use AriLux AL-LC06 in CW/WW mode set
> 
> GPIO05 > PWM1
> GPIO15 > PWM2
> 
> Or use the following config:
> 

> In order to use AriLux AL-LC06 in CW/WW mode set
> 
> GPIO05 > PWM1
> GPIO15 > PWM2
> 
> Or use the following config:
> 
> `{"NAME":"Arilux LC06 in WW/CW mode","GPIO":[17,0,0,0,0,37,0,0,0,0,0,38,0],"FLAG":0,"BASE":18}`

Just to confirm, this is what fixed it for me.