arendst/Tasmota #2118: Report air quality and sound level from Sonoff SC to Domoticz (resolved)

mikizvi, 2018-03-09:
Hi,

The patch below is needed to get the Sonoff SC to report air quality and sound level to Domoticz.  The sound level is reported as the "Sensor idx 6 Count" (air quality is already "Sensor idx 9 AirQuality").

diff --git a/sonoff/xsns_04_snfsc.ino b/sonoff/xsns_04_snfsc.ino
index 51e3e1b..3c8ca23 100644
--- a/sonoff/xsns_04_snfsc.ino
+++ b/sonoff/xsns_04_snfsc.ino
@@ -125,6 +125,8 @@ void SonoffScShow(boolean json)
 #ifdef USE_DOMOTICZ
       DomoticzTempHumSensor(temperature, humidity);
       DomoticzSensor(DZ_ILLUMINANCE, sc_value[2]);
+      DomoticzSensor(DZ_COUNT, sc_value[3]);
+      DomoticzSensor(DZ_AIRQUALITY, sc_value[4]);
 #endif  // USE_DOMOTICZ
 #ifdef USE_WEBSERVER
     } else {

Hardware: Sonoff SC V2

Status 0:
12:28:11 CMD: status 0
12:28:11 MQT: domain/stat/sonoff_sc/STATUS = {"Status":{"Module":21,"FriendlyName":"SonoffSc","Topic":"sonoff_sc","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
12:28:11 MQT: domain/stat/sonoff_sc/STATUS1 = {"StatusPRM":{"Baudrate":19200,"GroupTopic":"sonoffs","OtaUrl":"http://192.168.1.81:8008/api/arduino/sonoff.ino.bin","Uptime":"0T00:09:20","StartupUTC":"2018-03-09T10:18:51","Sleep":0,"BootCount":10,"SaveCount":18,"SaveAddress":"FA000"}}
12:28:11 MQT: domain/stat/sonoff_sc/STATUS2 = {"StatusFWR":{"Version":"5.12.0d","BuildDateTime":"2018-03-09T12:14:29","Boot":6,"Core":"2_4_0","SDK":"2.1.0(deb1901)"}}
12:28:11 MQT: domain/stat/sonoff_sc/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId1":"SSID","SSId2":"SSIDMichael","TelePeriod":300,"SetOption":"00000009"}}
12:28:11 MQT: domain/stat/sonoff_sc/STATUS4 = {"StatusMEM":{"ProgramSize":481,"Free":520,"Heap":20,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3}}
12:28:11 MQT: domain/stat/sonoff_sc/STATUS5 = {"StatusNET":{"Hostname":"sonoff_sc-1636","IPAddress":"192.168.1.119","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"68:C6:3A:80:A6:64","Webserver":2,"WifiConfig":3}}
12:28:11 MQT: domain/stat/sonoff_sc/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.81","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_80A664","MqttUser":"MQTT","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
12:28:11 MQT: domain/stat/sonoff_sc/STATUS7 = {"StatusTIM":{"UTC":"Fri Mar 09 10:28:11 2018","Local":"Fri Mar 09 12:28:11 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":2}}
12:28:11 MQT: domoticz/in = {"idx":11,"nvalue":0,"svalue":"25.0;17.0;2"}
12:28:11 MQT: domoticz/in = {"idx":12,"nvalue":0,"svalue":"20"}
12:28:11 MQT: domoticz/in = {"idx":14,"nvalue":0,"svalue":"30"}
12:28:11 MQT: domoticz/in = {"idx":13,"nvalue":90}
12:28:11 MQT: domain/stat/sonoff_sc/STATUS10 = {"StatusSNS":{"Time":"2018-03-09T12:28:11","SonoffSC":{"Temperature":25.0,"Humidity":17.0,"Light":20,"Noise":30,"AirQuality":90},"TempUnit":"C"}}
12:28:11 MQT: domain/stat/sonoff_sc/STATUS11 = {"StatusSTS":{"Time":"2018-03-09T12:28:11","Uptime":"0T00:09:20","Vcc":3.514,"Wifi":{"AP":1,"SSId":"SSID","RSSI":100,"APMac":"80:1F:02:C5:4B:7C"}}}

arendst, 2018-03-09:
As Air quality is related to CO2 PPM in domoticz I suggest I will add a formula to the simple Sonoff SC air quality percentage as 100% would be 500ppm (Excellent) to 10% as 2300ppm (Poor) - 500 + ((100 - sc_value[4]) * 20)

closed 2018-03-09 by arendst

mikizvi, 2018-04-25:
@arendst Can I ask you to please add a similar solution for the BME680 VOC measurement -> Domoticz?

As an initial hack, I tried the change below, but it's reporting air quality of 0 constantly (the web page reports kOhms in the range 20 to 200).

`$ git diff sonoff/xsns_09_bmp.ino
diff --git a/sonoff/xsns_09_bmp.ino b/sonoff/xsns_09_bmp.ino
index 01eb297..5b9c95d 100644
--- a/sonoff/xsns_09_bmp.ino
+++ b/sonoff/xsns_09_bmp.ino
@@ -465,7 +465,10 @@ void BmpShow(boolean json)
         mqtt_data, bmp_name, temperature, (bmp_model >= 2) ? json_humidity : "", pressure, (Settings.altitude != 0) ? json_sealevel : "");
 #endif  // USE_BME680
 #ifdef USE_DOMOTICZ
-      if (0 == tele_period) DomoticzTempHumPressureSensor(temperature, humidity, pressure);
+      if (0 == tele_period) {
+        DomoticzTempHumPressureSensor(temperature, humidity, pressure);
+        DomoticzSensor(DZ_AIRQUALITY, gas_resistance);
+      }
 #endif // USE_DOMOTICZ

 #ifdef USE_KNX
`

arendst, 2018-04-26:
Could you replace the following code:
```
#ifdef USE_DOMOTICZ
      if (0 == tele_period) DomoticzTempHumPressureSensor(temperature, humidity, pressure);
#endif // USE_DOMOTICZ
```
with the one below:
```
#ifdef USE_DOMOTICZ
      if (0 == tele_period) {
        DomoticzTempHumPressureSensor(temperature, humidity, pressure);
#ifdef USE_BME680
        if (bmp_model >= 3) { DomoticzSensor(DZ_AIRQUALITY, gas_resistance); }
//        if (bmp_model >= 3) { DomoticzSensor(DZ_AIRQUALITY, (char *)gas_resistance); }
#endif  // USE_BME680
      }
#endif // USE_DOMOTICZ
```
Play with the two gas_resistance lines and report which one provides to correct result.

mikizvi, 2018-04-26:
This one reports the kOhm value correctly:

~~~~
#ifdef USE_DOMOTICZ
      if (0 == tele_period) {
        DomoticzTempHumPressureSensor(temperature, humidity, pressure);
#ifdef USE_BME680
//        if (bmp_model >= 3) { DomoticzSensor(DZ_AIRQUALITY, gas_resistance); }
//        if (bmp_model >= 3) { DomoticzSensor(DZ_AIRQUALITY, (char *)gas_resistance); }
        if (bmp_model >= 3) { DomoticzSensor(DZ_AIRQUALITY, (uint32_t) g); }
#endif  // USE_BME680
~~~~

Domoticz is reporting this a ppm rather than kOhm.  Would appreciate ideas how to massage this into something that Domoticz could interpret as indoor air quality.  This code is from the Domoticz main/WebServer.cpp file:

~~~~
                                                if (airquality < 700)
                                                        root["result"][ii]["Quality"] = "Excellent";
                                                else if (airquality < 900)
                                                        root["result"][ii]["Quality"] = "Good";
                                                else if (airquality < 1100)
                                                        root["result"][ii]["Quality"] = "Fair";
                                                else if (airquality < 1600)
                                                        root["result"][ii]["Quality"] = "Mediocre";
                                                else
                                                        root["result"][ii]["Quality"] = "Bad";
~~~~