arendst/Tasmota #2119: Relay ON after reboot

alexmc1510, 2018-03-09:
Make sure these boxes are checked before submitting your issue - Thank you!

- Sonff Dual R2

Hi team,

I am experiencing an issue related relay status after reboot/energy failure.
I have a sonoff dual R2 installed in a roller shutter and a raspberry pi with domoticz and mosquitto as broker. My problem is very simple. Every time the sonoff reboot, the roller shutter opens automatically. I have buttons programmed in toggle mode. I have tried poweronstate command, I also have tried disonnecting Raspberry. All tests with bad results. I attached logfile after reboot.

00:00:00 Project sonoff Persianapasillo (Topic persianapasillo, Fallback persianapasillo, GroupTopic sonoffs) Version 5.12.0
00:00:00 RSL: domoticz/in = {"idx":1,"nvalue":1,"svalue":""}
00:00:00 RSL: stat/persianapasillo/RESULT = {"POWER1":"ON"}
00:00:00 RSL: stat/persianapasillo/POWER1 = ON
00:00:00 WIF: Connecting to AP1 vodafone7630 in mode 11N as persianapasillo-6340...
00:00:06 WIF: Connected
00:00:06 DNS: Initialized
00:00:06 HTP: Web server active on persianapasillo-6340.local with IP address 192.168.0.158
00:12:48 MQT: Attempting connection...
00:12:48 MQT: Connected

Can you provide a solution? Am I doing something wrong?

Thank you for your support.

Regards,

curzon01, 2018-03-09:
- [ ] Provide the output of command ``Status 0``
- [ ] Provide the output of command ``PowerOnState``
- [ ] Provide the output of command ``PowerRetain``
- [ ] Provide the output of command ``SwitchMode1`` and ``SwitchMode2``
- [ ] Provide the output of command ``SetOption14``

alexmc1510, 2018-03-09:
19:32:05 CMD: status 0
19:32:05 MQT: stat/persianapasillo/STATUS = {"Status":{"Module":39,"FriendlyName":"Persianapasillo","Topic":"persianapasillo","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
19:32:05 MQT: stat/persianapasillo/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://domus1:80/api/arduino/sonoff.ino.bin","Uptime":"0T07:53:16","Sleep":0,"BootCount":73,"SaveCount":370,"SaveAddress":"FA000"}}
19:32:05 MQT: stat/persianapasillo/STATUS2 = {"StatusFWR":{"Version":"5.12.0","BuildDateTime":"2018-03-03T16:35:50","Boot":31,"Core":"2_4_0","SDK":"2.1.0(deb1901)"}}
19:32:05 MQT: stat/persianapasillo/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"domus1","LogPort":514,"SSId1":"vodafone7630","SSId2":"indebuurt2","TelePeriod":300,"SetOption":"00004009"}}
19:32:05 MQT: stat/persianapasillo/STATUS4 = {"StatusMEM":{"ProgramSize":495,"Free":508,"Heap":18,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3}}
19:32:05 MQT: stat/persianapasillo/STATUS5 = {"StatusNET":{"Hostname":"persianapasillo-6340","IPAddress":"192.168.0.158","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.1","Mac":"60:01:94:9D:78:C4","Webserver":2,"WifiConfig":3}}
19:32:05 MQT: stat/persianapasillo/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.10","MqttPort":1883,"MqttClientMask":"persianapasillo","MqttClient":"persianapasillo","MqttUser":"DVES_USER","MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
19:32:05 MQT: stat/persianapasillo/STATUS7 = {"StatusTIM":{"UTC":"Fri Mar 09 18:32:05 2018","Local":"Fri Mar 09 19:32:05 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1}}
19:32:05 MQT: stat/persianapasillo/STATUS10 = {"StatusSNS":{"Time":"2018-03-09T19:32:05","Switch1":"ON","Switch2":"ON"}}
19:32:05 MQT: stat/persianapasillo/STATUS11 = {"StatusSTS":{"Time":"2018-03-09T19:32:05","Uptime":"0T07:53:16","Vcc":3.485,"POWER1":"ON","POWER2":"OFF","Wifi":{"AP":1,"SSId":"vodafone7630","RSSI":70,"APMac":"80:3F:5D:6D:69:A8"}}}

19:32:54 CMD: poweronstate
19:32:54 MQT: stat/persianapasillo/RESULT = {"PowerOnState":3}

19:33:19 CMD: powerretain
19:33:19 MQT: stat/persianapasillo/RESULT = {"PowerRetain":"OFF"}

19:33:32 CMD: SwitchMode1
19:33:32 MQT: stat/persianapasillo/RESULT = {"SwitchMode1":0}

19:34:00 CMD: switchmode2
19:34:00 MQT: stat/persianapasillo/RESULT = {"SwitchMode2":0}

19:34:16 CMD: setoption14
19:34:16 MQT: stat/persianapasillo/RESULT = {"SetOption14":"ON"}

Last action I don't know how to perform it.

alexmc1510, 2018-03-09:
I have noticed that every 5 minutes I have receiving a MQTT order or info. Fortunately, I have interlock mode activated because I am receiving both relays ON at the same time.

18:44:12 MQT: tele/persianapasillo/STATE = {"Time":"2018-03-09T18:44:12","Uptime":"0T07:05:23","Vcc":3.484,"POWER1":"ON","POWER2":"OFF","Wifi":{"AP":1,"SSId":"vodafone7630","RSSI":60,"APMac":"80:3F:5D:6D:69:A8"}}
18:44:12 MQT: tele/persianapasillo/SENSOR = {"Time":"2018-03-09T18:44:12","Switch1":"ON","Switch2":"ON"}
18:49:12 MQT: tele/persianapasillo/STATE = {"Time":"2018-03-09T18:49:12","Uptime":"0T07:10:23","Vcc":3.482,"POWER1":"ON","POWER2":"OFF","Wifi":{"AP":1,"SSId":"vodafone7630","RSSI":72,"APMac":"80:3F:5D:6D:69:A8"}}
18:49:12 MQT: tele/persianapasillo/SENSOR = {"Time":"2018-03-09T18:49:12","Switch1":"ON","Switch2":"ON"}
18:54:12 MQT: tele/persianapasillo/STATE = {"Time":"2018-03-09T18:54:12","Uptime":"0T07:15:23","Vcc":3.482,"POWER1":"ON","POWER2":"OFF","Wifi":{"AP":1,"SSId":"vodafone7630","RSSI":56,"APMac":"80:3F:5D:6D:69:A8"}}
18:54:12 MQT: tele/persianapasillo/SENSOR = {"Time":"2018-03-09T18:54:12","Switch1":"ON","Switch2":"ON"}
18:59:12 MQT: tele/persianapasillo/STATE = {"Time":"2018-03-09T18:59:12","Uptime":"0T07:20:23","Vcc":3.481,"POWER1":"ON","POWER2":"OFF","Wifi":{"AP":1,"SSId":"vodafone7630","RSSI":56,"APMac":"80:3F:5D:6D:69:A8"}}
18:59:12 MQT: tele/persianapasillo/SENSOR = {"Time":"2018-03-09T18:59:12","Switch1":"ON","Switch2":"ON"}

curzon01, 2018-03-09:
How do you has connected Switches for your Dual R2?

What's the output for `Gpio`?

alexmc1510, 2018-03-09:
Directly connected, without pull-up/down. 1,5m cable

curzon01, 2018-03-10:
What's the output for `Gpio`?

alexmc1510, 2018-03-10:
I have tried what @Jason2866  has said with same results. @curzon01 I have made a capture of module configuration.
[image]

Jason2866, 2018-03-10:
Hmmm, we have to try some things to get closer...
Setup GPIO 0 to none. Connect a Pulldown resistor (4k7) on GPIO 9.
Does it work with this setup?

alexmc1510, 2018-03-10:
Hi guys. Good news!! Thanks @Jason2866. We are closer to the solution. I have setup GPIO0 to None (without doing anything in GPIO 9) and the roller shutter doesn't move after reboot. Now it is only necessary to know what are happening.
Thanks,

curzon01, 2018-03-10:
Simply use pull-ups (2k2 - 10k is enough) on each - GPIO0 and GPIO9

Jason2866, 2018-03-10:
For GPIO 0 you can only use a pull up resistor! With a pulldown you will be in flash mode after power down and power up.

curzon01, 2018-03-10:
It's difficult to help at all, because we do not know if alexmc1510 switches agains GND or Vcc.

@alexmc1510: Switch to GND and use Pullups or use other PIN instead GPIO0 for Switch1 (e.g. serial pins are also working if you switch `SerialLog OFF`)

alexmc1510, 2018-03-11:
Hello all,

Finally I have solved it using TX and RX pins as button inputs disabling seriallog. After all the trials done my conclusion is that GPIO0 activates without order on every reboot. If I disconnect it, sonoff works fine. Probably, pull-up resistor doesn't exists in this pin. From my point of view this topic can be closed. I will maintain it opened until tomorrow if someone want to add some comments. Thanks a lot for your support!!

curzon01, 2018-03-11:
> From my point of view this topic can be closed.

You are the maintainer of your issue, you can and have to close your own issue.

closed 2018-06-02 by ascillato