arendst/Tasmota #2120: MQTT and SonOff 4h PRO to command physical switch

SaverioConigliaro, 2018-03-09:
Hello,
I have a physical switch (just an ordinary one used to switch on/off a couple of light into my house) and I'd like to use one on my sonoff relè channel to switch that switch (sorry for the words trick) ON or OFF... Is it possible? I of course need to preserve physical use for the switch but I also want to press it ON or OFF using SonOff from remote... How to cable it?
Thanks!!

Xavier

SaverioConigliaro, 2018-03-09:
[image]

Leseratte10, 2018-03-09:
Wiring it like that doesn't work, you wouldn't be able to turn the light off like that. 

Completely remove the switch from any wiring, so it's just Phase and Neutral from power source to Sonoff, and Phase and Neutral from Sonoff to light, without any switch; then add the switch inbetween GPIO14 and GND (the pins inside the sonoff).

SaverioConigliaro, 2018-03-09:
Not my solution because I don't want to connect all the three lights into my SonOff 4 Pro and use the button soldered as you described because in this way I am going to use all the sonoff ports... I just want to control the wall switch. I image to leave it ON the I leave the house and tell sonoff to keep the contact open until, for example, I have a motion event on my ip camera....if that events occurs the sonoff closes the contact and lights are on.

curzon01, 2018-03-10:
This can be work using a normal [Multiway switching](https://en.wikipedia.org/wiki/Multiway_switching) solution but be aware using this solution the 4CH channel works as a toggle. You can not know if ON/OFF switches the lamp off or on.

Condition: Your wall switch is a SPDT (Single Pole Double Throw - or "three-way"), not a simple On/Off - your  4CH Pro aleady has 4x SPDT.

To have a real on/off, it would be possible but than you need to define which switch has priority.

The [two three-ways switches picture on wiki](https://upload.wikimedia.org/wikipedia/commons/thumb/2/2e/3-way_switches_position_1.svg/300px-3-way_switches_position_1.svg.png) shows how to wire:
- the left "three-way" is your wall-switch
- the right "three-way" is one of your 4CH Pro relay

Here the full article in [English](https://en.wikipedia.org/wiki/Multiway_switching) and [German](https://de.wikipedia.org/wiki/Wechselschaltung).

SaverioConigliaro, 2018-03-15:
I solved connecting N wire from the phisical button to the sonoff. In this way i can simulate button press via software... But... There is a problem: if i press the phisical button 1 time the two right lamps switch on, pressing It the second time makes the two lefts lamps to switch on and finally by pressing button for the third time I Can see all 4 lamps on. This is a problem because from remote i can't read from sonoff the state of the lights. I can't tell to my sonoff ti switch them all ON because It is not able ti understand which is the starting state. Any hint?