arendst/Tasmota #2124: Compiling error

darkdevil91, 2018-03-09:
Hi, I'm having some problems compiling firmware for my Sonoff basic. via Arduino IDE.
I'm following this guide and when I try to compile/verify the .ino file I have this error:

Arduino:1.8.5 (Windows 10), Scheda:"Generic ESP8266 Module, Serial, 80 MHz, nodemcu, 26 MHz, 40MHz, DOUT, 1M (no SPIFFS), v1.4 Prebuilt, Disabled, None, 115200"
Arduino:1.8.5 (Windows 10), Scheda:"Generic ESP8266 Module, Serial, 80 MHz, nodemcu, 26 MHz, 40MHz, DOUT, 1M (no SPIFFS), v1.4 Prebuilt, Disabled, None, 115200"

C:\Users\mikyv\Desktop\Nuova cartella\arduino-1.8.5\arduino-builder -dump-prefs -logger=machine -hardware C:\Users\mikyv\Desktop\Nuova cartella\arduino-1.8.5\hardware -hardware C:\Users\mikyv\Desktop\Nuova cartella\arduino-1.8.5\portable\packages -tools C:\Users\mikyv\Desktop\Nuova cartella\arduino-1.8.5\tools-builder -tools C:\Users\mikyv\Desktop\Nuova cartella\arduino-1.8.5\hardware\tools\avr -tools C:\Users\mikyv\Desktop\Nuova cartella\arduino-1.8.5\portable\packages -built-in-libraries C:\Users\mikyv\Desktop\Nuova cartella\arduino-1.8.5\libraries -libraries C:\Users\mikyv\Desktop\Nuova cartella\arduino-1.8.5\portable\sketchbook\libraries -fqbn=esp8266:esp8266:generic:UploadTool=esptool,CpuFrequency=80,ResetMethod=nodemcu,CrystalFreq=26,FlashFreq=40,FlashMode=dout,FlashSize=1M0,LwIPVariant=Prebuilt,Debug=Disabled,DebugLevel=None____,UploadSpeed=115200 -ide-version=10805 -build-path C:\Users\mikyv\AppData\Local\Temp\arduino_build_638301 -warnings=none -build-cache C:\Users\mikyv\AppData\Local\Temp\arduino_cache_966481 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.mkspiffs.path=C:\Users\mikyv\Desktop\Nuova cartella\arduino-1.8.5\portable\packages\esp8266\tools\mkspiffs\0.2.0 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Users\mikyv\Desktop\Nuova cartella\arduino-1.8.5\portable\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2 -prefs=runtime.tools.esptool.path=C:\Users\mikyv\Desktop\Nuova cartella\arduino-1.8.5\portable\packages\esp8266\tools\esptool\0.4.13 -verbose C:\Users\mikyv\Desktop\Nuova cartella\arduino-1.8.5\portable\sketchbook\sonoff\sonoff.ino
C:\Users\mikyv\Desktop\Nuova cartella\arduino-1.8.5\arduino-builder -compile -logger=machine -hardware C:\Users\mikyv\Desktop\Nuova cartella\arduino-1.8.5\hardware -hardware C:\Users\mikyv\Desktop\Nuova cartella\arduino-1.8.5\portable\packages -tools C:\Users\mikyv\Desktop\Nuova cartella\arduino-1.8.5\tools-builder -tools C:\Users\mikyv\Desktop\Nuova cartella\arduino-1.8.5\hardware\tools\avr -tools C:\Users\mikyv\Desktop\Nuova cartella\arduino-1.8.5\portable\packages -built-in-libraries C:\Users\mikyv\Desktop\Nuova cartella\arduino-1.8.5\libraries -libraries C:\Users\mikyv\Desktop\Nuova cartella\arduino-1.8.5\portable\sketchbook\libraries -fqbn=esp8266:esp8266:generic:UploadTool=esptool,CpuFrequency=80,ResetMethod=nodemcu,CrystalFreq=26,FlashFreq=40,FlashMode=dout,FlashSize=1M0,LwIPVariant=Prebuilt,Debug=Disabled,DebugLevel=None____,UploadSpeed=115200 -ide-version=10805 -build-path C:\Users\mikyv\AppData\Local\Temp\arduino_build_638301 -warnings=none -build-cache C:\Users\mikyv\AppData\Local\Temp\arduino_cache_966481 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.mkspiffs.path=C:\Users\mikyv\Desktop\Nuova cartella\arduino-1.8.5\portable\packages\esp8266\tools\mkspiffs\0.2.0 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Users\mikyv\Desktop\Nuova cartella\arduino-1.8.5\portable\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2 -prefs=runtime.tools.esptool.path=C:\Users\mikyv\Desktop\Nuova cartella\arduino-1.8.5\portable\packages\esp8266\tools\esptool\0.4.13 -verbose C:\Users\mikyv\Desktop\Nuova cartella\arduino-1.8.5\portable\sketchbook\sonoff\sonoff.ino
Using board 'generic' from platform in folder: C:\Users\mikyv\Desktop\Nuova cartella\arduino-1.8.5\portable\packages\esp8266\hardware\esp8266\2.4.1
Using core 'esp8266' from platform in folder: C:\Users\mikyv\Desktop\Nuova cartella\arduino-1.8.5\portable\packages\esp8266\hardware\esp8266\2.4.1
Detecting libraries used...
"C:\Users\mikyv\Desktop\Nuova cartella\arduino-1.8.5\portable\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\mikyv\Desktop\Nuova cartella\arduino-1.8.5\portable\packages\esp8266\hardware\esp8266\2.4.1/tools/sdk/include" "-IC:\Users\mikyv\Desktop\Nuova cartella\arduino-1.8.5\portable\packages\esp8266\hardware\esp8266\2.4.1/tools/sdk/lwip/include" "-IC:\Users\mikyv\Desktop\Nuova cartella\arduino-1.8.5\portable\packages\esp8266\hardware\esp8266\2.4.1/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\Users\mikyv\AppData\Local\Temp\arduino_build_638301/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC   -DARDUINO=10805 -DARDUINO_ESP8266_GENERIC -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_GENERIC"  -DESP8266 "-IC:\Users\mikyv\Desktop\Nuova cartella\arduino-1.8.5\portable\packages\esp8266\hardware\esp8266\2.4.1\cores\esp8266" "-IC:\Users\mikyv\Desktop\Nuova cartella\arduino-1.8.5\portable\packages\esp8266\hardware\esp8266\2.4.1\variants\generic" "C:\Users\mikyv\AppData\Local\Temp\arduino_build_638301\sketch\sonoff.ino.cpp" -o "nul"
"C:\Users\mikyv\Desktop\Nuova cartella\arduino-1.8.5\portable\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\mikyv\Desktop\Nuova cartella\arduino-1.8.5\portable\packages\esp8266\hardware\esp8266\2.4.1/tools/sdk/include" "-IC:\Users\mikyv\Desktop\Nuova cartella\arduino-1.8.5\portable\packages\esp8266\hardware\esp8266\2.4.1/tools/sdk/lwip/include" "-IC:\Users\mikyv\Desktop\Nuova cartella\arduino-1.8.5\portable\packages\esp8266\hardware\esp8266\2.4.1/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\Users\mikyv\AppData\Local\Temp\arduino_build_638301/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC   -DARDUINO=10805 -DARDUINO_ESP8266_GENERIC -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_GENERIC"  -DESP8266 "-IC:\Users\mikyv\Desktop\Nuova cartella\arduino-1.8.5\portable\packages\esp8266\hardware\esp8266\2.4.1\cores\esp8266" "-IC:\Users\mikyv\Desktop\Nuova cartella\arduino-1.8.5\portable\packages\esp8266\hardware\esp8266\2.4.1\variants\generic" "-IC:\Users\mikyv\Desktop\Nuova cartella\arduino-1.8.5\portable\sketchbook\libraries\PubSubClient-2.6.09\src" "C:\Users\mikyv\AppData\Local\Temp\arduino_build_638301\sketch\sonoff.ino.cpp" -o "nul"
"C:\Users\mikyv\Desktop\Nuova cartella\arduino-1.8.5\portable\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\mikyv\Desktop\Nuova cartella\arduino-1.8.5\portable\packages\esp8266\hardware\esp8266\2.4.1/tools/sdk/include" "-IC:\Users\mikyv\Desktop\Nuova cartella\arduino-1.8.5\portable\packages\esp8266\hardware\esp8266\2.4.1/tools/sdk/lwip/include" "-IC:\Users\mikyv\Desktop\Nuova cartella\arduino-1.8.5\portable\packages\esp8266\hardware\esp8266\2.4.1/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\Users\mikyv\AppData\Local\Temp\arduino_build_638301/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC   -DARDUINO=10805 -DARDUINO_ESP8266_GENERIC -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_GENERIC"  -DESP8266 "-IC:\Users\mikyv\Desktop\Nuova cartella\arduino-1.8.5\portable\packages\esp8266\hardware\esp8266\2.4.1\cores\esp8266" "-IC:\Users\mikyv\Desktop\Nuova cartella\arduino-1.8.5\portable\packages\esp8266\hardware\esp8266\2.4.1\variants\generic" "-IC:\Users\mikyv\Desktop\Nuova cartella\arduino-1.8.5\portable\sketchbook\libraries\PubSubClient-2.6.09\src" "-IC:\Users\mikyv\Desktop\Nuova cartella\arduino-1.8.5\portable\packages\esp8266\hardware\esp8266\2.4.1\libraries\Ticker" "C:\Users\mikyv\AppData\Local\Temp\arduino_build_638301\sketch\sonoff.ino.cpp" -o "nul"
"C:\Users\mikyv\Desktop\Nuova cartella\arduino-1.8.5\portable\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\mikyv\Desktop\Nuova cartella\ard [... cut ...]
Uso la libreria TasmotaSerial-1.1.0 alla versione 1.0 nella cartella: C:\Users\mikyv\Desktop\Nuova cartella\arduino-1.8.5\portable\sketchbook\libraries\TasmotaSerial-1.1.0 
Uso la libreria NeoPixelBus-2.2.9 alla versione 2.2.9 nella cartella: C:\Users\mikyv\Desktop\Nuova cartella\arduino-1.8.5\portable\sketchbook\libraries\NeoPixelBus-2.2.9 
Uso la libreria SPI alla versione 1.0 nella cartella: C:\Users\mikyv\Desktop\Nuova cartella\arduino-1.8.5\portable\packages\esp8266\hardware\esp8266\2.4.1\libraries\SPI 
exit status 1
Errore durante la compilazione per la scheda Generic ESP8266 Module.

What could we do? I'm new to Arduino and sonoff firmwares

Snipercaine, 2018-03-13:
First guess is there's a problem with the board manager.  
You have the Additional Boards Manager URL in the Preferences page? 
https://www.dropbox.com/s/vvxm4m454n85thp/Screenshot%202018-03-13%2011.23.29.png?dl=0

Then go to Tools - Board - Boards Manager, then search for ESP8266 and Install it. 

Someone was just telling me today there may be a problem with ver 2.4.1 of the board manager.  
You may need to select ver 2.4.0 when you install. 
https://www.dropbox.com/s/qi7l1cvst0zrq1z/Screenshot%202018-03-13%2011.32.16.png?dl=0

angrytomatoe, 2018-03-20:
I ran into the same issue today. Afer downgrading via Boards Manager to esp8266 board version 2.4.0 everything compiled smoothly 👍 :-)

cashflow4me, 2018-03-28:
I was able to compile and flash with 2.4.1 as per jasetheace2 instructions