arendst/Tasmota #2125: Home Assistant don't show the real status off the Sonoff

tuga49, 2018-03-10:
Hello everyone,

First off all, I'm new to Home Assistant. I have hass.io running on my raspberry pi3. Yesterday, I configured mqtt,samba,ssh and duckdns. Today I flashed tasmota on two Sonoff S20 with success. Now, on home assistant, some times it work and some times it dont work well. Some times, for example, I turn on the S20, then it show me sonoff S20 is off, but its on. And some times I can change the stauts. I change, but it back again to the previous status. And some times it work well without problem. I have latest version of HA and latest firmware on tasmota. Why it's happening? 
NOTE: This happen on both Sonoffs

Please remember I'm new, and explain me what should I do with steps that I can understant. Thanks

codenamej6, 2018-03-10:
I'm having a similar problem, although not with every flashed sonoffs.  I have 5 that work perfectly that I set up a week or two ago, but the ones I tried doing this week using the same method keep disconnecting/reconnecting or giving me timeout issues.  I'm also very new to this, so I'm trying to learn why it's doing this so that I can fix it.
In home assistant, if you click the hass.io button on the left menu, then click the mosquitto mqtt add-on and scroll down, there's a log there of all the activity.  There might be a clue there as to what's happening and to see if it's disconnecting for any reason.  What does your log say?

tuga49, 2018-03-10:
Hi, I solved it. My problem was with MQTT of the sonoff. They cant have the same Client ID. Use DVES_%06X in all of them, or make sure to keep all different. I used DVES_%06X in both of them and now it work :) but now Im facing other issue. I have them connected to the wall, it doest matter if is on or off. If the lighs goes off, when it came back, they both turns on. I wan't they to keep the last state or then  keep off. Dont know why it happen

issacg, 2018-03-11:
Have you looked at the [PowerOnState](https://github.com/arendst/Sonoff-Tasmota/wiki/PowerOnState-Configuration) configuration?

tuga49, 2018-03-12:
Hi. I have solved one problem. The other one its bug. The bug is, for example. I have sonoff s20 turned on. If I remove it from the wall, it keeps on on ha. It happens with one my friend to, so its a bug. I hope it get solved soon

LucaAghemo, 2018-03-23:
I guess it cannot be resolved as a "bug", because if you unplug the S20 from the wall... he dies, therefore he cannot update his status. Maybe the mqtt broker or Home Assistant should "see" the thing and change status, but I don't know if this is possible.

issacg, 2018-03-23:
LWT topic takes care of this, of you set it up. You can tell home assistant to set "availability" of the device based on this.

tuga49, 2018-03-23:
@LucaAghemo, When I unplug my sonoff from the wall, it keeps showing he is on. Its should appear "unavailable or offline or so", but it just keeps the state of the sonoff. If he is on, it keep showing on, and then When I connect, it changes to PowerOnState that I have, in this case, I have 0, it change to off. but while sonoff its unplugged from the wall, it keep showing me the last known status. What should I do?

tuga49, 2018-03-23:
@issacg  how I set availability to home assistant know if my device is offline, and chage status to offline or unavaillable? thanks

issacg, 2018-03-23:
https://home-assistant.io/docs/mqtt/birth_will/

tuga49, 2018-03-23:
@issacg this page is not found. Thanks

issacg, 2018-03-25:
Sorry, mail client bungled it

https://www.home-assistant.io/docs/mqtt/birth_will

tuga49, 2018-03-25:
@issacg, thanks. Solved :)