arendst/Tasmota #2126: Dimmer for Sonoff or external switch device

djtonyd, 2018-03-10:
hi all is there anyone on here that has created a dimmer device on the sonoff's or can dim it via the app
i've currently got a sonoff basic in series with a standard wall dimmer and can also control the on/off feature via the phone and wall switch, but if the dimmer on the wall was to be removed what feature / device could go in its place so that the smartthings or alex can dim it?

Make sure these boxes are checked before submitting your issue - Thank you!

- [Sonoff basic ] What hardware you are using
- [0 ] Provide the output of command ``status 0``
- [ ] If you have a stack dump decode it: https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.rst
- [ ] For better debug messages: https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.rst

Frogmore42, 2018-03-10:
The Sonoffs all have relays so don't support dimming.  Smartthings talks to zwave and zigbee devices so that would be your best choice.

webtop, 2018-03-10:
@djtonyd Is the dimmer digital? What type/model is it?
Depending on that it MAY be possible to use the output from the WS2812 through GPIO14

djtonyd, 2018-03-10:
i did find something on here that someone tagged this: https://www.amazon.com/RobotDyn-controller-control-Arduino-Raspberry/dp/B072K9P7KH/ would this be ideal to use coming off  the sonoff's I/O, and the wall dimmer is just a normal clipsal leading edge  https://golights.global.ssl.fastly.net/media/catalog/product/cache/1/image/1200x1200/040ec09b1e35df139433887a97daa66f/l/4/l400e_4.jpg

webtop, 2018-03-11:
Those dimmers use Shunt Voltage Regulators as the basis for the voltage regulation. Those can be controlled via PWM. I don't have the time today to look into it, but theoritically you should be able to build something very simple and cheap to allow the sonoff to talk to that dimmer. So I do believe it's do-able.