arendst/Tasmota #2131: Naming explanation needed

roblad, 2018-03-11:
Hi,

I do not know what is a code of that, haw to ged name for mqtt and server name - what options are avaliable and from that name is generated.

Haw to decode:

DVES_%06X

%s-%04d

arendst, 2018-03-11:
See https://en.wikipedia.org/wiki/Printf_format_string

For DVES_%06X the value is provided by the last six hex digits (%06X) from the mac address.
For %s-%04d the values are the topic name (%s) and the last 4 decimal digits (%04d) of the mac address.

roblad, 2018-03-11:
Hi,

I am not able to calculate for that

wemos-0818 | 192.168.100.157 | 60:01:94:21:c3:32 | 
sonoff-4400 | 192.168.100.192 | 60:01:94:85:71:30 |

arendst, 2018-03-11:
As can be seen on line 2167 the mac address is masked with 0x1FFF.

So get your windows calculator on your desktop.

Set it to Programmer mode and enter in decimal 0818. Oh surprise it shows 332 in hex being the last three digits of the mac address.

Same for 4400 being 1130 hex and is 7130 masked with 0x1FFF.

ascillato, 2018-06-04:
Hi

If your issue was solved, please close it. Thanks :+1:

closed 2018-06-04 by roblad

JsBergbau, 2021-04-02:
@arendst 
Is there a list with all strings? Sorry didn't find and Google brought me here. Especially I'd like to configure MQTT to use the `Device Name` from `/co` page. 
[image]

BTW: Whats the difference between Device Name and Friendly Name 1?