arendst/Tasmota #2134: Unable to update Tasmota

mokshmridul, 2018-03-11:
Hi,

I have a few Sonoff Basic's that were flashed onto Tasmota via the Serial/FTDI route. However i flashed it using 1m SPIFFS (according to some YT channel).

While they work perfectly fine, i am unable to update either via FTDI or via OTA (using the web URL provided). They continue to work and show the last build time at Feb 2018 "Build Date & Time | 2018-02-09T14:54:51".

While flashing them using OTA, im first uploading the minimal OTA and then the full OTA. 

Could someone please help me resolve this, as i am unable to google a proper fix for this.

Output of ``status 0`` is given below:

```15:11:21 CMD: status 0
15:11:21 MQT: stat/sonoff1/STATUS = {"Status":{"Module":1,"FriendlyName":"Sonoff","Topic":"sonoff1","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":0,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":1}}
15:11:21 MQT: stat/sonoff1/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","Uptime":"1T15:29:14","Sleep":0,"BootCount":89,"SaveCount":417,"SaveAddress":"F4000"}}
15:11:21 MQT: stat/sonoff1/STATUS2 = {"StatusFWR":{"Version":"5.12.0","BuildDateTime":"2018-02-09T14:54:51","Boot":6,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
15:11:21 MQT: stat/sonoff1/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"domus1","LogPort":514,"SSId1":"Home24G","SSId2":"indebuurt2","TelePeriod":300,"SetOption":"00000029"}}
15:11:21 MQT: stat/sonoff1/STATUS4 = {"StatusMEM":{"ProgramSize":485,"Free":516,"Heap":22,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3}}
15:11:21 MQT: stat/sonoff1/STATUS5 = {"StatusNET":{"Hostname":"sonoff1-2605","IPAddress":"192.168.0.51","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.1","Mac":"DC:4F:22:09:6A:2D","Webserver":2,"WifiConfig":3}}
15:11:21 MQT: stat/sonoff1/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.12","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_096A2D","MqttUser":"homeassistant","MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
15:11:21 MQT: stat/sonoff1/STATUS7 = {"StatusTIM":{"UTC":"Sun Mar 11 10:11:21 2018","Local":"Sun Mar 11 15:11:21 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":5}}
15:11:21 MQT: stat/sonoff1/STATUS10 = {"StatusSNS":{"Time":"2018-03-11T15:11:21"}}
15:11:21 MQT: stat/sonoff1/STATUS11 = {"StatusSTS":{"Time":"2018-03-11T15:11:21","Uptime":"1T15:29:14","Vcc":3.228,"POWER":"ON","Wifi":{"AP":1,"SSId":"Home24G","RSSI":100,"APMac":"9C:5C:8E:BA:64:C0"}}}```

arendst, 2018-03-11:
Follow these steps to get some more information:
- Open the console web page
- Enter command ``weblog 3``
- Enter command ``upload 1`` and see what happens
- If the upload is done (or if you see other useful info) Copy all text from the console and provide it here for further analysis.

mokshmridul, 2018-03-11:
Hi,

Thank you for the quick response. The details you wanted are given below:

"weblog 3" results:

``22:59:01 CMD: weblog 3
22:59:01 MQT: stat/sonoff1/RESULT = {"WebLog":3}
22:59:02 CFG: Saved to flash at F9, Count 420, Bytes 1648``

"upload 1" results:

``23:00:13 CMD: upload 1
23:00:13 RSL: Group 0, Index 1, Command UPLOAD, Data 1
23:00:13 MQT: stat/sonoff1/RESULT = {"Upload":"Version 5.12.0 from http://sonoff.maddox.co.uk/tasmota/sonoff.bin"}
23:00:13 CFG: Saved to flash at FB, Count 421, Bytes 1648``

arendst, 2018-03-11:
So what happens after the upload command? Make sure to catch the screen for at least 1 minute.

mokshmridul, 2018-03-12:
If i run "upload 1", i get the following items in the console:

```07:41:09 CMD: upload 1
07:41:09 RSL: Group 0, Index 1, Command UPLOAD, Data 1
07:41:09 MQT: stat/sonoff1/RESULT = {"Upload":"Version 5.12.0 from http://sonoff.maddox.co.uk/tasmota/sonoff.bin"}
07:41:10 CFG: Saved to flash at FB, Count 425, Bytes 1648
07:41:12 HTP: Web server stopped
07:41:12 UPL: http://sonoff.maddox.co.uk/tasmota/sonoff.bin
07:41:35 HTP: Web server active on sonoff1-2605.local with IP address 192.168.0.51
07:41:36 MQT: Attempting connection...
07:41:36 MQT: Connected
07:41:36 MQT: tele/sonoff1/LWT = Online (retained)
07:41:36 MQT: cmnd/sonoff1/POWER = 
07:41:36 MQT: Subscribe to cmnd/sonoff1/#
07:41:36 MQT: Subscribe to cmnd/sonoffs/#
07:41:36 MQT: Subscribe to cmnd/DVES_096A2D/#
00:00:00 CFG: Loaded from flash at FB, Count 425
00:00:00 APP: Boot Count 92
00:00:00 Project sonoff Sonoff (Topic sonoff1, Fallback DVES_096A2D, GroupTopic sonoffs) Version 5.12.0
00:00:00 CFG: Saved to flash at FA, Count 426, Bytes 1648
00:00:00 WIF: Attempting connection...
00:00:00 WIF: Patch issue 2186
00:00:00 WIF: Connecting to AP1 Home24G in mode 11N as sonoff1-2605...
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Attempting connection...
00:00:03 WIF: Attempting connection...
00:00:04 WIF: Attempting connection...
00:00:05 WIF: Attempting connection...
00:00:06 WIF: Attempting connection...
00:00:07 WIF: Attempting connection...
00:00:08 WIF: Attempting connection...
00:00:09 WIF: Attempting connection...
00:00:10 WIF: Attempting connection...
00:00:11 WIF: Connected
00:00:11 DNS: Initialized
00:00:11 HTP: Web server active on sonoff1-2605.local with IP address 192.168.0.51
00:00:14 MQT: Attempting connection...
00:00:15 MQT: Connected
00:00:15 MQT: tele/sonoff1/LWT = Online (retained)
00:00:15 MQT: cmnd/sonoff1/POWER = 
00:00:15 MQT: Subscribe to cmnd/sonoff1/#
00:00:15 MQT: Subscribe to cmnd/sonoffs/#
00:00:15 MQT: Subscribe to cmnd/DVES_096A2D/#
00:00:15 APP: (UTC) Mon Mar 12 02:42:07 2018, (DST) Sun Mar 25 02:00:00 2018, (STD) Sun Oct 28 03:00:00 2018
07:42:08 MQT: tele/sonoff1/INFO1 = {"Module":"Sonoff Basic","Version":"5.12.0","FallbackTopic":"DVES_096A2D","GroupTopic":"sonoffs"}
07:42:08 MQT: tele/sonoff1/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff1-2605","IPAddress":"192.168.0.51"}
07:42:09 MQT: tele/sonoff1/INFO3 = {"RestartReason":"Software/System restart"}
07:42:09 MQT: homeassistant/switch/sonoff1/config =  (retained)
07:42:15 RSL: stat/sonoff1/RESULT = {"POWER":"ON"}
07:42:20 RSL: stat/sonoff1/POWER = ON
07:42:37 RSL: tele/sonoff1/STATE = {"Time":"2018-03-12T07:42:32","Uptime":"0T00:00:22","Vcc":3.226,"POWER":"ON","Wifi":{"AP":1,"SSId":"Home24G","RSSI":100,"APMac":"9C:5C:8E:BA:64:C0"}}
07:42:39 MQT: Attempting connection...
07:42:45 MQT: Connect failed to 192.168.0.12:1883, rc -2. Retry in 10 sec```

arendst, 2018-03-12:
Looks like a succesful upload to me.

mokshmridul, 2018-03-12:
It does look successful from the console, but the version number and build details don't change. I've attached a picture of the information tab below, which was taken after i tried the update commands.

[image]