arendst/Tasmota #2135: wish: direct control with Amazon-Dash-Button

2000ede, 2018-03-11:
Hello!
I would like it if Tasmota devices are to be switched DIRECTLY with Amazon Dash button.
See also Implementation in ioBroker Adapter Module (https://github.com/PArns/ioBroker.amazon-dash).
Thank you!

Hallo!
Ich würde es gut finden,wenn Tasmota Geräte DIREKT mit Amazon-Dash-Button zu schalten sind.
Siehe auch Umsetzung in ioBroker-Adapter Modul (https://github.com/PArns/ioBroker.amazon-dash).
Danke!

2000ede, 2018-03-12:
Hi!
Is it not possible to store the MAC address and the IP address of the Dash Button and to recognize switching via a permanent ping command?
(Sorry, I'm not a developer)

curzon01, 2018-03-12:
@2000ede Wouldn't it be better to handle that specific request by a third system like Home Assistant. AFAIK their already exist add-on to handle ADBs.

Be aware: A lot of request make sense in some envirinments but - and **this is the sticking point** - **the resources of the ESP 8266 build in Sonoff devices are very limited** and still on their limits. Theo target with Tasmota are the Sonoff devices, not the logic to control them in any possible variations - for this third systems like Home Assistant, OpemHab, FHEM are better designed and they are having more resources like an ESP8266 with the smallest Flash within a Sonoff device.

2000ede, 2018-03-12:
Many thanks for your response.
It was just an idea if it was easy to implement.
I was hoping that you would not have to run another system (eg Raspberry Pi with ioBroker).

TASMOTA is very good! Thank you for this great commitment!

Jason2866, 2018-03-12:
@2000ede  Just for interest... How do you control your devices with TASMOTA?
In OpenHab (for example) exists a binding for the Dash Buttons.
You can run mqtt broker, OpenHab NodeRed on a single device ->  RPi3

2000ede, 2018-03-12:
gladly. Currently about "ioBroker" on a Raspi3. That works fine too.
However, you always need a running Mini PC for this. This can again lead to error cases. Furthermore, I would like to provide a simple system for my brother. Raspi needs to be set up again, etc.
Therefore, a direct connection is much better.
But as written in the beginning: It is only a wish. Without it works :-)