arendst/Tasmota #2136: Smart Home Automation Module Itead 5V jog / self-locking Switch phone app Remote Control Timing Wifi Smart Remote Access

RobertHx, 2018-03-11:
Make sure these boxes are checked before submitting your issue - Thank you!

Configuration/Configure modul = 1 - Basic sonoff (it is not, but i did not find any other that work)
In this sonoff, if "Itching mode" is off (you have manually turn on and off) it works just fine.

Problem is, if you set PulseTime x <> 0, so turns on, and after x/10 s back off. It works fine. That is not problem.

Problem is, if you disconnect device from power and then connect again. Then, after initialise itself turns on. That means, if my house lose el.power, and then comes back, my garage door will open. 

This is partial status of this sonoff:
MQT: stat/g1/STATUS = {"Status":{"Module":1,"FriendlyName":"g1","Topic":"g1","ButtonTopic":"0","Power":0,"PowerOnState":0,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
MQT: stat/g1/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://domus1:80/api/arduino/sonoff.ino.bin","Uptime":"0T00:22:16","Sleep":0,"BootCount":77,"SaveCount":220,"SaveAddress":"F8000"}}
MQT: stat/g1/STATUS2 = {"StatusFWR":{"Version":"5.12.0","BuildDateTime":"2018-03-04T18:18:14","Boot":4,"Core":"2_4_0_RC2","SDK":"2.1.0(7106d38)"}}
MQT: stat/g1/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"domus1","LogPort":514,"SSId1":"HorVet","SSId2":"indebuurt2","TelePeriod":300,"SetOption":"00000009"}}
MQT: stat/g1/STATUS4 = {"StatusMEM":{"ProgramSize":518,"Free":484,"Heap":16,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3}}

- [ ] What hardware you are using
- [ ] Provide the output of command ``status 0``
- [ ] If you have a stack dump decode it: https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.rst
- [ ] For better debug messages: https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.rst

arendst, 2018-03-11:
Just did a test with your settings and it works as expected; 

- Pulsetime is set to 1 minute, device is turned on and relay is on.
- Within 20 seconds power is removed from device and relay is turned off (obviously)
- Power is re-applied and relay stays off.

How do you test device disconnecting from power?

Keep in mind that wemos devices probably won't re-act as you might expect after re-inserting the USB connecter due to their reset circuit.

RobertHx, 2018-03-11:
Yes, sonoff basic works perfect, this sonoff not. I need this device as
garage door opener. I do not need 220v voltage pass thru as basic does.
Can you find this device on aliexpress.com? I copy topic from there.

arendst, 2018-03-11:
?????

RobertHx, 2018-03-11:
[image]

arendst, 2018-03-11:
What happens when configured as module motor c/ac?

RobertHx, 2018-03-11:
Its even worse. After reconnect to usb relay is turned on for 3-4 sec., then turns off.
Even after choosing basic sonoff again this does not go away.

RobertHx, 2018-03-11:
I have reset this sonoff. Now again works almost normal. Only think I did, was PowerOnState 0.
That is if Pulsetime is set to 0.
Then at reconnect stays off.

If Pulsetime  x <> 0 then for a short time after reconnect to usb turns on. (And garage door would open again if power is lost and comes back).

Jason2866, 2018-03-13:
@RobertHx  My info helps you not really to solve your problem, but maybe wasting your time.
I tried different cheap china relays. Nearly all work fine (finally), if the work. 
2 of 3 didnt work out of the box. Horrible design errors (missing pullup resistors, Chip Enable unconnected...).
One of this board, is switching the relay on startup ON and OFF. There is no reason for...
All other work as the should!
This one works right out of the box with no modification! an has a optocupler input too: 
https://www.aliexpress.com/item/ESP8266-WIFI-relay-network-relays/32741837053.html?spm=a2g0s.9042311.0.0.CpeR0I

RobertHx, 2018-03-13:
Progress!!!
But first: i have 4 such sonoffs. 2 are original itead software, 2 tasmota. Originals never turn relay on after reconnect to power (usb). So it has to be tasmota or HA.

In HomeAssistant (HA) i set retain to false and restarted HA.

Modul Type I changed to now 12 1 Channel.
Reset tasmota.

Setup tasmota 2 with different topic. It works! even after power reconnect.
Then backup configuration, and restore to tasmota 1. Tasmota 1 works now!
Changed tasmota 1 topic to old name. Again do not work. 

After that I renamed tasmota 1 and 2 to new Topic names. And all works now.

I did not change nothing else in HA but topic to match tasmota.

So I think something is wrong with HA. Not 100% but close.

Jason2866, 2018-03-13:
I saw many post with problems with Home Assistant...
In my humble opinion Home Assistant isnt a good choice!

RobertHx, 2018-03-13:
What is better?

Jason2866, 2018-03-13:
I am statisfied with OpenHab. It runs very well on a RasPi 3.
You can get a SD Card Image. Easy to install...
BUT configuration isnt done on a weekend, many possibilitys and a lot of things to understand.
If you got the Basic it is really powerful and nearly everything can be implemented AND it works

RobertHx, 2018-03-13:
Has multiple user accounts? Admin and users?

Jason2866, 2018-03-13:
No multiple user accounts. You hit the the main point what is missing in OpenHab.
You can define different panels for different users but thats not really multi user...
So as everthing in life, there is not the one and only thing thats fits for all!

RobertHx, 2018-03-13:
Anyhow tasmota is still not ideal. If you look at picture of this tasmota, lower switch has to be on, otherwise tasmota will not function.
If this switch is off (thats purpose why you would buy it with original itead software) it will not work. In that case tasmota hang up.

RobertHx, 2018-03-13:
Multiuser comes in future?

until then, different panels is not bad solution.

Jason2866, 2018-03-13:
Maybe... The basic design has to be changed. So not in near future. The main contributers (no paid people)
have it on there list, but other things needs to be done first.
Have a look at   www.openhab.org

RobertHx, 2018-03-13:
I will. But you said it is not done in weekend, it has to be hard to setup... :(

asantos15, 2018-03-22:
@RobertHx , I saw this one somewhere a while ago and it's been solving my issue with the 5V sonoff relay:
This is the command that publishes nothing and overrides the state that is causing the board to change state:
To check if there is any state pre selected: mosquitto_sub -p 1883 -u <username> -P <password> -t cmnd/+/power changing username and password only. 
Then...
mosquitto_pub -h IP -p 1883 -r -t 'cmnd/sonoff/POWER' -n -u username -P password