arendst/Tasmota #2137: DHT11 Sensor for Sonoff Touch and T2

Fabian2009, 2018-03-11:
Hello,

I am using several Sonoff Touch's and one T2 where I would like to add a DHT11 sensor.
Is there any picture how to connector the sensor ?

According to the webinterface, I need to connector it to GPIO3 for the Touch version, but I only have pictures showing GPIO0 for flashing.

thank you very much for your support,
best regards,
FAbian

curzon01, 2018-03-11:
How did you flashed your Touch with Tasmota? If you used an USB connector then you already used RX/TX line. These are the same as GPIO1/GPIO3 and so your are already know the pins for GPIO1 and GPIO3.
On webinterface you can configure both GPIOs for DHT11 sensor. Choose one and connect the DHT11 to it, that's it.

Jason2866, 2018-03-11:
Just to be complete, disable serial logging, if you use GPIO 1 and GPIO 3.
This can be done by command or in the Web view