arendst/Tasmota #2141: DS18X20 not reading with sonoff SV

lordgadgetman, 2018-03-12:
Hi All,

I am running a sonoff SV with Tasmota 5.12.0 but cant get it to read the temp on any of the IO ports.

I see the board sending MQTT readings which are 0c.

I only have one temp sensor connected to the board to IO14 and I have also tried configuring IO4 with the same results. 

I am using DS18S20 rather than the B but dont believe this should make a difference, plkease correct me if I am wrong.

01:59:34 MQT: house/Garage_door/SENSOR = {"Time":"2018-03-12T01:59:34","DS18B20":{"Temperature":0.0},"TempUnit":"C"}

The temperatire also appears on the web admin for thios device but again shows 0c

Any ideas on the issue?

Thanks

Frogmore42, 2018-03-12:
0 degrees C is what you get when the data line is either shorted to ground (or you did not put a pull up on it).  So, how do you have it connected?  Does your DS18S20 have 2 pins or 3 and what did you connect them to?  I am pretty sure you have to use a pull up resistor (4K7 is what most people use) from 3V3 to the data line.  You also have to hook up 3V3 and gnd to the other two pins on the DS18S20.

lordgadgetman, 2018-03-12:
It’s a 3 pin and your right I haven’t used a pull up resistor. So if I am reading this right the data line that is the the Dq pin needs to be connected to signal but also needs a pull up 4.7k between Vcc at the Dq pin as well. 

Have I read that right?

The other pins are already connected to 3.3 and gnd as required.

curzon01, 2018-03-12:
All DS18S20 having 3 pins, but they can be used in different powering mode (Normal and Parasite) and therefor sometimes is looks like you use two pins instead 3. But using 2 pins only, the third pin Vcc has to be connected to GND:

- **Normal Power Mode**: is using all three pins Vcc, GND and Data. Data is Pullup to Vcc - see [Normale Power Mode picture here](https://www.tweaking4all.com/wp-content/uploads/2014/03/ds18b20-normal-power-400x184.jpg)
- **Parasite Power Mode**: is using two pins GND and Data Pullup to Vcc. Vcc must be set to GND in that case - see [Parasite Power Mode picture here](https://www.tweaking4all.com/wp-content/uploads/2014/03/ds18b20-parasite-power-400x181.jpg)

Frogmore42, 2018-03-12:
Some manufactured sensors only include 2 wires.  Using parasite power is tricky.  The spec says you must use a strong drive on the data line, not just a pull-up, during the measurement time.  It might work with just a pull-up, but it is likely to not be reliable.