arendst/Tasmota #2142: More of a query than a issue with the iTead Sonoff RF Bridge 433

TristanMW, 2018-03-12:
Morning,
I am using a variety sonoff switches around my house which I control with node-red.
I am looking at getting the sonoff door and window sensors installed as well for extra security which connect to the iTead Sonoff RF Bridge 433 using the eLink app.

My question is, would it be possible to flash the iTead Sonoff RF Bridge 433 with the Tasmota software and still be able to pair the door and window switch to it?

Thanks for the help

Jason2866, 2018-03-12:
What do you mean with pairing? There is no logic left in the bridge for RF 433Mhz devices after flashing Tasmota. You will only get a mqtt message if a sensor "fires". Every magic has to be done in the HA.

TristanMW, 2018-03-12:
By pairing I mean that the sensor needs to be connected bridge in order to be added as a device on the official app. 

If I receive a mqtt message from the bridge every time the sensor activates then that would be perfect.
I normally use the <mqtt out> node to send commands to the switches but I have noticed a <mqtt in> node which probably would work

arendst, 2018-03-12:
The bridge should still be able to learn the code from DW1 and send a message to MQTT. See tasmota commands with regards to the bridge in the wiki.

TristanMW, 2018-03-12:
Perfect thanks for the help

TristanMW, 2018-03-12:
If anyone in the future is interested in the results, ill post them.

closed 2018-03-12 by TristanMW

nuinoa, 2018-03-12:
Hi TristanMW,
I am interested in the sonoff RF bridge MQTT integration with openHAB App.  Is this someting you are working on.   I would like to knowhow to add Window Door contact in my OpenHAB items and how to extract Rfkey status to initial an action like turn on a sonoff basic switch.  Thanks.