arendst/Tasmota #2143: Streange bechaviour of Sonoff CH4 Pro when WiFi is lost for longer time

roblad, 2018-03-12:
Hi,

Sonoff Ch4 Pro, has strange behavior after update to v5.12.0c. When WiFi router is lost for more than 3-5 min the CH4 Pro, hangs and switch on Relay 1 (A) and LED of Relay 1 (A) is switched on. Only restart can solve the problem. When it happened there is no communication with the Sonoff, I do not have Debug and Serial log, in Web log there is only connection retrying and after a few min hangs with the relay and LED powering on.  What could be a problem for that , what I need to review to find the reason ?

Regards,
RL

Jason2866, 2018-03-12:
Which version of Iwip do you use in your build? If Version 2.0 change to 1.4 prebuilt. V2.0 is buggy.
How much free memory do you have? Low memory is a problem. >20k would be good.

roblad, 2018-03-12:
Hi,

What is a  "Iwip",  my amount is 23 kb 

1024kB
479kB
524kB
23kB

"Version 2.0 change to 1.4 prebuilt. V2.0 is buggy." what version do you talking about ? What module,lib etc , what is that ?

Jason2866, 2018-03-12:
Look here #2042

roblad, 2018-03-12:
Hi,
I have c version and I add that checking 

overrides for all previous includes
+#include "lwip/init.h"
 
 
 // Libraries
 // Libraries
 #include <PubSubClient.h>                   // MQTT
 #include <PubSubClient.h>                   // MQTT
@@ -44,6 +45,11 @@
   #error "MQTT_MAX_PACKET_SIZE is too small in libraries/PubSubClient/src/PubSubClient.h, increase it to at least 1000"
   #error "MQTT_MAX_PACKET_SIZE is too small in libraries/PubSubClient/src/PubSubClient.h, increase it to at least 1000"
 #endif
 #endif
 
 
+// Check if the lwIP version is the stable one
+#if LWIP_VERSION_MAJOR != 1
+  #error Please use lwIP v1.4
+#endif
+
and it did not help after 30 sec without wifi sonoff 4ch pro hangs

Jason2866, 2018-03-12:
Hmm, no idea...
Do you have a specific reason for updating the software?
Try your previous build. If it works why not stay at this software?

roblad, 2018-03-12:
Hi,

There is different ESP on that Sonoff 4ch Pro 

https://www.itead.cc/wiki/Sonoff_4CH_Pro

Core library for Tasmota version c for ESP is 2.4.0 for d is 2.4.1

Jason2866, 2018-03-13:
Maybe the new core is the problem... 
For all my builds (5.12.0d too) i still use library 2.3.0 (1.5.3).
This builds are running without problems for me.
For Atom.io this setting is in platformio.ini

roblad, 2018-03-13:
Hi,

Haw you set  2.3.0 version in PlatformIO ?

Jason2866, 2018-03-13:
Snipplet of platformio.ini

[env:sonoff]
platform = espressif8266@1.5.0  ; v2.3.0
;platform = espressif8266@1.6.0  ; v2.4.0
platform = espressif8266
framework = arduino
board = esp01_1m
board_flash_mode = dout
build_flags = -Wl,-Tesp8266.flash.1m0.ld
lib_deps = PubSubClient, NeoPixelBus, IRremoteESP8266, ArduinoJSON
extra_scripts = pio/strip-floats.py

roblad, 2018-03-15:
Hi,

I recompiled with platform = espressif8266@1.5.0 ; v2.3.0  and it did not solved that problem

Jason2866, 2018-03-15:
Hi, ideas running out...
My last idea is trying the prebuild image from the master branch. 
If the problem is still there???

roblad, 2018-03-16:
Hi,

haw to disable mDNS ?

What does that mean "prebuild image from the master branch" ?

Jason2866, 2018-03-17:
https://github.com/arendst/Sonoff-Tasmota/releases

roblad, 2018-03-31:
Hi,

problem disapeared after upgrade to version "i"

closed 2018-03-31 by roblad