arendst/Tasmota #2145: Which platformio.ini settings to use for WeMOS?

RaymondMouthaan, 2018-03-12:
Currently I am updating / maintaining the [Tasmota Wiki page for Wemos](https://github.com/arendst/Sonoff-Tasmota/wiki/Wemos%20D1%20Mini) where I like to add the correct settings for compiling and uploading Tasmota firmware to Wemos devices (D1 Mini and D1 Mini Pro). 

Below the platformio.ini settings I use:
```
[env:wemos-d1-mini]
platform = espressif8266
framework = arduino
board = d1_mini
board_flash_mode = dout
build_flags = -Wl,-Tesp8266.flash.1m0.ld
lib_deps = PubSubClient, NeoPixelBus, IRremoteESP8266, ArduinoJSON

; *** Serial Monitor options
monitor_baud = 115200
```

Are these the correct settings?

arendst, 2018-03-12:
To remove any possible problems with regard to flash size and flash mode I only support the settings as provided with the released and development version.

So a wemos needs to be treated the same as a sonoff in platformio.ini.

```
[env:sonoff]
;platform = espressif8266@1.5.0  ; v2.3.0
;platform = espressif8266@1.6.0  ; v2.4.0
platform = espressif8266
framework = arduino
board = esp01_1m
board_flash_mode = dout
build_flags = -Wl,-Tesp8266.flash.1m0.ld -DMQTT_MAX_PACKET_SIZE=1000
lib_deps = PubSubClient, NeoPixelBus, IRremoteESP8266, ArduinoJSON
extra_scripts = pio/strip-floats.py

; *** Serial Monitor options
monitor_baud = 115200

; *** Upload Serial reset method for Wemos and NodeMCU
;upload_resetmethod = nodemcu
;upload_port = COM6
;upload_speed = 115200

; *** Upload file to OTA server using SCP
;upload_port = user@host:/path
;extra_scripts = pio/strip-floats.py, pio/sftp-uploader.py

; *** Upload file to OTA server using HTTP
;upload_port = domus1:80/api/upload-arduino.php
;extra_scripts = pio/strip-floats.py, pio/http-uploader.py
```

Enabling the serial upload_resetmethod as nodemcu makes flashing wemos very easy.

RaymondMouthaan, 2018-03-12:
@arendst, I just flashed a brand new Wemos D1 Mini Pro v1.0.0 with the settings you recommend and it works :) 

To be exactly:
```
[env:wemos-d1-mini]
platform = espressif8266
framework = arduino
board = esp01_1m
board_flash_mode = dout
build_flags = -Wl,-Tesp8266.flash.1m0.ld -DMQTT_MAX_PACKET_SIZE=1000
lib_deps = PubSubClient, NeoPixelBus, IRremoteESP8266, ArduinoJSON
extra_scripts = pio/strip-floats.py

; *** Serial Monitor options
monitor_baud = 115200

; *** Upload Serial reset method for Wemos and NodeMCU
upload_resetmethod = nodemcu
upload_speed = 115200
;upload_port = COM6

; *** Upload file to OTA server using SCP
;upload_port = user@host:/path
;extra_scripts = pio/strip-floats.py, pio/sftp-uploader.py

; *** Upload file to OTA server using HTTP
;upload_port = domus1:80/api/upload-arduino.php
;extra_scripts = pio/strip-floats.py, pio/http-uploader.py
```

Ill add these to the wiki page.

Thanks!

RaymondMouthaan, 2018-03-12:
[Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki/WeMOS-D1-Mini) updated accordingly

closed 2018-03-12 by RaymondMouthaan

ngalfas, 2018-11-25:
i think the wiki needs some update with the current Platformio version

ie. 
Warning! `monitor_baud` option is deprecated and will be removed in the next release! Please use `monitor_speed` instead.
Warning! `board_flash_mode` option is deprecated and will be removed in the next release! Please use `board_build.flash_mode` instead.

RaymondMouthaan, 2018-11-25:
You can update the wiki accordingly 👍🏽

vel2000, 2019-02-22:
When i use the above settings in "platformio.ini", I get about 80 problems/warnings while compiling in Visual Studio and platformio.
I use development source "Sonoff-Tasmota 6.4.1.18"
**[Full log is here](https://paste.ubuntu.com/p/Q88VVcxwGv/)**
Example messages:             
> Compiling .pioenvs\wemos-d1-mini\libbc2\TasmotaMqtt-1.1.1\mqtt\mqtt.c.o
In file included from lib\TasmotaMqtt-1.1.1\src\mqtt\mqtt.c:37:0:
lib\TasmotaMqtt-1.1.1\src\mqtt\mqtt_msg.h:48:1: warning: "/*" within comment [-Wcomment]
/*                    Remaining Length                 */
^
lib\TasmotaMqtt-1.1.1\src\mqtt\mqtt.c: In function 'mqtt_tcpclient_recv':
lib\TasmotaMqtt-1.1.1\src\mqtt\mqtt.c:311:5: warning: enumeration value 'WIFI_INIT' not handled in switch [-Wswitch]
switch (client->connState) {
^
lib\TasmotaMqtt-1.1.1\src\mqtt\mqtt.c:311:5: warning: enumeration value 'WIFI_CONNECTING' not handled in switch [-Wswitch]
lib\TasmotaMqtt-1.1.1\src\mqtt\mqtt.c:311:5: warning: enumeration value 'WIFI_CONNECTING_ERROR' not handled in switch [-Wswitch]
lib\TasmotaMqtt-1.1.1\src\mqtt\mqtt.c:311:5: warning: enumeration value 'WIFI_CONNECTED' not handled in switch [-Wswitch]
lib\TasmotaMqtt-1.1.1\src\mqtt\mqtt.c:311:5: warning: enumeration value 'DNS_RESOLVE' not handled in switch [-Wswitch]
lib\TasmotaMqtt-1.1.1\src\mqtt\mqtt.c:311:5: warning: enumeration value 'TCP_DISCONNECTING' not handled in switch [-Wswitch]
lib\TasmotaMqtt-1.1.1\src\mqtt\mqtt.c:311:5: warning: enumeration value 'TCP_DISCONNECTED' not handled in switch [-Wswitch]
lib\TasmotaMqtt-1.1.1\src\mqtt\mqtt.c:311:5: warning: enumeration value 'TCP_RECONNECT_DISCONNECTING' not handled in switch [-Wswitch]
lib\TasmotaMqtt-1.1.1\src\mqtt\mqtt.c:311:5: warning: enumeration value 'TCP_RECONNECT_REQ' not handled in switch [-Wswitch]
lib\TasmotaMqtt-1.1.1\src\mqtt\mqtt.c:311:5: warning: enumeration value 'TCP_RECONNECT' not handled in switch [-Wswitch]
lib\TasmotaMqtt-1.1.1\src\mqtt\mqtt.c:311:5: warning: enumeration value 'TCP_CONNECTING' not handled in switch [-Wswitch]
lib\TasmotaMqtt-1.1.1\src\mqtt\mqtt.c:311:5: warning: enumeration value 'TCP_CONNECTING_ERROR' not handled in switch [-Wswitch]
lib\TasmotaMqtt-1.1.1\src\mqtt\mqtt.c:311:5: warning: enumeration value 'TCP_CONNECTED' not handled in switch [-Wswitch]
lib\TasmotaMqtt-1.1.1\src\mqtt\mqtt.c:311:5: warning: enumeration value 'MQTT_CONNECT_SEND' not handled in switch [-Wswitch]
lib\TasmotaMqtt-1.1.1\src\mqtt\mqtt.c:311:5: warning: enumeration value 'MQTT_SUBSCIBE_SEND' not handled in switch [-Wswitch]
lib\TasmotaMqtt-1.1.1\src\mqtt\mqtt.c:311:5: warning: enumeration value 'MQTT_SUBSCIBE_SENDING' not handled in switch [-Wswitch]
lib\TasmotaMqtt-1.1.1\src\mqtt\mqtt.c:311:5: warning: enumeration value 'MQTT_PUBLISH_RECV' not handled in switch [-Wswitch]
lib\TasmotaMqtt-1.1.1\src\mqtt\mqtt.c:311:5: warning: enumeration value 'MQTT_PUBLISHING' not handled in switch [-Wswitch]
lib\TasmotaMqtt-1.1.1\src\mqtt\mqtt.c:311:5: warning: enumeration value 'MQTT_DELETING' not handled in switch [-Wswitch]
lib\TasmotaMqtt-1.1.1\src\mqtt\mqtt.c:311:5: warning: enumeration value 'MQTT_DELETED' not handled in switch [-Wswitch]
lib\TasmotaMqtt-1.1.1\src\mqtt\mqtt.c: In function 'MQTT_Task':
lib\TasmotaMqtt-1.1.1\src\mqtt\mqtt.c:704:3: warning: enumeration value 'WIFI_INIT' not handled in switch [-Wswitch]
switch (client->connState) {
^

ngalfas, 2019-02-24:
just leave the standard settings...  there is no need to try and use the 4MB memory on the wemos

vel2000, 2019-02-24:
The same error messages also appear with standard settings.
Completely independent of whether 1MB or 4MB is used in `platformio.ini`.

erichegit, 2019-03-03:
If I would get warnings I would be happy.
I get an Error
```
Linking .pioenvs\sonoff-DE\firmware.elf
Building .pioenvs\sonoff-DE\firmware.bin
Retrieving maximum program size .pioenvs\sonoff-DE\firmware.elf
Checking size .pioenvs\sonoff-DE\firmware.elf
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [======    ]  59.8% (used 48956 bytes from 81920 bytes)
PROGRAM: [=====     ]  52.1% (used 533712 bytes from 1023984 bytes)
Configuring upload protocol...
Looking for upload port...
Use manually specified: /dev/cu.wchusbserial1460
Uploading .pioenvs\sonoff-DE\firmware.bin
*** [upload] Error -1
error: Failed to open /dev/cu.wchusbserial1460
error: espcomm_open failed
error: espcomm_upload_mem failed

```