arendst/Tasmota #2146: Tasmota and Google Home?

OxygenGS, 2018-03-12:
Ok peeps i am about to flash my sonoff's to Tasmota. But, will it after that still work the same way with my google home? So i can still say: Hey google, turn on the light (specific room)?

arendst, 2018-03-12:
Nope. You'll need a home automation tool for that. Happy flashing.

OxygenGS, 2018-03-12:
home automation too? Like what exactly? Because i wanna keep using it like i am using now that i can control it with my voice.

OxygenGS, 2018-03-12:
Would it run on a Asus Tinker Board??

OxygenGS, 2018-03-12:
or a banana board?

Jason2866, 2018-03-12:
If you want to be save. Take a RasPi 3. You will get prebuilt images from nearly all know HA sytems for it.
So this way no problems to setup the hardware for a working system out of the box...
You will have enough fun setting the HA sytem to your wishes😎

RyanEwen, 2018-05-15:
To add some clarity for others who come across this..

If you remove the stock firmware, which uses eWeLink cloud service to talk to the device, you will lose the ability to control your Sonoff's using the eWeLink app. You will also lose the ability for Google Assistant to control the Sonoff's since in a stock setup, it goes through eWeLink to do that.

However, if you flash Tasmota f/w and configure your Sonoff's to connect to an MQTT server, and you also connect some home automation software (such as openHAB) to the same MQTT server, then the home automation software can be used to control the Sonoff's. Then you can setup Google to go through your home automation software to control them as well. For example:
- Sonoff's with Tasmota <-> local MQTT server
- local openHAB server <-> local MQTT server
- local openHAB server <-> myopenhab.org cloud add-on
- Google Assistant <-> myopenhab.org <-> local openHAB server <-> local MQTT server <-> Sonoff's with Tasmota

You're basically replacing the eWeLink cloud service with a locally-hosted openHAB (or HomeAssistant, or whatever) service. And Google can talk to that to control your Sonoff's.

There are also ways to set it up so that you wouldn't need to run any home automation software at home. Something like this:
Google Assistant <-> IFTTT Webhooks -> Sonoff's with Tasmota (via port forwarding rules setup in your router to let IFTTT in).

ascillato, 2018-06-04:
Hi

If your issue was solved, please close it. Thanks :+1:

closed 2018-06-13 by ascillato2

ascillato, 2019-01-03:
Thanks. Link added to the wiki :+1:

TiibCD, 2020-03-17:
FYI, "The gBridge.io service will be shut down on 15th March 2020"

guipsfr, 2020-04-05:
Someone have any alternative solution to gBridge ?

badziomiki, 2020-04-27:
+1 - anyone aware of any alternative solution to gBridge?

ruhlandm1, 2020-11-21:
Hello Together - I´ve found an alternative  solution even though it is relatively expensive and therefore surely not for everyone. 

I anyhow used a light manager air from jbmedia to controll different RF and IF devices. The lightmanager is also capable of controlling network devices by sending http-push and http-get commands into your network. Since a few week weeks this thing is also controllable through google home. 

So I have created devices in the light manager, have teached them the network commands for the tasmota devices. After connecting the jb media light manager and google home the devices appear in my google home dashboard. 

That works fine, it´s just not super fast (takes from 1-3 seconds to respond). So for everyone having such a device at home this could be a solution.