arendst/Tasmota #2147: Problem saving settings with backlog

issacg, 2018-03-12:
I'm testing 5.12.0e-2_4_0, and running a configuration script via Backlog, but the changes don't seem to be persisting.

Sending the command `Backlog MqttHost 192.168.13.4; MqttPort 8883; MqttUser 0; MqttPassword 0; OtaUrl http://192.168.13.4:80/ota/firmware.bin` gives me the following in the serial log

```
15:44:49 MQT: stat/DVES_116CF9/RESULT = {"Backlog":"Appended"}
15:44:49 RSL: stat/sonoff/RESULT = {"MqttHost":"192.168.13.4"}
15:44:49 RSL: stat/sonoff/RESULT = {"MqttPort":8883}
15:44:49 RSL: stat/sonoff/RESULT = {"MqttUser":""}
15:44:50 RSL: stat/sonoff/RESULT = {"MqttPassword":""}
15:44:50 MQT: Attempting connection...
15:44:55 MQT: Connect failed to 192.168.13.4:8883, rc -2. Retry in 10 sec
15:44:55 RSL: stat/sonoff/RESULT = {"OtaUrl":"http://192.168.13.4:80/ota/firmware.bin"}
15:44:55 RSL: stat/sonoff/RESULT = {"POWER":"ON"}
15:44:55 RSL: stat/sonoff/POWER = ON
15:44:56 APP: Restarting

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v4ceabea9
~ld

00:00:00 CFG: Use defaults
00:00:00 Project sonoff Sonoff (Topic sonoff, Fallback DVES_116CF9, GroupTopic sonoffs) Version 5.12.0e-2_4_0
...
```
And then the device proceeds to connect with the old MQTT settings (which are in user_settings.h)

I'm unsure about the bootloading output if it's supposed to look like that or not.  I'm flashing a nodemcu for development, but can try on a sonoff if there's significant chance that it will work differently.

arendst, 2018-03-12:
Well the important message is:
```
Use defaults
```
Resetting all you just entered....

Reconsider your user_config.h. Reboot first before doing your backlog and monitor the info scrolling along. At least the Use defaults should not be there.

issacg, 2018-03-12:
I saw that, but I don't understand why I'm getting it.Basically I'm connecting to an mqtt server which pushes the information of a second "internal" mqtt server. What ends up happening is I get into an endless loop where I boot, connect to the first server, it pushes the information for the internal server, reboots the device and then, since it loads the factory defaults, goes around and around. It's not actually doing anything else, which is why I'm puzzled about what would cause it to load the default config after the reboot...