arendst/Tasmota #2149: No connection via IP / MQTT reports connected/not connected

JueBag, 2018-03-12:
I flashed several Sonoff switches with tasmota(5.12.0), on some of them I see this weird behaviour:

The switch is reported to be connected to the WLAN (Fritzbox menu)
While trying to connect to the IP I get no response.
Checking the MQTT I see lots of the following reports

```
1520872882: New client connected from 192.168.178.57 as DVES_10DE89 (c1, k15, u$
1520872903: Client DVES_10DE89 has exceeded timeout, disconnecting.
1520872903: Socket error on client DVES_10DE89, disconnecting.
1520872908: New connection from 192.168.178.57 on port 1883.
1520872908: New client connected from 192.168.178.57 as DVES_10DE89 (c1, k15, u$
1520872929: Client DVES_10DE89 has exceeded timeout, disconnecting.
1520872929: Socket error on client DVES_10DE89, disconnecting.
1520872958: New connection from 192.168.178.57 on port 1883.
1520872958: New client connected from 192.168.178.57 as DVES_10DE89 (c1, k15, u$
1520872980: Client DVES_10DE89 has exceeded timeout, disconnecting.
1520872980: Socket error on client DVES_10DE89, disconnecting.

```
Does this point to a hardware failure??

arendst, 2018-03-12:
Verify unique clientids

JueBag, 2018-03-12:
I use them with numbered IDs (the one in question is: "Sonoff-4"), so I confirm they all have distinct IDs.

codenamej6, 2018-03-12:
I have the same problem.  Some work, but others get that same timeout error or try to reconnect even though they're already connected.  Made sure that all of the client ID's were unique.  I did find that after a few days of letting them run, some of them sorted themselves out on their own and stay connected now, but it bothers me to no end not knowing why they sometimes work and sometimes don't.

Barnabybear, 2018-03-14:
Hi, I had a similar problem that caused me to pull my hair out for a couple of days but now fixed.
10 Sonoff Basics all flashed with Sonoff-Tasmota -5.12.0 (Arduino-IDE-1.8.5).
All connect, stay connected and function as expected when using my BT Home Hub 4 router as the Wi-Fi connection point.
BUT:
All 10 continuously report connecting, disconnecting and are generally unresponsive to MQTT and HTTP requests when the Wi-Fi connection point is an Ubiquiti UAP-AC-PRO (Access point).

The fix for me:
In ‘user_config.h’ in the section marked ‘mDNS’
Comment out the line ‘#define USE_DESCOVERY’

If using MQTT ensure that the following are set:
‘#define MQTT_HOST’ needs to be the IP address not the name
‘#define MQTT_HOST’ needs to be the port number

All 10 have been up for 2 days now without any issues. Hope that helps.
To the devs – Thanks for the great work.

codenamej6, 2018-03-14:
Barnabybear, thank you wholeheartedly.  I commented out #define USE_DISCOVERY and it solved all of my timeout and reconnect issues.  Such a simple fix to an infuriating problem.

JueBag, 2018-03-14:
Thanks @Barnabybear , that tip worked for me as well.
Closing this one!

Barnabybear, 2018-03-14:
Hi, thanks for posting back and good to know this worked for you. If anyone can offer some insight I would still be interested in knowing:
1.	Given that everything was fine when the Sonoff Wi-Fi connected to my router and the problems only occurred when I connected to the same router via a Wi-Fi access point. What did the access point block, delay or disrupt to cause the need for re-connections?
2.	Why does commenting out #define USE_DISCOVERY fix this? 
Thanks.

codenamej6, 2018-03-14:
In my case, it was a bit different, I was getting connection errors while connected to a router and not an access point.  Might be a router/access point brand issue.

closed 2018-03-16 by JueBag