arendst/Tasmota #2151: Help please with flashing error

OxygenGS, 2018-03-12:
Ok i am trying to flash my S20.
Now i did first get an error saying that sayd:
MQTT_MAX_PACKET_SIZE is too small in libraries/PubSubClient/src/PubSubClient.h, increase it to at least 1000

So i did change it 1000, however, now i get this error shown on the screenshot below.
http://prntscr.com/iqco3a

Please help!?

Jason2866, 2018-03-13:
Your general Installation seems not to be ok. Error states that a lib file could not be found.
Did you follow exactly the wiki? See here:
https://github.com/arendst/Sonoff-Tasmota/wiki/Arduino-IDE
Your change for setting setup is fine -> newer version

If you get Arduino IDE not running try ATOM.IO it is a nice IDE and easy to setup. look here:
https://github.com/arendst/Sonoff-Tasmota/wiki/PlatformIO

curzon01, 2018-03-13:
@OxygenGS - why do not using the ready compiled binary under [Releases](https://github.com/arendst/Sonoff-Tasmota/releases)?
I'm asking because some newbie is fighting with the compiling something which is not neccessary at all. Start with ready binary, then you can try your own...