arendst/Tasmota #2152: Help: 2 buttons, 1 relay

rokrodic, 2018-03-13:
Hi. I use Sonoff basic. I have 2 external switches (on/off = +3V3/gnd). One is connected to gpio0, other to gpio14. I have a light on the relay. I set the module:
"gpio14 9" (= switch)
"topic son001"
"switchtopic son002"
"switchmode1 1"
"switchmode2 1"
When I switch the gpio14, I get ON/OFF on mqtt (cmnd/son002/POWER1) - everything OK. The problem is the gpio0. If I press the little button on the sonoff itself (pushbutton), the light toggles. But if I turn on or off the external switch nothing happens. If I simulate the pushbutton with my switch (turn on and suddenly off), the light toggles. This happens when it is freshly setup. But if I leave the switch ON, it seams the tasmota thinks I am pushing the button for a long time and it resets and restarts - all above settings are erased.
My plan is: when either switch is changed, tasmota sends mqtt message and it turns relay according to the mqtt message it receives. No direct internal relay changes.
How do I set the tasmota to correctly use the external switch on gpio0? (I will not be using the sonoff's pushbutton)
Tnx.

Frogmore42, 2018-03-13:
Using GPIO 0 as an input, especially connected to a switch that is applying gnd to the pin (and it sounds like that is how you have it wired) is a really bad idea.  Your best bet is to use a different GPIO.  GPIO 4 is generally easy to find and a much better choice for your task.

How do you have the switch wired up?   GPIO 0 has an internal pull-up, which means for your switch to have any impact it needs to connect GPIO 0 to ground.

The reason this is a bad idea is if the esp8266 resets for any reason when the switch is closed it will enter bootloader mode and not run your firmware.

rokrodic, 2018-03-13:
Hi. Smart notice about gpio0. I forgot about its function at startup. Sadly the gpio4 is so small on the chip.  It seems only gpio1,3,4,14 are available intasmota. Sadly no gpio13 (led). Using led pin would be easier. What do you think about using rx/tx pins?

Jason2866, 2018-03-13:
Use rx/tx that will work. You just have to turn of serial logging

closed 2018-03-13 by rokrodic