arendst/Tasmota #2153: Using Tasmota as remote turn off and turn on again without MQTT

binderth, 2018-03-13:
I use some Sonoff Pow and Switches.

My use case is I'd like to use the Tasmota on a Sonoff to turn off the power and turn it on after some time. But - and that's the tricky part: it should turn on/off the router, meaning after turning the power off, there's no more chance to send another MQTT command to turn it on. Is there a way to achieve this?

Jason2866, 2018-03-13:
Not yet tried. I think it should be possible with command pulsetime
Take a look in the wiki -> commands

binderth, 2018-03-13:
Unfortunately pulsetime only works with ON and it'll shut OFF after specified time... Would be great you if it worked both ways!
So you can not only use it as staircase circuit, but also to shut something down after a specified time! Would come in handy not only in my use case I bet!

whyvas, 2018-03-14:
I was also thinking about a solution for an unreliable modem and router using tasmota. I think it would be even better to leave this tasmota in AP mode so you can connect to it directly in case your wifi goes down completely. This would remove the need to have it power cycle automatically and you can use the existing firmware.

binderth, 2018-03-14:
hmmm... if the Sonoff is in AP mode in tasmota how would it get "normal" mqtt-messages?

on the other hand, I'm still thinking, if pulsetime worked both ways it would be not only good for our modem/router issue, but other use cases also. I don't think, it had unwanted side effects, does it? for the very least, pulsetime could have an option for 

- use it only with OFF
- use it only with ON or
- use it both ways?

Jason2866, 2018-03-14:
@binderth back to your first topic. Quick and dirty solution
Define your own module with inverted (false definend) relay. 
With that you could invert command Pulsetime (mqtt states ON, in real it is OFF)

binderth, 2018-03-14:
Thanks for the tipp: But how do I make a own module? ;)
could you point me to a start? Thanks!

Jason2866, 2018-03-14:
Take a look for the configuration for your module in: sonoff_template.h
After that go in the web console and change module to type "generic"
Copy config from orig. setup 1:1 except relay. Invert this (if orig Relay1 new Relay1i)

curzon01, 2018-03-14:
@binderth - don't use PulseTime for delayed power on, simply use the [Backlog](https://github.com/arendst/Sonoff-Tasmota/wiki/Commands#using-backlog) feature with Delay, e. g.

`Power OFF;Delay 600;Power ON` 

switches the power off, wait 60 sec and switches the power on

Jason2866, 2018-03-14:
@curzon01 Better solution!

binderth, 2018-03-14:
That's great @curzon01  Thank you!

closed 2018-03-14 by binderth

krue81, 2020-11-06:
> Unfortunately pulsetime only works with ON and it'll shut OFF after specified time... Would be great you if it worked both ways!
> So you can not only use it as staircase circuit, but also to shut something down after a specified time! Would come in handy not only in my use case I bet! So I can shut on some device after shutting it down without further commands.

Had the same problem; needed a switch to turn on 10 seconds after turning it off and ended using this solution (on tasmota console):

RULE1 ON Power1#state=0 DO ruletimer1 10 ENDON ON rules#timer=1 DO Power1 1 ENDON

RULE1 1

Jason2866, 2020-11-06:
Many ways to solve with Tasmota 👍