arendst/Tasmota #2154: Possible memory issue what else can I disable?

EGP01, 2018-03-13:
Make sure these boxes are checked before submitting your issue - Thank you!

- [Sonoff T1 3CH ] What hardware you are using
- [ Not applicable] Provide the output of command ``status 0``
- [Don't have ] If you have a stack dump decode it: https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.rst
- [ Don't have] For better debug messages: https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.rst

I have been having issues flashing my Sonoff t1. I have been seeing the classic issues of memory as described on the troubleshooting guide - after a while wifi connection lost, web management page unresponsive. I have been trying to disable as many unnecessary features as possible but I am running out! The only thing I have had to leave on is the webserver page as I don't know of anyway to select the correct sonoff module I have.

Yesterday I reflashed again. Memory was 26kb and file size was 44%! Today seems more stable and is working even though the web management page has stopped working.  Not sure what else I can disable or is this another issue that I am missing?

Frogmore42, 2018-03-13:
Which lwip are you using?
Search the issues and you will find lots of examples of people having issues with using the latest lwip.

EGP01, 2018-03-13:
I tried the previous build (2.4 board manager) a few times and used 1.4 prebuilt but wasn’t successful. Thought I would try the latest and selected v2 low memory. 
I went with that as when I compiled it saved 4% memory over the other.

All features I can feasibly turn off are off. Domoticz, emulator, TLs, sensors, ws2812 and ir, mdns.

Frogmore42, 2018-03-13:
You might try one of the prebuilt images to see if they work better for you.  Until you get familiar with the process of making changes, building, installing, debugging, it is best to use something that has worked well for others.  Right now, there are too many things that could be wrong to say what actually is wrong.

EGP01, 2018-03-14:
So my latest sketch has been working for 24hours. No connection to the web page but switch is working and responds to on off commands from hassio which is the main thing.

curzon01, 2018-03-14:
@EGP01 wrote
> No connection to the web page

Try prebuild binary or use Library 2.3.0 instead of the 2.4.0 from developer branch adn you might see web page works too. How to select an another lib can be read under https://github.com/arendst/Sonoff-Tasmota/wiki/Theo's-Tasmota-Tips#20180103---tasmota-is-ready-for-esp8266arduino-version-240

EGP01, 2018-03-14:
Thanks for tip. I think I might leave it a bit. The switch itself is working and remembers settings on reboot so I am reluctant to meddle and break it again! But I will certainly try if I run into any more issues