arendst/Tasmota #2155: new H801 unflashable ?

Zixim, 2018-03-13:
Make sure these boxes are checked before submitting your issue - Thank you!

- [H801] What hardware you are using
- [n/a ] Provide the output of command ``status 0``

Have been testing an 2 H801 devices for a few weeks with Tasmota + mqtt + Home Assistant.
All has been going well, so i decided to get a few more H801's.
Just received a batch of 3 H801 devices and I tried flashing Tasmota on them, without any success.
After reboot, this is what serial console shows: 

```
 ets Jan  8 2013,rst cause:2, boot mode:(3,5)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v401052cc
~ld
```

Flashing a backup of the original firmware, yields the same result.
Any suggestions ?

Zixim, 2018-03-14:
this gets weirder...those devices that will not accept Tasmota :  they will flash properly to espurna ?!?!
using the exact same esptool.exe command

curzon01, 2018-03-14:
As we do not know from where you get your Tasmota binary an how (self-compiled, [development branch](https://github.com/arendst/Sonoff-Tasmota/tree/development) or [master branch](https://github.com/arendst/Sonoff-Tasmota/tree/master)) I **suggest to try one of the ready compiled and tested release binaries** from [Releases](https://github.com/arendst/Sonoff-Tasmota/releases) to exlcude all possible side-fx from a self-compiled binary.

If those binaries also not working, try first the [minimal](https://github.com/arendst/Sonoff-Tasmota/releases/download/v5.12.0/sonoff-minimal.bin) one.

Zixim, 2018-03-14:
I have only used downloads from the Releases as per your link.

curzon01, 2018-03-14:
minimal too?

Zixim, 2018-03-14:
my edit must have crossed your reply.

Zixim, 2018-03-14:
to recap: weeks ago, i downloaded minimal, then flashed it to my lab testing devices.
Now i tried to flash the same file to 2 new H801 ... renders them useless....redownload, no fun, full version downloads, no fun, older versions, no fun, backups from original fw , no fun...

curzon01, 2018-03-14:
esptool or esptool.py?

Personal I'm using pthon [Esptool](https://github.com/arendst/Sonoff-Tasmota/wiki/Esptool) `esptool.py`.

If I get new ESP 8266 devices (doesn't matter which) I always do also an erase flash
`esptool.py --port /dev/ttyUSB0 erase_flash`

Zixim, 2018-03-14:
i always erase_flash as well... have done so from day 1.
I usually flash from a Raspberry Pi, with esptool.py using Pi pins 1(3.3v), 6 (GND), 8 (TXD) and 10 (RXD).

With this weirdness, i also tried esptool.exe on windows with the exact same bad result.
Then i tried esptool.exe with an espurna image...and it booted.
Go figure.

Zixim, 2018-03-14:
for the sake of completeness : 
```
PS C:\flash> .\esptool.exe --port COM10 erase_flash
esptool.py v2.1
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266
Uploading stub...
Running stub...
Stub running...
Erasing flash (this may take a while)...
Chip erase completed successfully in 0.9s
Hard resetting...
PS C:\flash> .\esptool.exe --port COM10 write_flash -fs 1MB -fm dout 0x0 .\sonoff-minimal.bin
esptool.py v2.1
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Compressed 431904 bytes to 301134...
Wrote 431904 bytes (301134 compressed) at 0x00000000 in 26.8 seconds (effective 128.7 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting...
```

reboot & open the comm port : 
```
 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v401052cc
~ld
```

arendst, 2018-03-14:
After flashing and re-applying power so the device is reset are you able to access the webpage and select module H801?

The H801 uses GPIO2 for serial Tx where all other devices use GPIO1. This is handled by selecting module H801.

As it's the Tx signal you might even try a serial command  ``module 20`` to select the correct H801.

Zixim, 2018-03-14:
I don't think i'm understanding you ?
These devices don't seem to boot after flash & power-cycle...while the devices from 6 weeks ago are behaving as expected.
I have now taken an older one and used the same connections to get serial comms going...works as expected.

Yes, compared to a Sonoff device the H801 needs Tx connected to Tx and Rx to Rx. Is thos what you mean when you say "The H801 uses GPIO2 for serial Tx where all other devices use GPIO1" ??

So you would like me to try typing blindly ```module 20``` after flashing ?

On the other hand, serial comms after flash & reboot seem fine, since i'm getting output...granted it isn't the output that we would like (Tasmota serial logging), but there'e data coming over the line...

curzon01, 2018-03-15:
@Zixim: For completeness - you posted the flash parms using esptool.exe for minimal:
`.\esptool.exe --port COM10 write_flash -fs 1MB -fm dout 0x0 .\sonoff-minimal.bin`

Did you used the same parms when you successful flashed espurna?

Zixim, 2018-03-15:
I promise you i did indeed: 
```
.\esptool.exe --port COM10 write_flash -fs 1MB -fm dout 0x0 .\ESPEasy_v2.0-20180312\espurna-1.12.4-huacanxing-h801.bin
esptool.py v2.1
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Compressed 487728 bytes to 352868...
Wrote 487728 bytes (352868 compressed) at 0x00000000 in 31.4 seconds (effective 124.4 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting...
```

reboot : 

```
[000325] [INIT] ESPURNA 1.12.4
[000325] [INIT] xose.perez@gmail.com
[000325] [INIT] http://tinkerman.cat

[000326] [INIT] CPU chip ID: 0x6F94DD
[000328] [INIT] CPU frequency: 80 MHz
[000331] [INIT] SDK version: 1.5.3(aec24ac9)
[000335] [INIT] Core version: 2_3_0
[000338] [INIT] Core revision: 
[000341] 
[000342] [INIT] Flash chip ID: 0x1440EF
[000345] [INIT] Flash speed: 40000000 Hz
[000349] [INIT] Flash mode: DOUT
[000351] 
[000352] [INIT] Flash sector size:     4096 bytes
[000357] [INIT] Flash size (CHIP):  1048576 bytes
[000361] [INIT] Flash size (SDK):   1048576 bytes /  256 sectors
[000367] [INIT] Firmware size:       487728 bytes /  120 sectors
[000372] [INIT] OTA size:            536576 bytes /  131 sectors
[000378] [INIT] SPIFFS size:              0 bytes /    0 sectors
[000384] [INIT] EEPROM size:           4096 bytes /    1 sectors
[000389] [INIT] Empty space:          16384 bytes /    4 sectors
[000395] 
[000396] [INIT] BUILD_FLAGS: -g -Wl,-Tesp8266.flash.1m0.ld -DHUACANXING_H801
[000403] [INIT] BOARD: HUACANXING_H801
[000406] [INIT] SUPPORT: ALEXA BROKER DEBUG_SERIAL DEBUG_TELNET DOMOTICZ HOMEASSISTANT MDNS_SERVER NTP SCHEDULER TELNET TERMINAL THINGSPEAK WEB

[000418] [INIT] Last reset reason: External System
[000423] [INIT] Settings size: 416 bytes
[000427] [INIT] Free heap: 29608 bytes
[000430] [INIT] Power: 3126 mV
[000432] [INIT] Power saving delay value: 10 ms
[000436] 

```

curzon01, 2018-03-15:
I know it sounds ridiculous but can you verify (hex view, compare with downloaded) your binary `./sonoff-minimal.bin` - exact this you tried to flash.

That's incredibly. May be checking flash healthy could help finding the reason
- write known local binary to ESP flash
- read it from ESP and compare with local stored

What I'm also missing is the bootloader output `,rst cause:a, boot mode:(x,y)` when you boot espurna, it's the same (`mode(3,y)`)?

Zixim, 2018-03-15:
@curzon01 
What I'm also missing is the bootloader output ,rst cause:a, boot mode:(x,y) when you boot espurna, it's the same (mode(3,y))?

How do I get to the bootloader output ?? I think I copied all output from the serial console...

I will do MD5 hashes of the binaries, before flashing, and after readback from device.
We will see if they are different.

Zixim, 2018-03-15:
i have just 

erased flash :  `.\esptool.py_21.exe --port COM10 erase_flash`

written 487.725 bytes :  `.\esptool.py_21.exe --port COM10 --baud 115200 write_flash -fs 1MB -fm dout 0x0 .\espurna-1.12.4-huacanxing-h801.bin`

rebooted straight back into download mode

read 487.725 bytes : `.\esptool.py_21.exe --port COM10 --baud 115200 read_flash 0x00000 0x77130 image1M.bin`

made 2 md5 hashes, one for each file : 
```
4142f8ba021286dd5fc3fdeb9529c428 .\espurna-1.12.4-huacanxing-h801.bin
4142f8ba021286dd5fc3fdeb9529c428 .\image1m.bin

```

Meinsss, 2018-03-18:
I can confirm this issue.
My H801 is working with espurna (and others) fine, but when it comes to flash tasmota I'm getting the same errors as @Zixim.

Zixim, 2018-03-18:
@Meinsss which other firmwares are working for you ? Espurna, and ... ?

Meinsss, 2018-03-18:
I also tried these two:
https://github.com/open-homeautomation/h801/tree/master/mqtt
https://community.smartthings.com/t/release-smartlife-h801-rgbw-led-strip-wifi-controller-bulb/51182

(Goal: Firmware between HASS and CCT Led Strips (warm/cold white)

Zixim, 2018-03-19:
@curzon01 , do you have further ideas?
Seems i'm no longer the only one experiencing this...

@Meinsss , the H801 that are refusing Tasmota, are they recently purchased? Have you ever flashed Tasmota on them?

Meinsss, 2018-03-19:
@Zixim, this is my first one bought on 3rd February this year. This was my first try, so no I did not flashed tasmota successfully on them.

namadori, 2018-03-20:
Can you send a picture of the board? I can compare to a working one I have near to check for differences

Zixim, 2018-03-25:
[image]
[image]
[image]
[image]

Skaronator, 2018-03-30:
Got my H801 yesterday. (Shipping was ~26th Feb.) I'm able to flash Sonoff with the first try. Here is the whole cmd log:
```
C:\Users\Skaronator\Desktop\H801>esptool --port COM6 erase_flash
esptool.py v2.3.1
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
Uploading stub...
Running stub...
Stub running...
Erasing flash (this may take a while)...
Chip erase completed successfully in 0.9s
Hard resetting via RTS pin...
```
```
C:\Users\Skaronator\Desktop\H801>esptool --port COM5 write_flash -fs 1MB -fm dout 0x0 sonoff.bin
esptool.py v2.3.1
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Compressed 496960 bytes to 342591...
Wrote 496960 bytes (342591 compressed) at 0x00000000 in 32.8 seconds (effective 121.2 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

Zixim, 2018-03-31:
@Skaronator and does it boot after the flash ?

Skaronator, 2018-03-31:
Yes everything works. (Just one problem: Where can I find all the MQTT commands?)

Zixim, 2018-03-31:
https://github.com/arendst/Sonoff-Tasmota/wiki/Commands

Skaronator, 2018-03-31:
Can't find any turn on/off commands in H801 table and `http://192.168.178.201/cm?cmnd=Power%20Toggle` doesn't work

Zixim, 2018-03-31:
easiest way to enter commands is through the web console.
Browse to http://IP-ADDRESS/cs?
and enter commands in the line below the textbox.

Sanfam, 2018-04-02:
To rule out the obvious, have you tried flashing using another usb-serial adapter?   I originally had a high failure rate following flash attempts using an FT232 adapter and later a CH340g, but found myself nearly 100% successful with a CP2104 board.

soprater, 2018-04-02:
I'm using a CP2102 and flashed espurna flawlessly.
Executed the same commands with same output like @Skaronator, but getting this weird
`ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v401052cc
~ld`
on the monitor and nothing else.

Zixim, 2018-04-02:
I have tried a FT232RL adapter, and have also tried with the serial pins on a Raspberry Pi... these boards do not boot after flashing.
Flashing completes without errors, the board never boots.
Using the _same_ serial connections, other firmwares flash & boot.

soprater, 2018-04-08:
I don't know what kind of sorcery this is, but it works now.

Today my second H801 arrived, flashed it with 5.12 (used .zip from Releases), edited platformio.ini (uncommented one env_default), added wifi and mqtt credentials to user_config.h and disabled domoticz + everything after line 175 (One wire sensors).
End of the story: It boots up. Same with the one which ran into the error which we discussed here.

Hope it helps!

irl, 2018-04-21:
I've erased the flash and written sonoff.bin 5.12.0 from the releases page using esptool.py v2.3.1 and I am now also seeing the following in an endless reboot loop:

```
 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v09826c6d
~ld
```

Zixim, 2018-04-22:
unfortunately this isn't being looked into.

irl, 2018-04-22:
@Zixim Looking at your photographs, I have the same board that you do. Has anyone been able to confirm that there has been a change in the board hardware? This is my first H801 so I have nothing to compare to.

irl, 2018-04-22:
I got it working. I'm no expert at this, but I was inspired by @soprater's comment and gave compiling my own image a go. If it's helpful, I can give you details on versions I used.

I had no serial output when I first booted it, but it appeared after I set the module to be an H801 through the web interface. As I had set already the wifi credentials, it was already connected to the network. I suspect that previously it had been rebooting every few seconds as it wasn't finding a wireless network or being configured via the "button" into a mode. (I observed this with the S20s I flashed yesterday, I assume this is expected behaviour).

I wonder if actually, everything was working, but the problem was instead just the lack of serial output and lack of being able to get to the web interface or some other method to set the module type.

irl, 2018-04-23:
Awesome! Thanks for the explanation. I can look at updating the documentation in the wiki tonight and perhaps that is enough to close the issue (i.e. no code changes required, this issue is expected)?

arendst, 2018-04-23:
Pls do. Issue is as expected.

Zixim, 2018-04-23:
Blindly setting ``module 20`` does not work with minimal bin.
Confirmed it solves the issue with 5.12 (full bin).

Does not explain why some H801 devices do not need the blind parameter setting, and some do.

irl, 2018-04-23:
@Zixim Maybe some H801 actually do use GPIO1 for Serial TX? Do you have a board on which you can reproduce it working out-of-the-box with the 5.12 full binary?

Zixim, 2018-04-23:
Yes, i do have boards that work oob.

Zixim, 2018-04-23:
[image]
This one flashes oob.

irl, 2018-04-23:
The only difference I can spot is the winbond chip, which seems to be a slightly different model. I think they are functionally equivalent though. When you say it works oob, do you mean that there is serial console output or that you can connect to the wifi?

I have seen comments (though I can't remember where) that talk about needing to power it from an external supply and not via the 3.3v header in order to boot the image successfully. As I already had a power supply hooked up, I didn't actually try booting it with the 3.3v header pin, so I don't know if that would have worked or not.

Zixim, 2018-04-23:
When I say it works, I mean I can flash minimal (or full) bin, and it boots normally.  Allowing me to further configure it.
As opposed to what I described in post #1. (Which is now solved, when blindly typing ``module 20`` into the serial console. This 'trick' does not explain why some boards work oob, others don't. Seems funky to me.

I always power the boards via the actual power connector. Used to hook up the 3.3v on the serial header, but the other way (green screw terminal pwr connector) is much more practical for me.

irl, 2018-04-30:
https://github.com/arendst/Sonoff-Tasmota/wiki/H801

@arendst I believe this issue can be closed now that this is documented.

@Zixim While your situation is interesting, it's a case of Tasmota working better than expected, not really a bug. If you ever find out why some of your boards work oob and some don't then I'd be interested to know. I unfortunately don't have the time to look into it further.

thecomalley, 2018-05-21:
Am a bit stuck trying to get the H801 to work

Once the device flashes i am unplugging the USB adapter, taking it out of flash mode (removing jumper wire) and plugging it back into the PC open a putty console to blindly type `Module 20`?

after doing this i am only getting the following from putty (Serial COM3, 9600) have also left it plugged into the mains post flash with no change, what am i doing wrong?

[image]

```
esptool.py.exe --port COM3 write_flash -fs 1MB -fm dout 0x0 sonoff.bin

esptool.py v2.3.1
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Compressed 542192 bytes to 370949...
Wrote 542192 bytes (370949 compressed) at 0x00000000 in 32.9 seconds (effective 131.9 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

Zixim, 2018-05-21:
serial speed should be 115200

thecomalley, 2018-05-22:
Ah that's better thanks.
Now getting the following, is this where i need to blindly type `module 20`?
have tried this and multiple reboots, flashed using esptool and 5.14.0
```
 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v401052cc
~ld
```

Zixim, 2018-06-02:
sorry for the late reply...
yes, it is the moment when you would issue the command, BUT don't use the venerable Putty... I've never gotten it to work for sending commands to the ESP device.
Use Termite instead : https://www.compuphase.com/software_termite.htm

orsgah, 2018-08-19:
Same problem after  istall 6.1.1, after than i cannot istall any version, module boot loop with error:

```
ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v401052cc
~ld

```

ozett, 2019-01-05:
i have problems to get the h801 into flash-mode.
means it will not flash.
i power it from 3.3 volts.

can anybody confirm that this is difficult from 3.3v 
and it must better be powered by normal power-in?

like the magichome-controllr?
https://github.com/arendst/Sonoff-Tasmota/wiki/MagicHome-LED-strip-controller

ozett, 2019-01-05:
i thought i did bad soldering and inspected my work,
but i found bad soldering on the board. 
it is so tiny, i did not manage to get rid of it. maybe the esp8266ex is already dead?
i think, i go now with the magichomecontroller... 🙈 

[image]

egeu5, 2020-06-23:
> @orsgah : The wiki explains a work-around, as does this thread.

Hi,
could you explain the workaround? In the wiki is only the command "module 20" to change the tx-pin from 0 to 1. This did not solve the problem. I tried to flash minimal but this is not working and ends in a boot-loop.

egeu5, 2020-07-26:
> 
> 
> @orsgah : The wiki explains a work-around, as does this thread.

@Zixim 
Hi,
could you explain the workaround? In the wiki is only the command "module 20" to change the tx-pin from 0 to 1. This did not solve the boot-loop problem. I tried to flash minimal but this is not working and ends in a boot-loop. My flash size is 1MB - that should be enough for normal firmware but even minimal is not working.

Zixim, 2020-07-27:
@egeu5 
The solution for this bootloop is explained in gruesome detail over this thread.

If it doesn't work for you, you've got another kind of problem, or you're not doing the `module 20` command properly.
Either way, without more info from your part, no way to help more.

CzarofAK, 2021-11-15:
@Zixim, somehow i am too stupid. where shall i enter the "module 20"?
Using "esptool" on Windows 10, and getting this failure:
C:\>esptool -cp COM4 -cb 115200 -bm dout -cf tasmota-DE.bin -v
esptool v0.4.13 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
setting flash mode from qio to dout
opening bootloader
resetting board
trying to connect
trying to connect
trying to connect
resetting board
trying to connect
trying to connect
trying to connect
resetting board
trying to connect
trying to connect
trying to connect
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed

C:\>module20
Der Befehl "module20" ist entweder falsch geschrieben oder
konnte nicht gefunden werden.

CzarofAK, 2021-11-15:
@barbudor 
thanks for your quick response. i was able to flash it, on of my dupont wire was defect. sorry for my inconvenience.

klrock, 2022-03-12:
As noted by [ozett](https://github.com/ozett) I also found bridged connections on the esp chip.  Tried a repair but too small and too little skill.  Had it for a couple of years, so no hope for replacement.  My lighting project now on hold for who knows how long.