arendst/Tasmota #2156: POW bricked after reset?

Phrede, 2018-03-13:
Fairly new and still learning, please be gentle.  If there is a better place to post things like this please letmeno.

A relatively new POW Sonoff had been working fine for about a week (Other than the timezone change which I was going to fix with a reflash and timezone 99)

I went to EnergyReset in console but issued the wrong command Reset instead.  The unit then lost WiFi communication but still would switch on and off.  I then did multiple switch presses in an attempt to do a full reset.  That is when everything quit working.  The only thing that happens is a quick blue flash when I apply AC power.  I did a reflash which appeared to go well and finished normally but had the same results.

I expect it is now bricked but am open to try if there are suggestions.   I don't think the value of the unit justifies attempting anything more than ordering a replacement.  I do hope to learn something, however.  What did I do wrong?  Is this common, I need spares on hand?  Any spare parts salvageable?

Jason2866, 2018-03-13:
Have you changed this value (in user_config.h) to a different date before reflashing?
#define CFG_HOLDER             0x20161209        // [Reset 1] Change this value to load following default configuration parameters
If not the device wont reset to default values!

curzon01, 2018-03-14:
Be aware changing CFG_HOLDER the user has to change this value to their own every time new releases are compiled - otherwise they lost their own config on next power up.

Phrede, 2018-03-17:
Thanks, that fixed it.   I seem to now have issues doing an OTA update, but re-flash thru hardware solved it.  Always learning....  

I feel like I'm missing something when troubleshooting/researching problems.  Maybe just rookieitis.

Hugo477, 2018-04-06:
I have exactly the same issue, caused by my mistake (Reset 1 instead of restart 1)
I´ve changed date in CFG_HOLDER into my last complilation date, reflash and the same as before, just quick blue led flash after power up, relay switches controlled by a button, no wifi and thats all.
Than I erase flash, reflash and again no change. 
Des it mean that wifi is broken or I did any mistake?