arendst/Tasmota #2157: Feature request Sonoff POW: Send power reports when power changes more than a specified percentage

epmsmid, 2018-03-13:
Is it possible to add a feature to send a power report when the power usage changes more than a certain percentage? E.g. like the Fibaro Wallplug E/F?

From the usermanual page 13 and 15:
[image]

[image]

This way the reporting interval can higher and you can still respond to power changes.

arendst, 2018-03-14:
Working on it...

arendst, 2018-03-14:
Provided command PowerDelta \<percentage\> to latest pre-release. See Wiki > Commands for option usage.

arendst, 2018-03-14:
I'm now struggling with the definition 80%.

Current release calculates the percentage by dividing the difference by max_power. This will send one report for coarse changes but small changes are unnoticed.

I'm testing with dividing the difference by min_power. This works ok for fine changes but will send several reports with coarse power load changes...

I'm in doubt what to release...

epmsmid, 2018-03-14:
Hi Theo,
Thanks for your quick response and the great work so far!I think the implementation is like this:Reports are send with 2 priorities:- A big power change (user defined % in this case set to 80%) has a high priority. - A small power change (user defined % in this case set to 15%) has a low priority.
The change is the change relative to the previous power measurement.So if the previous measurement was 10 Watt and the current measurement is 20 Watt, the change is (20 W - 10 W) / 10 W = 0.1 = 10%
A high priority change is always send, so if the the power change from 100 W to 10 W -> 90% a report is send.
For low priority messages a bucket is used to decide if a message may be send. Every interval (in this case 30 seconds) the bucket is set to 5. Every time a low priority message is send, the bucket is decreased by 1. When the bucket is 0, no low priority message message will be send.The interval for setting the bucket to 5 is another interval than the periodic reporting interval.
When the power changes from 10W to 30W -> change 20% a report is send unless the low priority bucket is 0.
I hope this description helps.If you still struggle, please let me know.
Regards,
Eric

epmsmid, 2018-03-15:
While rereading the above it's possibly better not to calculate the percentage of the change against the last value, but against the lowest value of the previous and the current power.

I think the the formula for the change should be:

<power change percentage> = absolute value of (<current power measurement> - <previous power measurement>) / minimum value of (<current power measurement> and <previous power measurement> * 100%

I think this gives a better resolution.
So a change from 20W to 100W will give a percentage of 80/20*100% = 400%
and a change from 100W to 20 W will also gie a percentage of 80/20*100% = 400%

Otherwise a change from 20W to 100W will give a percentage of (100-20)/20*100% = 400%
And a change from 100W to 20W will give a percentage of (20-100)/100*100% = -80%

It's possibly a good idea to also implement a bucket for high priority message, to protect against flooding reports.
Maybe it's a good idea to make the buckets user configurable.
Then you get these parameters:
- Bucket reset interval (Default 30 seconds?)
- Bucket size for high priority reports (Default 6? -> average reporting interval every 5 seconds)
- Bucket size for low priority reports (Default 3? -> average reporting interval every 10 seconds)
- High priority report change percentage (Default 80%?)
- Low priority report change percentage (Default 15%?)

epmsmid, 2018-03-15:
Sorry, accidentally closed the issue.

arendst, 2018-03-15:
Other than reporting on power change I see no reason to discriminate in high or low percentage change. The home automation tool already monitors overall energy uses so the only advantage is the possibility to react on any major power change (percentage set with PowerDelta).

Keep in mind that power measuring could take several seconds so reporting every 5 seconds is a no-go anyway.

I'll need to fine tune current solution, that uses your last proposed solution already, for the pow and will come back here when done.

epmsmid, 2018-03-16:
The reason for discriminating high and low percentage is to protect agains flooding reports.
When the used power fluctuates just above the low percentage, a lot of power messages will be sent. This is why Fibaro limits the number of low percentage messages per time interval. E.g. 1,0W -> 1,2W -> 1,0W ->1,2W etc. When the percenage is set to 15% every change causes an immediate report.
I think big fluctuations as high is 80% are rare.
The reason for this feature request is that I use power changes to trigger scripts. With this feature, the power change is reported earlier.

arendst, 2018-03-17:
For now I leave it as it is. I suggest you play with the percentage to get a functional environment. If you come across a life problem just let me know and show.

ascillato, 2018-06-02:
Hi @epmsmid 

Now there are rules in Tasmota. Please, see if this new feature can solve your issue. Thanks!