arendst/Tasmota #2158: Flashed but now not connecting to my wifi

OxygenGS, 2018-03-13:
Hey guys,

well i did succeed to flash my sonoff :D However, if i wanna check the Serial Monitor in Atom, i can see that it fails to connect to my wifi
[image]

Help please?!

OxygenGS, 2018-03-13:
I did press 4 times quick, now i can acces it with my phone. However, even if i change the settings in there and reboot it and let it try to connect to wifi, it does not work. Help please?

curzon01, 2018-03-14:
might you want to close your own issue #2151 if it success - because this normaly is not a forum, that's an issue list. And if you can access your device even using AP please use `Status 0` and give us some info (at min. the FW):
- [ ] What hardware you are using
- [ ] Provide the output of command ``status 0``