arendst/Tasmota #2160: Sonoff 4CH Pro proper GPIO for sensor

kajmaj, 2018-03-14:
I would like to connect a sensor to Sonoff 4Ch Pro V1.1 as there are GPIOs 1,2,3 available in the module configuration.
There should be 3 GPIOs based on the schematic. GPIO 1 and 3 in J6 (UART TX/RX). GPIO 2 (UART1 TX) - this one  I am not able to find on a board.
Nonetheless if I select GPIO 1 or 3 and sensor "Switch1" in "Configuration"  it is permanently "ON" even if sensor (switch) is not connected.
MQT: tele/4CH/SENSOR = {"Time":"2018-03-14T14:16:26","Switch1":"ON"}
Modul is flashed by Sonoff-Tasmota 5.12.0, MQTT is configured, everything else works well.
Did I overlook anything or my conclusion about GPIOs location is wrong? Or anything completely else?
I would appreciate any help, guidance or location of related GPIOs on the board.

Jason2866, 2018-03-15:
Have you disabled serial logging? You have to if you use RX/TX GPIOs

kajmaj, 2018-03-15:
I´ve disabled serial logging. Thank you for advice, but still the same : 
`MQT: tele/4CH/SENSOR = {"Time":"2018-03-15T14:08:47","Switch1":"ON"}` does not matter on selected GPIO 1,2 or 3.
If I understand well there should be "OFF" if sensor selected but not connected.

Jason2866, 2018-03-15:
Do you realy use the right TX/RX Pins? J6 is false this the UART!!! You have to use J1.
In schematic namend E-RX and E-TX!

kajmaj, 2018-03-15:
Hmm, this is the reason. Thaks a lot. I will try it tomorrow morning.  But my assumption that w/o sensor(switch1)  activated in configuration but not connected it shoud publish "OFF" is not correct?

Jason2866, 2018-03-15:
There are no pullup/pulldown resistors installed. So it can be both (on or off) or change randomly.
So you have to install pullup resistor (s) if you switch to GND. You can try 10k Ohm at the start.
If light switches without a reason go down to 4k7

kajmaj, 2018-03-15:
Thanks a lot. Pullup resistor together with proper pin could make the magic.
Will report after test.

kajmaj, 2018-03-16:
Progress report:
J1 is the rigt location of GPIO 1 and 3. It works perfectly without pullup/pulldown resistor. With resistor installed no way (neither 4k7 nor 10k). It is permanently ON, withh switch activated it goes OFF, no random changes so far.
Now it is getting more complex than I expected. 

I am facing a few more issues related solely to 4CH Pro:

- I need to decouple SWITCH4 from relay 4 (POWER4) and use this SWITCH4 as a sensor for HA. 
Sensor connected either to GPIO1 or 3 and assigned as Switch1,2,3 or 4 in Sonoff "Configuration" controls related relay, therefore I need to decouple it as well. 
- still did not find GPIO2 location.

Thanks in advance for any guidance.

kajmaj, 2018-03-16:
`SwitchTopic`

kajmaj, 2018-03-17:
UR right. My desired scenario is what I should start with.
I want to use Sonoff 4CH Pro to control the irrigation system. 
The system has 3 zones+a rain sensor.
3 zones it means 3 relays controlled either by Homeassistant through MQTT and/or manually by 3 buttons on Sonoff (for the case of WiFi failure) - **solved**.
Sensor is connected to GPIO3 (once more thank you for guiding me) sending Telemetry in 5 minutes period which is fine for such a sensor. - **solved**.

Button 4 I would like decouple from relay 4 and use it to manualy activate the watering cycle script through Homeassistant.

Jason2866, 2018-03-17:
Ok, got it. I would define my own module definition. This is easily done. Change the module type to GENERIC.  In sonoff_template.h is the definition for the modules. Here is the code snipplet for the 4Chpro:
  { "Sonoff 4CH Pro",  // Sonoff 4CH Pro (ESP8285)
     GPIO_KEY1,        // GPIO00 Button 1
     GPIO_USER,        // GPIO01 Serial RXD and Optional sensor
     GPIO_USER,        // GPIO02 Optional sensor
     GPIO_USER,        // GPIO03 Serial TXD and Optional sensor
     GPIO_REL3,        // GPIO04 Sonoff 4CH Red Led and Relay 3 (0 = Off, 1 = On)
     GPIO_REL2,        // GPIO05 Sonoff 4CH Red Led and Relay 2 (0 = Off, 1 = On)
     0, 0, 0,          // Flash connection
     GPIO_KEY2,        // GPIO09 Button 2
     GPIO_KEY3,        // GPIO10 Button 3
     0,                // Flash connection
     GPIO_REL1,        // GPIO12 Red Led and Relay 1 (0 = Off, 1 = On)
     GPIO_LED1_INV,    // GPIO13 Blue Led (0 = On, 1 = Off)
     GPIO_KEY4,        // GPIO14 Button 4
     GPIO_REL4,        // GPIO15 Red Led and Relay 4 (0 = Off, 1 = On)
     0, 0
  },
Copy everything 1:1 except for the definition for GPIO 15 (=Relay 4)
With this setup the relay is unknow and cannot switch

Jason2866, 2018-03-17:
Or better name the relay: relay5
If i am not wrong ist should be still switchable as power5 via mqtt....

kajmaj, 2018-03-17:
Do you mean to rename REL4 to REL5?
`GPIO_REL5, // GPIO15 Red Led and Relay 4 (0 = Off, 1 = On)`
It sounds good.
Button 4 will create `MQT: stat/4CH/POWER4`
Relay "4" now "5" will be controlled via `MQT: cmnd/4CH/POWER5`
Do I understand well?

kajmaj, 2018-03-17:
Hm, it means enable serial logging again and reflash it with modified FW via PlatformIO? Or is it possible to do via OTA?

Jason2866, 2018-03-17:
Yes correct. No need to reflash!!  Just change the config via the Web Site of the module.

kajmaj, 2018-03-17:
I should modify the snippet, copy it, paste it to the command line of Console and hit the Enter?
If yes, it´s easier than expected.

Jason2866, 2018-03-17:
No, nothing like that. My explanation was a bit short.
Every sonoff module has a esp chip inside. The ESP chip has GPIO pins numbered from 0 to 16
If you setup the module typ a fix designed correct pin assignment is selected.
In the snipplet above you see the pin assignment for the Sonoff 4ch pro.
If you select the module Generic no pin assignment is made. You can and have to! assign the GPIO with
your functions that fits to the module.

kajmaj, 2018-03-17:
Up to "In the snipplet ..." I understand.
Last two lines are greek to me a bit.
If I select "Generic" there are GPIO0,1,2,3,4,5,12,13,14,15,16 i.e. GPIO9,10 are missing.
Using "modified" snipplet, I can assign all GPIOs, except 9,10.
I can add and assign 9 and 10 via MQTT command. `Gpio<pin>  <sensor>` ?
Sonoff remain with "my" settings untill OTA update or settings "survive" FW update?

Jason2866, 2018-03-17:
S**t i overlooked GPIO 9 and 10.  You are right cannot assigned this way.
So not that easy way. I take a look in the command set... just a moment.

Jason2866, 2018-03-17:
Command Gpio not possible for 9 and 10. The are not defined in module generic as user configurable.
This is a good idea. On almost all modules this Gpio pins are needed to adress the flash...
So a liitle harder way. Compile your own firmware. You have to change in sonoff_template.h
one setting in the configuration of the module sonoff 4ch pro
Change
GPIO_REL4, // GPIO15 Red Led and Relay 4 (0 = Off, 1 = On)
In
GPIO_REL5, // GPIO15 Red Led and Relay 5 (0 = Off, 1 = On)

Jason2866, 2018-03-17:
Compile. You have made it!  Upload file via the web browser upgrade button.
Your done. If you compile your Tasmota change in user_config.h the settings for your wlan.
Gateway etc. The module starts and connects to your wifi. You  can do all from your PC

kajmaj, 2018-03-17:
Thank you for your effort.
Pricipally back to my first idea - create modified sonoff.bin and either flash it via PC or flash it as OTA.
And do it before any further FW update.

Jason2866, 2018-03-17:
Yes thats the way. One big minus off this way. On every new firmware you upload to your 4ch you have
to made this change by yourself.

kajmaj, 2018-03-17:
I will try it and report :)
You made my day. Thank you.

Jason2866, 2018-03-17:
Your welcome

kajmaj, 2018-03-17:
Hmm, nothing is easy.
When I changed Relay 4 to Relay 5 there popped up 5 errors in Linter in Platformio after "Build and Save".
Error remain there even if I returned it back to Relay 4.
The only way is to delete `Sonoff-Tasmota-development`  file from the PC and unzip it again
Linter error report:
`Error	GCC	expected initializer before 'PROGMEM'	112:27
Error	GCC	expected initializer before 'PROGMEM'	186:25
Error	GCC	'uint8_t' does not name a type	189:3
Error	GCC	'uint8_t' does not name a type	197:7
Error	GCC	expected initializer before 'PROGMEM'	242:35`

```
112:  const char kSensorNames[] PROGMEM =const char PINS_WEMOS[] PROGMEM = "D3TXD4RXD2D1flashcontrolD6D7D5D8D0A0";
186:  const uint8_t      io[MAX_GPIO_PIN];
189:  const uint8_t kNiceList[MAXMODULE] PROGMEM = {
197:  const uint8_t kNiceList[MAXMODULE] PROGMEM = {
242:  const mytmplt kModules[MAXMODULE] PROGMEM = {

```

What I did is:
I changed  `GPIO_REL4, // GPIO15 Red Led and Relay 4 (0 = Off, 1 = On)`
to `GPIO_REL5, // GPIO15 Red Led and Relay 5 (0 = Off, 1 = On)`
And hit "Build and Save"

kajmaj, 2018-03-18:
I do not know what the errors above mean, but compilled file is apparently OK. 
I've flashed it to 4CH Pro via website.
Button is apparently decoupled with following result.
Button 4 creates `MQT: stat/4CH/POWER4` and `cmnd/4CH/POWER4 = ON` as well, Relay4 does not react.
Nonetheless Relay4 now "5" is not controlled via `MQT: cmnd/4CH/POWER5` because such a command is considered as unknown. 
`CMD: cmnd/4CH/POWER5`
`MQT: stat/4CH/RESULT = {"Command":"Unknown"}`

My desired state is moreless achieved, I needed decoupled Relay4 and Button4.
If I need to use Relay4 in the future again, FW change back to original is needed.

Jason2866, 2018-03-18:
If i have time next week, i will try this setup config...
Up to 8 relay should be possible to config

kajmaj, 2018-03-18:
To have a complex solution ready it would be great if you find a way.

Jason2866, 2018-03-18:
I just did:
changed GPIO_REL4, // GPIO15 Red Led and Relay 4 (0 = Off, 1 = On)
to GPIO_REL5, // GPIO15 Red Led and Relay 5 (0 = Off, 1 = On)
Save. Build -> no error -> successfull! Since i have no 4ch i cant test it.
4ch is on the way from china...
So there must be something wrong in your change of sonoff_template.h or your Atom.io
Setup is buggy. I use Atom.io too.

kajmaj, 2018-03-18:
Lets assume that my build is OK and I believe that it is. 
Those errors are questionable, I checked related lines before and after and they are unchanged.
I've tested it on Sonoff Basic - flash OK, functions OK. So I've updated FW on 4CH Pro afterwards as well - update OK, all functions OK, button4 works as Switch4, exactly as expected (providing stat/4CH/ Power4/ON or OFF" and not controling Relay4).
The one and only issue remains - there is no way how to control "newly" created Relay5 neither from console nor from the HomeAssistant by a published mqtt command with topic /4CH/Power5.
During a week I could try to rebuild FW again and will see if something changed.

kajmaj, 2018-03-20:
Small update, it lloks like the 4CH Pro died.
 
As 4CH Pro started to behave oddly ie. randomly on/off switches, published doubled mqt/stat messages for 1 switch, if I manualy switched between 2 switches it perform in a strange sequence like: "from switch1 to switch2-back to switch1-back to switch2" It was like hesitating what to do.
I decided to upload "clean/unmodified " FW via website.
Since that it is "dead" no communication at all.
If powered, the only visible activity is blue LED flashing in following pattern "30 x flash pause, 2x flash pause, cca 8 sec light, pause" and again.

I tried reflash it via FTDI adapter, but:
`Uploading .pioenvs\sonoff\firmware.bin
warning: espcomm_sync failed
error: espcomm_open failed

error: espcomm_upload_mem failed
*** [upload] Error -1`
RX/TX are connected correctly. (just to be sure I changed to TX/RX but the same)
During the upload proces is flashing LED indicating TX only (I guess, nonetheless second LED is OFF)

I tried t throubleshot following [https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting](url)
But:  
`c:\Users\K\.platformio\python27\Scripts>esptool.py --port COM7 erase_region 0x0F4000 0x008000
esptool.py v2.3.1
Connecting........_____....._____....._____....._____....._____....._____....._____....._____....._____....._____
A fatal error occurred: Failed to connect to Espressif device: Timed out waiting for packet header`

Is there anything else I could do before I will smash it and throw it to the trash and order a new one?

Jason2866, 2018-03-20:
Have you tried the long press button mode to reset?
https://github.com/arendst/Sonoff-Tasmota/wiki/Button-usage

kajmaj, 2018-03-20:
Yes. Only blue led flashing pattern changed. 30x short - 1x long - 1x short - approx 30 sec pause.
In addition I´ve tried 
- ESP8266 SmartConfig app - no connection
- Access Point - no connection
I am either doing sthg wrong or it is over.

BTW, I compiled FW once more today (just to prepare it for a new 4CH Pro) and it went without errors.
Maybe I killed this piece by improperly compiled FW.

fanofsonoff, 2018-07-18:
I am trying to make Sonoff 4Ch Pro R2 as Garage Door Openers and Sensors for my 2 door garage( 2 separate doors). I disabled serial logging. I tried whatever possibilities I can think of to configure the sensors. But, they are not working. Can anyone post config.yaml, console configuration and anything else , that would be greatful

Jason2866, 2018-07-19:
Read the wiki for switchmode and switchtopic.
If you need help for HomeAssistant ask there.

dastrix80, 2019-03-28:
Hi All

I want to setup a 2 wire dry contact sensor for a Door.

Can I use RX/TX for this on my 4CH PRO ? GPIO1 & 3? I assume it doesnt matter which wire connects to which pin? Thanks

dastrix80, 2019-03-28:
Do I need one pin on ground and one on either RX or TX? Confused as to hwo to wire it.

flyboy013, 2020-02-17:
I was having difficulties figuring it where to post for help, so I figured I would post here since I’m having the same issue.   

I using a Sonoff 4CH Pro r2 and having difficulties getting a GPIO switch to publish to MQTT as opposed to activating on of the relays.

I’ve installed tasmota 8.1.0 and have changed the module type to Sonoff 4CH Pro (23). After reboot I defined switches 1-3 which are associated with GPIO1-3, respectively. I’ve disabled serial logging with command “SerialLog 0”. I read that the default behavior of a switch is for the corresponding relay to be activated and that a rule should be set to have a MQTT message published instead. I’ve created three rules:

```
Rule1 on Switch1#State do publish cmnd/custom-topic/SWITCH1 %value% endow Rule1 1
Rule2 on switch2#state do publish cmnd/custom-topic/SWITCH2 %value% endon Rule2 1
Rule3 on switch3#state do publish cmnd/custom-topic/SWITCH3 %value% endon Rule3 1
```

I have rebooted the devices since setting up this configuration. I’ve tried shorting Tx, Rx, and SDA (on J1 header) to ground, but instead of getting the correspond messages published, the relay is activated. I feel like I’m missing something pretty basic, but after trying various configurations and searching the web for the past few days, I cannot figure it out. Some guidance would be appreciated. Thanks.

ascillato, 2020-02-17:
@flyboy013 

have you defined those GPIOs as switches?

also you syntax is not ok. The final part of a rule is ENDON not ENDOW
also you can write all the rules in the same rule slot. Please try in the console the following:

```yaml
rule1 1
rule1 on Switch1#State do publish cmnd/custom-topic/SWITCH1 %value% endon on switch2#state do publish cmnd/custom-topic/SWITCH2 %value% endon on switch3#state do publish cmnd/custom-topic/SWITCH3 %value% endon
```

If you need further assistance, please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you.

Thanks.

---------------------------

### Support Information 
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

flyboy013, 2020-02-17:
@ascillato 

The typo was only in the post, but combining the rules into one solved the problem.  Thank you for your help.