arendst/Tasmota #2162: Please set WifiConfig 5 as default

rcarmo, 2018-03-14:
I am using S20 sockets, and every time there is a minor outage or a Wi-Fi issue I have to go around setting WifiConfig 5 on all sockets, because they reboot and lose state, going into Wifi Manager mode.

Is there a way to have it be the default without recompiling the firmware?

In fact, why isn't it the default?

Jason2866, 2018-03-15:
That is not the normal behaviour of the Tasmota Software. If set WifiConfig this states doesnt change normally at power loss. The config is stored in flash. I think your used software is faulty in some way.
What Version do you use and where have you donwloaded it?

rcarmo, 2018-03-15:
5.11.1, from the releases page. I upgrade regularly (about every other release).

Jason2866, 2018-03-15:
Hmm, strange... My suggestion the best would be to try a own compiled version with your wifi setup.
I didnt use any prebuild image, so perhaps prebuilt images react different

rcarmo, 2018-03-15:
I'm confused. So you step into a discussion questioning the origin of the firmware image and outlining the base functionality of the original software and now suggest building a custom image and point out you didn't use a prebuilt yourself? 

That makes me curious: what kind of experience do you have with the default behaviour of the prebuilt image?

Jason2866, 2018-03-15:
I never used a prebuild image. But in the configuration of self compiled version there is no option in there is wificonfig configured... So there should no difference in this behaviour from a precompiled to a selfcompiled one. My idea is to try with a selfmade one if the error goes away or it stays

reloxx13, 2018-03-15:
build it yourself.

rcarmo, 2018-03-17:
I’ve just had this happen again, and the repeated reboots after 90s (with the relay flashing upon reboot) fried a lamp. And I am positive I set WifiConfig 5 on that particular socket.

I think there are two distinct issues here: reboots in wifimanager mode and relay toggle, and the loss of WiFiConfig 5 (which I set manually on all sockets) without any apparent reason.

Jason2866, 2018-03-17:
You reference to questions from users who needs help in configuration -> no issues
I can only repeat what @reloxx13 said. Build your own.

rcarmo, 2018-03-17:
I think you’re misrepresenting the issues here. Regardless of cause, two things are happening:

* the device loses configuration after using WiFiConfig 5, and enters a reboot loop in wifimanager
* the relay toggles during repeated reboots on the S20 and causes hardware failure overnight

Also, I’m starting to suspect this is a firmware stability issue and not necessary a WiFi one. I am using homebridge and logging the RSSI values using Node-RED, and the two most problematic outlets have the same signal quality (on average) than the others.

My building my own image won’t fix this.

Jason2866, 2018-03-17:
Your last sentence is not correct, you think that wont fix that. Others and i do.
Your turn (or not).

rcarmo, 2018-03-17:
I'm honestly quite sorry that you have that attitude. Regardless, for reference, here are the socket RSSI charts. Note that Socket 4 (which was the one that failed today) still has a pretty good RSSI compared to the others both before the crash (which is beyond the chart limit, but recorded) and after.
[image]

(Sockets 7, 8 and 9 were removed from the chart to make it easier to see 4)

Jason2866, 2018-03-17:
Ok. Yes the crash and your wlan setup is good. 
So one more reason that you try to build your own firmware to see if things are going fine with that.
I have over 10 sonoff and about ten Esp devices with Tasmota in use, and not even one crashed since last firmwareupdate 5.12.... (Uptime 26T19:18:06)

arendst, 2018-03-17:
Make sure these boxes are checked before submitting your issue - Thank you!

- [ ] What hardware you are using
- [ ] Provide the output of command ``status 0``
- [ ] If you have a stack dump decode it: https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.rst
- [ ] For better debug messages: https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.rst

rcarmo, 2018-03-17:
- ITEAD S20
- `status 0` gives me (some IDs redacted):

```
15:36:26 CMD: status 0
15:36:26 MQT: stat/socket4/STATUS = {"Status":{"Module":8,"FriendlyName":"Socket 4","Topic":"socket4","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
15:36:26 MQT: stat/socket4/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://domus1:80/api/arduino/sonoff.ino.bin","Uptime":6,"Sleep":0,"BootCount":44,"SaveCount":96,"SaveAddress":"F4000"}}
15:36:26 MQT: stat/socket4/STATUS2 = {"StatusFWR":{"Version":"5.11.1","BuildDateTime":"2018-01-07T21:51:09","Boot":6,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
15:36:26 MQT: stat/socket4/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"home.lan","LogPort":514,"SSId1":"REDACTED","SSId2":"REDACTED","TelePeriod":300,"SetOption":"00000009"}}
15:36:26 MQT: stat/socket4/STATUS4 = {"StatusMEM":{"ProgramSize":484,"Free":516,"Heap":24,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3}}
15:36:26 MQT: stat/socket4/STATUS5 = {"StatusNET":{"Hostname":"socket4","IPAddress":"192.168.1.117","Gateway":"192.168.1.254","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.254","Mac":"68:C6:XX:XX:XX:XX","Webserver":2,"WifiConfig":5}}
15:36:26 MQT: stat/socket4/STATUS6 = {"StatusMQT":{"MqttHost":"home.lan","MqttPort":1883,"MqttClientMask":"DVES_X","MqttClient":"DVES_81013D","MqttUser":"DVES_USER","MAX_PACKET_SIZE":512,"KEEPALIVE":15}}
15:36:26 MQT: stat/socket4/STATUS7 = {"StatusTIM":{"UTC":"Sat Mar 17 14:36:26 2018","Local":"Sat Mar 17 15:36:26 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1}}
15:36:26 MQT: stat/socket4/STATUS10 = {"StatusSNS":{"Time":"2018-03-17T15:36:26"}}
15:36:26 MQT: stat/socket4/STATUS11 = {"StatusSTS":{"Time":"2018-03-17T15:36:26","Uptime":6,"Vcc":3.212,"POWER":"OFF","Wifi":{"AP":2,"SSId":"7","RSSI":64,"APMac":"64:XX:XX:XX:XX:XX"}}}
```

No stack dumps or debugging. syslog (which I also collect) shows nothing special from this device prior to the time of the crash (as far as I can estimate from the chart).

arendst, 2018-03-17:
Status 0 looks fine.
From the settings shown they should have been saved.

- What settings do you have to restore after the fault?
- Are all set to default?
- What happens if you perform a normal restart, are all settings still there?
- If you enabled syslog what was the RestartReason as returned by INFO3 during reboot or at the Information webpage.
- What features did you disable considering a ProgramSize of just 484kB 

You might want to consider upgrade to the latest released version (5.12.0) or development version 5.12.0f

Jason2866, 2018-03-17:
@arendst: 5.11.1  sonoff.bin is 484kb

rcarmo, 2018-03-17:
By the same order:

* Everything - the S20 loops back to using a `sonof-XXYY`SSID
* Yes
* Yes
* I had syslog set to 0 on this device even though the hostname was configured (this device crashed more than once, so I obviously forgot to set it the last time around)
* I'm using the official build

arendst, 2018-03-17:
Pls provide RestartReason for further analysis.

You might consider the type of load too as other users experienced strange behaviour with inductive loads interfering with the device flash.

Other than that all my 40+ devices never lost settings...

I suggest to update the one that fails the most to version 5.12.0 (prefferably development) and see what happens.

rcarmo, 2018-03-17:
I will, but am still curious as to what the reboot behaviour will be. Is there any way to make sure it won't keep toggling the relay over and over during the night every few seconds if it enters a reboot loop?

Jason2866, 2018-03-17:
I did just a restart 1 on a S20 device with firmware 5.12. Relay does not toggle.
Some users had problems with mDNS. I have this always disabled by default in my settings (no need).
Worth a try: use the ip adress in settings of your broker not a name (needs to be resolved)

arendst, 2018-03-17:
If you keep the relay off it shouldn't toggle when power is restored. What happens at wifi restoration is currently your main problem as mine just wait for wifi to be available and connect again without toggling the relay.

I guess in your case the S20 experiences an exception causing the restart but I still cannot explain the settings loss.

arendst, 2018-03-17:
You could also set PowerOnState 0 so the relay will always be off after a power restore.

rcarmo, 2018-03-17:
Let's hope the exception doesn't toggle the relay, but yes, `PowerOnState 0` is a good idea.

arendst, 2018-03-17:
The exception will toggle the relay if it was on as it will restart the device and every restart will toggle the relay shortly due to SDK default GPIO state.

rcarmo, 2018-03-17:
OK, so that's likely what burned out the bulb. Is that fixable at all?

arendst, 2018-03-17:
Nope. Unless you build a device yourself and configure a relay the opposite way than itead did. That way an On relay will stay On but would toggle shortly to On if it was Off. You probably won't have that either...