arendst/Tasmota #2163: Unable to connect to web server

namurdock, 2018-03-14:
Setup:

- Sonoff Basic
- Noob issue... Tried entering Status 0 into the serial monitor in Arduino serial monitor, no status was shown, i see the activity on the board but no status results. 

Hi, I need help. I have flashed and re-flashed a new Sonoff Basic unit. Everything looks fine, it joins to my wlan, gets assigned an ip address via DHCP. and also communicates perfectly with my MQTT server.

However i am unable to connect to the web server. I have set the WIFI_CONFIG_TOOL to WIFI_MANAGER. This works fine, the config page is displayed and i can connect to my SSID.

I can ping the assigned IP from my router but not from machines on my network so this could be a router issue,

However still i cannot connect to the webserver.

Any help would be greatly appreciated.

Jason2866, 2018-03-15:
Which software Version? Downloaded self compiled? Pls give more infos

namurdock, 2018-03-15:
Apologies...

software version Sonoff-Tasmota-5.12.0, downloaded sourcecode.zip file from [https://github.com/arendst/Sonoff-Tasmota/releases](url)

Then compiled and uploaded via Arduino, as per [https://www.youtube.com/watch?v=c2IUIUnxjUY&t=841s](url)

Jason2866, 2018-03-15:
Why do you compile it? Flash the ready build one!
Compiling is usefull when you setup for your personal setting wifi, dns...

namurdock, 2018-03-15:
Doesn't clicking upload in Arduino IDE just compile anyway?

Surely you have to set some settings in the user_config file otherwise it will try to connect to the wrong said?

By the sounds of it I have missed something fundamental when reading through the wiki.

Jason2866, 2018-03-15:
You, can upload in Arduino Ide a ready made bin -> no compiling
So did you change something in sonoff_user_config.h or not?
There is no need to change something in this file for a successful build.
But when compiling the software by your own it is useful. With the setup for your wifi you have
a running device after flash and power up.

namurdock, 2018-03-19:
Ok, finally managed to get this done.

- Flashed with the ready made bin.
- Put the device into Wifi Manager mode.
- Connected it to my SSID
- wLan issues an ip address.

But still i cannot get into the web server to make any configuration changes.

Jason2866, 2018-03-19:
What happens when you enter the assigned ip address?

namurdock, 2018-03-19:
I get 

- ERR_ADDRESS_UNREACHABLE in Chrome.
- The connection has timed out in Firefox

I can ping the IP address but a scan on fing from a mobile device does not list it.

wayne47git, 2018-03-23:
Are you **sure** you have the sonoff.bin from the release section? I can not compile a version that will talk to Chrome or FIrefox (but I **can** talk to my compiled version with lynx). Once I install the downloadable sonoff.bin, things work fine.

namurdock, 2018-03-23:
Interesting that you can't use a regular browser. 

I tried easyesp as well and had the same experience. Flashes fine, but cannot connect to the web server.

Increasingly thinking it is my network that is the problem. 

I have read that these things are pretty sensitive to WiFi fluctuations.

WagnerBeccard, 2018-03-25:
Jea.. here too.. Reflashed from 5.10 to 5.12 with platform IO, after the module stopped working (double clicking every 20 to 40 seconds, thought maybe a out of memory)
It's got their ip in the router but the UI showed up bearly. Only when I restart with 6 button presses I come into the home UI, but no further 
ERR_ADDRESS_UNREACHABLE, or timeout afterwards.

jokerigno, 2018-07-22:
Same issue. Tried with 6.1, 5.14 and 5.13 sonoff.bin with esptools and arduino IDE. I can see my sonoff in my router map. But can't connect to it using IP. 

I also tried to ping it without success.

ascillato2, 2018-08-25:
Hi,

Have you managed to solve your issue?

Please, try erase all flash and try last version by flashing by wire.

If you still experience a problem, please, open a new issue with all the troubleshooting completed in order to properly help you.

Thanks

closed 2018-08-25 by ascillato2