arendst/Tasmota #2164: I cannot find Sonoff Touch after Wifi router change

vpomax, 2018-03-14:
My configuration is:
- Raspberry Pi 3
- Home Assistant 0.65.4 with Hassio

After I changed my wifi router I cannot find any of my seven Sonoff Touch switches flashed with Tasmota 5.12.0.
I cannot find them neither pinging their IPs... why???

vpomax, 2018-03-15:
OK, I must changed SSID and password to the new router and used the old SSID and password 
Four Sonoff Touch now are ok in wifi network, but other three Sonoff Touch are not visible in the wifi network... why?

Zixim, 2018-03-17:
i have had the same thing happen to me.
Have a tasmota sonoff connected to a wifi router with some ssid, take away that router, put in place another router with the same ssid... sometimes the sonoff will never connect to the wifi.

Only way out I have found : restart the sonoff in download mode, then restart it in normal mode, it should reconnect.

vpomax, 2018-03-17:
Hi Zixim
how do I reboot the sonoff in download mode?

Zixim, 2018-03-17:
https://github.com/arendst/Sonoff-Tasmota/wiki/Hardware-Preparation#bringing-the-module-in-flash-mode

vpomax, 2018-03-17:
ahhh... ok, I did it when I reflashed again
But today I do no see in my wifi network two sonoff... their blue light do not flashes... but I do not find  them connected to the router and if I call their web interface(e.g. http://191.168.1.90) they do not answer

Zixim, 2018-03-17:
you're going to have to physically get the 2 devices, restart them into flash mode, then restart them normally.
If they still don't show up, you will need to connect the serial port, and configure the wifi settings via the serial console.