arendst/Tasmota #2165: switch publishing 'stat' to different topics when 'cmnd' sent in different way

6laz3r, 2018-03-15:
hello, 2 days into tasmota and raspberry pi and MQTT and HA, so pretty limited knowledge on this. i have a sonoff basic with tasmota configured as generic module with the following settings (for this switch) in user_config:

#define PROJECT                "swiMainTV"       // Drawing Room TV

#define MQTT_FULLTOPIC         "drawingRoom/%prefix%/%topic%/"  // all devices in drawing room
//#define MQTT_FULLTOPIC         "livingRoom/%prefix%/%topic%/"   // all devices in living room

// %prefix% token options
#define SUB_PREFIX             "cmnd"
#define PUB_PREFIX             "stat"
#define PUB_PREFIX2            "tele"

// %topic% token options ---------- MQTT device topic / Group topic / client ID 
#define MQTT_TOPIC             "swiMainTV" //3 lines for Drawing Room TV
#define MQTT_GRPTOPIC          "switch"
#define MQTT_CLIENT_ID         "MainTV"

this is also configured in home-assistant as a MQTT switch. i am able to switch this on and off from HA, web interface and by publishing to the following topic

drawingRoom/cmnd/swiMainTV/POWER

however, the 'stat' being returned is on different topics. when cmnd is sent from HA or a mqtt client, the response topic is - 
'stat/MainTV/POWER'
and when the switch is toggled on web interface, response topic is - 
'drawingRoom/stat/swiMainTV/POWER'

this is the only switch doing this and i have tried compiling/flashing it multiple times. i have tasmota on a nodeMCU which always returns in the latter topic for all its connected relays.

what am i doing wrong? thanks

curzon01, 2018-03-15:
MQTT topic response is not as defined from `FullTopic` when your Client ID is a substring of the fulltopic - see also #1893

Actually suggestion for Client ID is leaving it as defined `DVES_%06X`
Note: The client ID is only an ID where the MQTT client registers on MQTT server - so more or less a technical ID - not neccessary for your HA to be human readable.

6laz3r, 2018-03-15:
thank you! this drove me crazy for 3 hours. though no practical issues due to this, just wanted to know why! thanks again.

closed 2018-03-15 by 6laz3r