arendst/Tasmota #2166: [Help wanted] Disable Sonoff 4ch pro RF

SaverioConigliaro, 2018-03-15:
Hey There,
I nave a tasmota flashed Sonoff 4ch pro connected to severamente house lights and i keep getting interference from the allarm wireless PIR. Every time i trigger the PIR by moving somewhere in front of It then my house lights switch on and off!!!
Since I don't Plan to use any remote Command... Is It possible to Disable radio (RF not wifi) interface?

arendst, 2018-03-15:
RF is not controlled by Tasmota.

- Unlearn the RF code from your PIR see itead website how to.
- Remove RF antenna
- Cut PCA circuit controlling RF signal

SaverioConigliaro, 2018-03-15:
Could you tell me where is the RF antenna in the SONOff 4 ch pro? Is this cutting going to influence wifi signal too?

SaverioConigliaro, 2018-03-15:
I didn't put an RF from the SONOff into my PIR!!!! This PIR is One of the components offline my allarm wireless system!!! PIR RF programming were performed by my allarm Central not by sonoff!!!! But when this PIR is triggered It trigger sonoff relè too!!!!

arendst, 2018-03-15:
It can only trigger the relay when Sonoff Pro has learned it. As Sonoff Pro learns as soon as you hold the button for a few seconds chances are you did it without knowing.

See wiki where to find the coiled RF antenna. Removing it has no consequences for wifi.

SaverioConigliaro, 2018-03-15:
You mean that while putting SONOff into the Wall i pressed the button without knowing It and It catched the code from the PIR sensor above my head???? LOL

SaverioConigliaro, 2018-03-15:
Last question: to replicate this behaviour (cause It could be usefull...) Do I have to trigger the PIR by moving in front of It while pressing the SONOff button?

arendst, 2018-03-15:
Depends. If the PIR sends RF every second then it's not needed but you probably have to, yes.

SecurityAddicted, 2018-03-15:
This can be useful to disable the RF part: https://github.com/arendst/Sonoff-Tasmota/issues/911#issuecomment-331694747

Snipercaine, 2018-06-10:
Opposite question:   I want to use the RF bridge to control the 4ch pro.  Is that possible?

ascillato, 2018-06-10:
Hi @Snipercaine ,

I think it should be possible. I don't have a 4Ch Pro to test it now, but the SOnOff Bridge can send any RF code and the SOnOff 4Ch Pro can learn any of these codes.

So, if you set your 4Ch in learning mode and you press any of the keys on the main page of your Sonoff Bridge, it should be working.

Besides, you can control the 4Ch from the Bridge using MQTT or directly (device to device) using RF or by websend command or by KNX.

Hope this helps.

BTW, Love your videos! :+1:

closed 2018-08-09 by stale[bot]