arendst/Tasmota #2167: Sonoff Dual R2 and rolling shutter position

B1G1, 2018-03-15:
Hi, i'm using Sonoff Dual R2 to automate my rolling shutters with Home Assistant. 
With my existing wall buttons i simply push the up button one time to open totally the rolling shutter, and if i want to stop it i have to push again one time the same button. If i push the down button while the rolling shutter is opening then it start to close. And so on...

My actual configuration of the sonoff is 
SwitchMode 3
PulseTime1 and PulseTime2 5
SetOption14 1 to prevent both relays from being activated at the same time.
seriallog 0
sleep 0
09 Switch 1 on GPIO0 and 10 Switch 2 on GPIO9

All it's working fine, the wall buttons too, but i would need a command that would pulse the relay on, initially, then pulse it again at the end of a variable time, to let me manage the rolling shutter position (or something like that), and that could work with the external buttons too, to get Home Assistant updated with the rolling shutter position.

How can i do that?

alesaund, 2018-03-15:
Same problem

curzon01, 2018-03-15:
Use [Backlog](https://github.com/arendst/Sonoff-Tasmota/wiki/Commands#using-backlog) feature:

e.g.

`Backlog Power1 ON;Delay 140;Power1 OFF`

switch Power1 On for 14 sec.

B1G1, 2018-03-15:
Great! Didn’t know about this Command, thank you.
But if i use the wall buttons how can i get from the sonoff the delay time, so that i can convert it in aperture percentage

curzon01, 2018-03-16:
Unfortunately the length a relay is switched on is reported only in sec resolution - and this is not very accurate to calc positioning. For shutter positioning you need the timing in higer resolution, otherwise the random seek will become inaccurate.
I did positioning calc for my roof window shutter controller (an Arduino with an [own written program](https://github.com/curzon01/FS20Uno)), but you realy need the runtime precise as possible for shutter runtime in both direction, than the positioning works very well.

For my other shutter - using Sonoff Dual R2 and rebuild FS20 RSU (replaced the 868MHz FS20 receiver with and ESP8266) switches - I simply has solved this (I'm currently using FHEM as Home Automation System) this way:
If position of shutter is unknown (e. g. "unknown" simply means "not open"), I power the shutter for open time for max. runtime. Then to position the shutter (e.g. shadowing) run the needed time to close.
Example close shadowing from unknown position:
`Backlog Power1 ON;Delay 185;Power1 OFF;Power2 ON;Delay 95;Power2 OFF`

Example close shadowing from open position:
`Backlog Power2 ON;Delay 95;Power2 OFF`

B1G1, 2018-03-16:
ok, if you don't know the position you just reset it opening the shutter totally. 

And what about using the physical buttons? I read in the wiki about the counter command, could i use that command to count opening time when i use the physical buttons

curzon01, 2018-03-16:
Counter<x> can be used but you need to connect the relay outputs GPIO to free GPIOs which are then configured as Counter<x>. This is the hardware solution.

A software solution is to add it to firmware - currently a counter can not be defined for an already use GPIO like GPIO for Relay1 or Relay2 to counter the time how long a relay is enabled.

B1G1, 2018-03-16:
So, I'm in the hands of arendst 😄

renne, 2018-03-19:
Shutter shafts are octagonal. An electrical contact activated by the edges would create 8 impulses per round. A second contact can reset the counter when the shutter is opened completely. :wink:

renne, 2018-03-20:
Glueing pieces of magnetic stripes on the shutter blades and connecting a hall sensor to a GPIO would allow to count the number of blades passing by without the need of exponential calculations.

B1G1, 2018-03-20:
this could be a good solution, but I would not like to add other components

pablitofernandez, 2018-05-18:
Hi @B1G1 , any updates with this issue? I'm very interested in a solution for your problem.

B1G1, 2018-05-18:
Hi @pablitofernandez, i think i will use a Xiaomi door/window sensor to be sure at least if the shutter is closed, then i could add some other magnet to know the position. I just need to find a good position for all these things

closed 2018-07-10 by ascillato2

thucar, 2018-11-07:
Would it be possible to add a roller shutter control with positioning to Tasmota using this library for isntance? https://github.com/marvinroger/arduino-shutters It relies on knowing the full travel time of the shutter and re-calibrates itself every time it is fully open or fully closed to maintain accuracy.