arendst/Tasmota #2168: MQTT temperature readings fluctuating to 85C

pmknowles, 2018-03-15:
I have a Sonoff TH with a DS18B20 sensor attached. I've updated to the 5.12.0e firmware but I keep on getting this kind of output 
08:20:40 MQT: stat/sonoff-bathroom/RESULT = {"POWER":"OFF"}
08:20:41 MQT: stat/sonoff-bathroom/POWER = OFF
08:20:48 MQT: tele/sonoff-bathroom/STATE = {"Time":"2018-03-15T08:20:48","Uptime":"0T00:00:14","Vcc":3.178,"POWER":"OFF","Wifi":{"AP":1,"SSId":"xxx","RSSI":88,"APMac":"E4:18:6B:xx:xx:xx"}}
08:20:48 MQT: tele/sonoff-bathroom/SENSOR = {"Time":"2018-03-15T08:20:48","DS18B20":{"Temperature":10.3},"TempUnit":"C"}
08:23:04 MQT: tele/sonoff-bathroom/STATE = {"Time":"2018-03-15T08:23:04","Uptime":"0T00:02:14","Vcc":3.178,"POWER":"OFF","Wifi":{"AP":1,"SSId":"xxx","RSSI":88,"APMac":"E4:18:6B:xx:xx:xx"}}
08:23:04 MQT: tele/sonoff-bathroom/SENSOR = {"Time":"2018-03-15T08:23:04","DS18B20":{"Temperature":85.0},"TempUnit":"C"}
08:25:09 MQT: tele/sonoff-bathroom/STATE = {"Time":"2018-03-15T08:25:09","Uptime":"0T00:04:14","Vcc":3.178,"POWER":"OFF","Wifi":{"AP":1,"SSId":"xxx","RSSI":88,"APMac":"E4:18:6B:xx:xx:xx"}}
08:25:09 MQT: tele/sonoff-bathroom/SENSOR = {"Time":"2018-03-15T08:25:09","DS18B20":{"Temperature":85.0},"TempUnit":"C"}
08:27:22 MQT: tele/sonoff-bathroom/STATE = {"Time":"2018-03-15T08:27:22","Uptime":"0T00:06:14","Vcc":3.178,"POWER":"OFF","Wifi":{"AP":1,"SSId":"xxx","RSSI":88,"APMac":"E4:18:6B:xx:xx:xx"}}
08:27:22 MQT: tele/sonoff-bathroom/SENSOR = {"Time":"2018-03-15T08:27:22","DS18B20":{"Temperature":85.0},"TempUnit":"C"}
08:29:28 MQT: tele/sonoff-bathroom/STATE = {"Time":"2018-03-15T08:29:28","Uptime":"0T00:08:14","Vcc":3.178,"POWER":"OFF","Wifi":{"AP":1,"SSId":"xxx","RSSI":90,"APMac":"E4:18:6B:xx:xx:xx"}}
08:29:28 MQT: tele/sonoff-bathroom/SENSOR = {"Time":"2018-03-15T08:29:28","DS18B20":{"Temperature":10.3},"TempUnit":"C"}
08:31:22 APP: Serial logging disabled
08:31:38 MQT: tele/sonoff-bathroom/STATE = {"Time":"2018-03-15T08:31:38","Uptime":"0T00:10:14","Vcc":3.178,"POWER":"OFF","Wifi":{"AP":1,"SSId":"xxx","RSSI":88,"APMac":"E4:18:6B:xx:xx:xx"}}
08:31:38 MQT: tele/sonoff-bathroom/SENSOR = {"Time":"2018-03-15T08:31:38","DS18B20":{"Temperature":10.3},"TempUnit":"C"}
When I go to view the Main Menu it briefly shows 85C before switching to the correct temperature.
Any ideas what could be going wrong?

Jason2866, 2018-03-15:
Resistor installed between data and Vcc on DS18B20?

arendst, 2018-03-15:
Make sure the DS18B20 connector has a firm connection in the Sonoff TH. Mine work OK.

pmknowles, 2018-03-15:
I powered it down (rather than restarting) and it’s been fine since.

Thanks

Phil

Frogmore42, 2018-03-16:
85 deg C is what you get when you query a DS18B20 before it has completed a measurement.  This can often be traced to a faulty connection.

pmknowles, 2018-05-25:
The problem has come back again. The reading on the webserver is steady until it's time to send an MQTT message. It briefly flashes to 85C then goes back to normal. That means openHAB only sees 85C. I've tried V5.14 classic and normal. I'm going to revert back to something like 5.10 to see if any better.

Frogmore42, 2018-05-25:
If you have a three wire connection, the only way this happens is due to a faulty connection.

pmknowles, 2018-05-25:
It’s a Sonoff DS18B20 plugged straight into the Sonoff. I downgraded to 5.10 (what is was on until I upgraded the firmware). It was fine so after an hour reflashed to 5.14 and it’s OK. Didn’t physically touch the device – all done OTA – maybe just temperamental.

closed 2018-05-25 by pmknowles

Frogmore42, 2018-05-25:
If you care about the results, you will need to figure this out.  If you look closely, I'll bet you will find the jack and/or plug are not making a solid connection.  I have a TH10/16 that had this problem.  I discovered that the jack was very loose.  It felt much looser than the other ones I had with the same plug, not that the ret felt very tight either.

I replaced the existing jack with a new better quality one and have not had a problem since then.  This is not a software issue.

Soundfaction, 2018-09-08:
It is definitely a poor connection. I tried everything and then finally settled on drilling  a 1.5mm hole in the end of the jack socket - insert the jack then screw a small 1.5mm screw (from an old 2.5inch HDD enclosure) to clamp the jack in place. Perfect results every time since.
[image]

Soundfaction, 2018-09-08:
I'm not using the cover - if you are - then drill a hole in the cover to allow you to screw it in once the cover is on or cut a slot to allow the cover to be put in place with the jack connected.

Frogmore42, 2018-09-08:
The Sonoff devices are cheap, in just about every meaning of the word.  The TH16 uses a very poor quality connector.  I have three in use.  One was so bad I had to replace it.  The others have been fine so far.  Quality control is not top of the list for iTead.