arendst/Tasmota #2169: Hardware supported?

robgardien, 2018-03-15:
Could I use the following hardware with the Sonoff-Tasmota software?

https://www.ebay.de/itm/WLAN-Schaltaktor-Unterputz-Montage-WLAN-switching-actuator-flush-mounting/222618241423?hash=item33d514c98f:g:fIsAAOSwDehZmH9k

Jason2866, 2018-03-15:
Yes

robgardien, 2018-03-15:
Thanks! Would you happen to know if there is a similar product with 2 relays?

Jason2866, 2018-03-15:
Sorry no. I said yes because i followed your link. One of the pictures is the schematic diagramm of the unit.
With a look in it, i am sure it will work with Tasmota.

curzon01, 2018-03-15:
Yes and it's supported by Tasmota (select module Luani)
https://luani.de/projekte/esp8266-hvio/

but you have to build it by yourself

robgardien, 2018-04-09:
Thank you for your help, would this one be supported by tasmota?

http://ehouse.biz/en/ewifir.html

Jason2866, 2018-04-09:
Sorry, i cant tell. There is no schematic diagram available.
Just try, if it is a ESP on the module it should/could work.