arendst/Tasmota #2171: Change reported name to router

Eno423, 2018-03-15:
Make sure these boxes are checked before submitting your issue - Thank you!

- [Etekcity voltson wifi outlet 
- [ 16:05:53 RSL: stat/sonoff/STATUS = {"Status":{"Module":41,"FriendlyName":"Haribo_power","Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
16:05:53 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://domus1:80/api/arduino/sonoff.ino.bin","Uptime":"0T00:02:34","Sleep":0,"BootCount":16,"SaveCount":60,"SaveAddress":"F8000"}}
16:05:53 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"5.12.0","BuildDateTime":"2018-03-08T15:25:20","Boot":31,"Core":"2_4_0","SDK":"2.1.0(deb1901)"}}
16:05:53 RSL: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"domus1","LogPort":514,"SSId1":"NOINTERNET","SSId2":"indebuurt2","TelePeriod":300,"SetOption":"00000009"}}
16:05:53 RSL: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":494,"Free":508,"Heap":22,"ProgramFlashSize":1024,"FlashSize":4096,"FlashMode":3}}
16:05:53 RSL: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"Haribo_power","IPAddress":"x.x.x.x","Gateway":"x.x.x.x","Subnetmask":"255.255.255.0","DNSServer":"x.x.x.x","Mac":"x:x:x:x:x:x","Webserver":2,"WifiConfig":3}}
16:05:53 RSL: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"x","MqttPort":x,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_8C437B","MqttUser":"DVES_USER","MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
16:05:53 RSL: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Thu Mar 15 15:05:53 2018","Local":"Thu Mar 15 16:05:53 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1}}
16:05:53 RSL: stat/sonoff/STATUS9 = {"StatusPTH":{"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
16:05:53 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2018-03-15T16:05:53","ENERGY":{"Total":0.049,"Yesterday":0.005,"Today":0.044,"Power":0,"Factor":0.00,"Voltage":0,"Current":0.000}}}
16:05:53 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2018-03-15T16:05:53","Uptime":"0T00:02:34","Vcc":3.427,"POWER":"OFF","Wifi":{"AP":1,"SSId":"xxxxx","RSSI":100,"APMac":"x:x:x:x:x:x"}}}
16:05:59 MQT: Attempting connection...] Provide the output of command ``status 0``-

I have set every where I can find, but my 2 outlets still report ESP_XXXXXX in my routers attached devices. If I go into the DHCP server I do see the friendly name I have set,

I would like to change whatever setting to change the name

I know that the numerical portion reported is the last four of the MAC, but cant figure out how to change it

arendst, 2018-03-15:
Wait 24 hours and it's probably solved.

Eno423, 2018-04-03:
i figured it out

closed 2018-04-03 by Eno423