arendst/Tasmota #2174: Z Cross detection for AC dimmer module?

Snipercaine, 2018-03-15:
I'd really like to integrate an AC dimmer like this: [AC Z-C PWM dimmer module](https://www.aliexpress.com/item/AC-Light-Dimmer-Module-for-PWM-control-1-Channel-3-3V-5V-logic-AC-50-60hz/32802025086.html?spm=2114.search0104.3.1.c4595357tKhVOX&ws_ab_test=searchweb0_0,searchweb201602_3_10152_10065_10151_10344_10068_10130_10324_10342_10547_10325_10343_10546_10340_10341_10548_10545_10696_10084_10083_10618_10307_5711215_10313_10059_10534_100031_10103_10627_10626_10624_10623_10622_5711315_10621_10620_10810_10811_5722415,searchweb201603_41,ppcSwitch_4&algo_expid=dc26ce29-c382-4e21-bd98-372034a9db8d-0&algo_pvid=dc26ce29-c382-4e21-bd98-372034a9db8d&priceBeautifyAB=0) with a Sonoff Basic. 

Looks like there's an output (Z-C) that needs to be interpreted to then tell the arduino when to cut the sinewave (PWM).  I know dimming is a much desired function in the Smart Home community.  Would something like this be possible with Tasmota?

Btw, you guys do great work! 
Thanks

curzon01, 2018-03-15:
See #2126 
Define a free GPIO as PWM, wire it to ctrl in and use PWM command to control it

smadds, 2018-03-15:
I would think hard before investing in those units, @Snipercaine. Some initial thoughts:-

- They are switching mains, so safety must be considered - e.g. the heatsink is probably at mains voltage
- They are triac (leading edge) dimmers, so not best suited to dimming LEDs. Unless you are only planning to dim tungsten lights it's probably better to look for MoSFET equipped trailing edge dimmer circuits
- They have no suppression on the board - this is a major part of the circuitry to prevent unwanted interference with other equipment
- You'll need to add circuit protection - fuses or polysnaps
- This may be picky - but only 2 mounting homes on the pcb - at the opposite end to the terminals & mains? Would be difficult to mount safely.

I hope itead do produce a dimmer soon though - it would be very useful.

The best (reasonably priced) alternative I have found for dimming mains LEDs is this: https://www.aliexpress.com/item/DMX303-3channels-high-voltage-dimmer-AC90-240V-Power-supply-3-channels-0-10V-output-signal-DMX/32684403650.html?spm=2114.search0104.8.63.2f0f23b8snZADR&priceBeautifyAB=0

I feed DMX from node-red using a udmx dongle. 

I realise this is a long way from your request, but sometimes it's helpful to consider other options!

Snipercaine, 2018-03-19:
That is very helpful info thanks.  I think I'll shelf the idea for now. 
It's kinda funny, we're in this sort of lighting limbo right now.  We have LED bulbs, but we are still running 110v to them individually.  
I close my eyes and dream of a perfect future where all our lights are programmable low voltage LEDs! 
What a wonderful world it would be.

cyberelectronics, 2018-03-19:
You can check this also 
https://www.tindie.com/products/nEXT_EVO1/ac-mains-dimmer-mpdmv75/

evolucja, 2018-03-22:
I really want to try this module. I se really no alternative to have a dimmalbe GU10 LED bulb from Tasmota..

cyberelectronics, 2018-03-22:
With the above module, the ZCD is already integrated with a separate sync/timer circuit onboard, so you just need to send out a PWM signal (from 0 to 100%, no need for ZC detection on the ESP side) or analog (0=On to 2.85V=Off) to control brightness/heat/fan speed in some cases (was designed especially for resistive loads). First you need to check out if your led bulb is dimmable.

evolucja, 2018-03-22:
> With the above module
I'm sorry I wasn't precise. I was talking about the unit from first post. 

MPDMv7.5 looks nice hovewer it's just expensive (I need to control 15 bulbs).

Would MPDMv4.1 be also okay? It's cheaper yet still more expensive than Aliexpress module..

cyberelectronics, 2018-03-22:
4.1 contains the same ZCD and sync/timer circuit (with external PWM / analog input), but there is no ESP or DAC circuit onboard.
For more info you can ask the designer Nextevo on tindie.

daenny, 2018-06-18:
I made a first try in implementing dimming on the MPDMv7.5 with DAC:
https://github.com/daenny/Sonoff-Tasmota
see also:
https://github.com/arendst/Sonoff-Tasmota/pull/3015
The dimming via the I2C DAC works. For my LED i needed to restrict the dimming range to a subrange of the possible outputs.

rt400, 2018-08-12:
hi to all..
this is what i looking for..
i want to connect RobotDyn Thyristor AC Dimmer to my sonoff t1 2ch. one of the channel is FAN .
any help will be useful for me..

ascillato2, 2018-08-26:
@rt400 

Tasmota is not real time software, so Tasmota can not manage zero-cross detection for that. You need extra electronics.

An Esp8266 compatible dimmer is MPDMv7.5 as @daenny explained. See #3015

closed 2018-08-26 by ascillato2

lucasdellasantina96, 2021-07-19:
SetOption99 | Enable zero-cross capable AC dimmer0 = no zero-cross AC dimmer connected (default)1 = zero-cross AC dimmer attached. Focus on raising edge and sync frequency