arendst/Tasmota #2176: hello everyone,

Ballistix321, 2018-03-15:
Make sure these boxes are checked before submitting your issue - Thank you!

- [ ] What hardware you are using
- [ ] Provide the output of command ``status 0``
- [ ] If you have a stack dump decode it: https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.rst
- [ ] For better debug messages: https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.rst

Ballistix321, 2018-03-15:
hello,

as the title states i have problems flashing and now even conncting my sonoff basics

after following the instructions in the video made by QuickPi i went on to upload on the sonoff.

the error i got was:

warning: espcomm_sync failed error:
espcomm_open failed error: 
espcomm_upload_mem failed *** [upload] Error -1

after this error the serial addapter disconnected. now every time i try to connect the sonoff it keeps connecting and disconnecting with random time intervalls. because of the disconnecting it is impossible to run any check.

as i am very very new to this, please, if i forgot some crucial information, ask me.

thank you.

smadds, 2018-03-17:
One possibility could be your programming adaptor having insufficient power to power up the Sonoff properly. There are times when the Sonoff takes more power (e.g. when transmitting WiFi), this can overload either the adaptor or USB port.

Have you tried a different USB port - e.g. a powered USB hub?

Other favourites include having the TX and RX the wrong way round, and not holding the button down at the instant of power-up.

Ballistix321, 2018-03-21:
this weekend im getting a different adapter so i can check.

also i've seen in the youtube videos that switching around the rx an tx leads may help, the problem is that after the first try with flashing, and getting the - 1 error, there was no way of getting a stable connection on any of the usbports.

if im not mistaken the tutorial showed that when the led is not blinking on the sonoff, it is in programming mode. so holding the button on power up should have worked.