arendst/Tasmota #2177: WiFi MAC address change

halacs, 2018-03-16:
Hi,

with the original firmware, I experienced that the wireless MAC address is changing after firmware upgrade. I don't know why, but it seems in case of SonOff MAC address is not a device specific, hardware coded parameter. This is a big problem for me because I use MAC address filtering on my wireless network.

I'm wondering if MAC address will also change when I upgrade my tasmata firmware or after I installed my first tasmata firmware, MAC address won't change anymore.

Somebody knows the answers:
- why MAC address was changed with original firmware?
- should I expect further MAC changing with tasmata firmware?

Thanks!

arendst, 2018-03-16:
Tasmota does not change MAC address unless the SDK is doing it but I, as the programmer of Tasmota, do not change it within the Tasmota code.

halacs, 2018-03-16:
Okay, thanks!
Do you have any idea what could be the reason behind the MAC change with original FW?

arendst, 2018-03-16:
Nope.

halacs, 2018-03-16:
Oks. Then have a nice day and congrats on you FW! bye

closed 2018-03-16 by halacs

simeonovp, 2020-08-07:
A have Tasmota 8.1.0 running on Sonoff S20. After power off on the grid the was the device resetted to AP mode.  I confgure WiFi again and click on the web button "Reset config" on the initial WiFi config page. After that was the device connected to the WiFI but its MAC was changed and the same as my Raspi server. Now I have MAC address conflict and no idea how to repai the MAC of the tasmota device

simeonovp, 2020-08-07:
I have an additional info. The info page displays the correct (old) MAC address, but the ethernet packets looked by Wireshark has set invalid MAC

prabbit237, 2023-04-25:
> Tasmota does not change MAC address unless the SDK is doing it but I, as the programmer of Tasmota, do not change it within the Tasmota code.

I understand that changing the MAC address is generally not a good idea for several reasons. But I have a couple of devices that quit working with Amazon Alexa and I'm thinking a MAC change may get Alexa to redetect them as new devices. I've tried everything from changing emulation to host name and deleted them from Alexa via Alexa's web interface, etc. so changing the MAC is like my "Hail Mary" play.

Is there something specific to Tasmota's code itself that you are aware of that would break by doing this or is your "do not change it" based on how it affects the device in regards to the outside world?