arendst/Tasmota #2178: Sonoff freeze

themask1987, 2018-03-16:
Make sure these boxes are checked before submitting your issue - Thank you!

- [x ] What hardware you are using
Sonoff Basic R2 V1.0 2017-10-11
- [x ] Provide the output of command ``status 0``
```
10:41:49 CMD: status 0
10:41:49 MQT: stat/sonoffsalaluce2/STATUS = {"Status":{"Module":1,"FriendlyName":"Sonoff","Topic":"sonoffsalaluce2","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":0,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
10:41:49 MQT: stat/sonoffsalaluce2/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://domus1:80/api/arduino/sonoffsalaluce2.ino.bin","Uptime":"0T00:15:38","Sleep":0,"BootCount":18,"SaveCount":59,"SaveAddress":"F9000"}}
10:41:49 MQT: stat/sonoffsalaluce2/STATUS2 = {"StatusFWR":{"Version":"5.12.0","BuildDateTime":"2018-03-15T17:54:58","Boot":6,"Core":"2_4_0","SDK":"2.1.0(deb1901)"}}
10:41:49 MQT: stat/sonoffsalaluce2/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"domus1","LogPort":514,"SSId1":"FRITZ!Box","SSId2":"indebuurt2","TelePeriod":300,"SetOption":"00000009"}}
10:41:49 MQT: stat/sonoffsalaluce2/STATUS4 = {"StatusMEM":{"ProgramSize":488,"Free":512,"Heap":18,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3}}
10:41:49 MQT: stat/sonoffsalaluce2/STATUS5 = {"StatusNET":{"Hostname":"sonoffsalaluce2-5252","IPAddress":"192.168.0.101","Gateway":"192.168.0.10","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.10","Mac":"EC:FA:BC:87:34:84","Webserver":2,"WifiConfig":3}}
10:41:49 MQT: stat/sonoffsalaluce2/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.150","MqttPort":1883,"MqttClientMask":"DVES-873484","MqttClient":"DVES-873484","MqttUser":"openhabian","MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
10:41:49 MQT: stat/sonoffsalaluce2/STATUS7 = {"StatusTIM":{"UTC":"Fri Mar 16 09:41:49 2018","Local":"Fri Mar 16 10:41:49 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1}}
10:41:49 MQT: stat/sonoffsalaluce2/STATUS10 = {"StatusSNS":{"Time":"2018-03-16T10:41:49","Switch1":"ON"}}
10:41:49 MQT: stat/sonoffsalaluce2/STATUS11 = {"StatusSTS":{"Time":"2018-03-16T10:41:49","Uptime":"0T00:15:38","Vcc":3.486,"POWER":"OFF","Wifi":{"AP":1,"SSId":"FRITZ!Box","RSSI":92,"APMac":"C8:0E:14:24:62:69"}}}
```
Hello,
I have a "freeze" problem with sonoff basic.
I had downloaded the zip file with last release (5.12.0) and customized few params (VS Code):
- Wifi SSID and password
- MQTT params in order to connect to my RPI3 with openhabian
- Put comment on USE_DOMOTICZ && USE_HOME_ASSISTANT

The sonoff immediately after flash works perfectly but after few hours without using it (no physical button, no command ...) is no more possible to reach it via ping, web interface and MQTT.
When I come in this situation if I try to use the physical button on soff (the standard one or the GPIO14), if I'm lucky, after few minutes the situation become normal and I can do everything (web interface command, MQTT).
During the freeze time if I try to ping it (ping -t x.x.x.x) I see just 30/40% of correct reply from the device.
I also tried to use old version of firmware but the situation is the same.
Another strange thing (for my point of view), in consolle I see this
```
10:45:35 MQT: Attempting connection...
10:45:40 MQT: Connect failed to 192.168.0.150:1883, rc -2. Retry in 10 sec
10:45:51 MQT: Attempting connection...
10:45:56 MQT: Connect failed to 192.168.0.150:1883, rc -2. Retry in 10 sec
10:46:07 MQT: Attempting connection...
10:46:12 MQT: Connect failed to 192.168.0.150:1883, rc -2. Retry in 10 sec
10:46:23 MQT: Attempting connection...
10:46:28 MQT: Connect failed to 192.168.0.150:1883, rc -2. Retry in 10 sec
10:46:39 MQT: Attempting connection...
10:46:44 MQT: Connect failed to 192.168.0.150:1883, rc -2. Retry in 10 sec
10:46:55 MQT: Attempting connection...
10:46:58 MQT: Connected
```
Seems like, when I lost the connection (but I don't know why) the MQTT try to restore it consuming all the resource of the sonoff because when the connection is restablished the system became to the normal condition.
So, any idea or test to check why the sonoff is working in this way?

Thanks

arendst, 2018-03-16:
While trying to connect to MQTT server all comms is stalled for some seconds by SDK. As the retry interval is default set to 10 seconds chances are you will hardly get to the device.

Solutions are:
- Get a stable MQTT environment
- Increase mqtt retry parameter (See wiki - commands)
- Change default PubSubClient with alternative MQTT client as introduced in development branch.

themask1987, 2018-03-16:
Ok first thanks for your support.
I've changed the retry to 60 and after to 600 but no change.
For the other 2 points, sorry but I'm completly new on this world.
If I understand correctly, PubSubClient folder is under root\lib\. Now I have v 2.6.09, which version do you suggest? And also, in order to take another version I need to download the complete zip of another firmware release and substituite to mine?
For the first point, are you talking about the mosquito enviroment on raspberry or I'm completly out of topic?

arendst, 2018-03-16:
With stable I mean the mqtt environment is accepting requests from tasmota and not connections failed due to -2. See wiki - for decoding -2 https://github.com/arendst/Sonoff-Tasmota/wiki/MQTT-Overview

DOwnload the development version form github and do the same as you did with the master version. It contains all the libs needed.

In user_config.h you select another library and re-compile with it. It's all in the package.

themask1987, 2018-03-16:
Ok I've download the dev versione and:
using MQTT_LIBRARY_TYPE      2 -> general failure. No web server, ping not response, MQTT not working.
using MQTT_LIBRARY_TYPE      1/3-> same situation as before. Most of time not reachable

arendst, 2018-03-16:
Great. 

What do you use to compile and upload?

Next step would be to completely erase the flash (See wiki) and re-flash it.

themask1987, 2018-03-16:
I'm using Visual Studio Code with extension PlatformIO IDE.
In the wiki is mentioned Esptool for erase flash, is also possible with VS?

arendst, 2018-03-16:
VSC and platformIO is OK.

I'm glad you find the answers so quick (honestly) 

I've never used VS for that. I wouldn't know how to select the correct options.

themask1987, 2018-03-19:
Ok i've done.
I have erase the flash with esptool and reupload firmware using esptool/vscode (I've done this procedure several times) but I have the same situation: the connection is closed very often.
The only difference now is the error code, 8 instead of -2
18:43:01 MQT: Connect failed to 192.168.0.150:1883, rc 8. Retry in 10 sec
18:43:11 MQT: Attempting connection...
18:43:22 MQT: Attempting connection...
18:43:25 MQT: Connected
Unfortunatelly on wiki there is no definition for error 8 but aniway what I think is intersting is to understand why so often the connection is close and on wich side is closed (sonoff o openhabian).
Any other suggest?

brnnrc, 2018-03-20:
Hi, I'm facing similar problem with a Sonoff basic (no problem with POW and S20 on equal configuration). It happens after a few seconds, not hours instead.
I have many ping lost booting with webserver and mqtt on. I get some MQT: Attempting connection... too.
No signal issue.

1) When I boot without webserver, I have no ping and mqtt connection loss.

2) Here the strange thing: disabling
//#define USE_ADC_VCC
I have no loss even with webserver on!

Try and report those points, please.