arendst/Tasmota #2179: Webserver admin/client

Katzechrisu, 2018-03-16:
Hello, i have set the webserver to client over console with webserver 1

now i can not come back to webserver admin. i have no console on the page and over http i dont now the right string to turn it back.

how can helo me?

curzon01, 2018-03-16:
Make sure these boxes are checked before submitting your issue - Thank you!

- [ ] What hardware you are using

arendst, 2018-03-16:
Use serial or mqtt to switch webserver to admin. See wiki - commands

Katzechrisu, 2018-03-16:
And another question?
Give it a Logik Diagramm that Show me what happends with Switches and Relais and how to Connect it on wemos? 
I have 4 Switches But 8 Relays

Katzechrisu, 2018-03-16:
For seriell Commands i can use seriellmonitor in arduino?

Zixim, 2018-03-16:
i use Termite : https://www.compuphase.com/software_termite.htm
on windows.
Do not use Putty for serial commands, it won't work.

Katzechrisu, 2018-03-16:
i have testet with arduino but it dont work.
can anybody show me the command string to take the webserver to admin back?

curzon01, 2018-03-16:
Wemos D1 is your hardware.

There are several ways to reset it:

1. If you use Sonoff or has a button on GPIO, you can simply use [Button usage](https://github.com/arendst/Sonoff-Tasmota/wiki/Button-usage) "Pressing the button for over four seconds" Reset settings to defaults
2. Use serial monitor and type command `WebServer 2` as suggested

Katzechrisu, 2018-03-16:
I do not want to reset it.
i have test it with WebServer 2 at seriell monitor but it dont works.
i have set a User and Webpasword is that the problem?

Katzechrisu, 2018-03-16:
I can  not make a wire to GPIO0 because i have no Button declariert in the menu befor start the client mode

Zixim, 2018-03-16:
are you connected to the serial port ?
If yes, try a serial console, and give it "restart 1" , it should just reboot.

Katzechrisu, 2018-03-16:
i see it in seriell monitor at arduino but it not takes the comand.
I have send restart 1
and webserver 2
but nothing happends

Zixim, 2018-03-16:
I do not know if you can issue serial commands from the serial monotir...i think it is just a monitor.
For serial commands, I use Termite : https://www.compuphase.com/software_termite.htm
on windows.
PS: Do not use Putty for serial commands, it won't work.

Katzechrisu, 2018-03-16:
Ok i have now use termite but its the same in see the modul but it dont take a comand

Zixim, 2018-03-16:
then...what is the output inside the serial console ?

Katzechrisu, 2018-03-16:
I see if i klick a Button in webinterface. For example power4 = on

Zixim, 2018-03-16:
weren't you connected with Termite, using the serial port ?
What is displayed in the Termite window?

Katzechrisu, 2018-03-16:
But if i write restart 1 it make nothing

Katzechrisu, 2018-03-16:
It Connect to wifi and Sent Status mails

Zixim, 2018-03-16:
Please, connect to the serial port on the motherboard of the device?

Katzechrisu, 2018-03-16:
Seriellog 2 is it default in Software? If have See it in another Forum?
Must it stay on 2?

Katzechrisu, 2018-03-16:
Motherboard of device?

Zixim, 2018-03-16:
leave the defaults, and just connect the device to your USB port of the computer.

curzon01, 2018-03-16:
Serial interface is able to accept command.
If the command is not accepted the reason might simply that it is not terminated.
Try using Ctrl-J and/or Ctlr-M instead of hitting enter/return key if enter is not accepted. Otherwise check your terminal program documentation how to send LF to console...

Zixim, 2018-03-16:
@curzon01 i don't get the feeling that the Serial port is connected yet.

Katzechrisu, 2018-03-16:
Ah of Computer motherboard direktly? Not on the Front USB ? Or on seriell Port with seriell Kabel? 

I have make the Software also over the Front USB without Problem

Zixim, 2018-03-16:
front USB <---> Wemos

Zixim, 2018-03-16:
then open Termite
ans give us what you see onscreen.

Zixim, 2018-03-16:
it should look smt like this : 

[image]

Katzechrisu, 2018-03-16:
Ctrl-J and/or Ctlr-M dont work make only bing on the monitor
[image]

Katzechrisu, 2018-03-16:
[image]

curzon01, 2018-03-16:
Open Termite settings and set "Transmitted text" to "Append LF":

Default might be "Append nothing" - this can not work:
[image]

In that case, the text "WebServer 2" is send to your Wemos but the command is not terminated, because if you hit Return, "Append nothing" is send - this means nothing is send.

Katzechrisu, 2018-03-16:
it was on Append LF and don´t work

Zixim, 2018-03-16:
@curzon01 is it possible that serial commands are turned off? is that even an option ??

curzon01, 2018-03-16:
@Zixim it's enabled at all.
@Katzechrisu be a little brave and simply try the other  possibilities CR and CR/LF - nothing will implode/explode

Zixim, 2018-03-16:
mine is set to "append LF", but trying others will not hurt anything.

Katzechrisu, 2018-03-16:
tomorrow i can test it.
i will test also with an other USB Kabel.
I can not send anything over the seriell conection. only recive.
But the flashing goes fine.

Zixim, 2018-03-17:
this is unusual, because flashing = sending.

Katzechrisu, 2018-03-17:
I have test. It dont`t work good.
I make a new flash on modul by serial. It takes the config from userconfig.h and works OK. In the config.h i set a WebClient"Katzechrisu" and a Password"12345678"
All works good. Then I write into Console WebServer 1. It change to Client...Good....But what Comand i must write to makes it back to Admin, and where i have to do it. The Console is not there(normaly).
Over serial it wont work. What Resolution i have to make it back?

Must i only write "WebServer 2" on Serial Programm)
Or can you say me the comand for MQTT. 

I have spending a lot of time for this an can´t  go forward with this Problem. 
Please help.

Zixim, 2018-03-18:
you need to start over and pay attention to all that you are doing.
Without the serial console, (Termite) you will be stuck.

1. flash the device again, from the beginning, do not change ANY config settings.
2. now is the time to figure out the serial console... close the software that you used for flashing, but leave the device connected to USB/serial.
3. open Termite, it should show approximately the same output as your last screenshot.
4. type STATUS 0 in the bottom line of Termite and click the enter button at the end of the line.
5. what appears on screen ?
6. do NOT go any further with configuring the device until steps 3 & 4 & 5 give the desired output !

CreasolTech, 2020-05-29:
From linux, I type the command
`minicom -b 115200 -D /dev/ttyUSB0`
to access the ESP8266 with Tasmota, connected to device /dev/ttyUSB0 at 115200baud , 
and type 
webserver 2 CTRL-j
to return to default admin web access (I type CTRL+j buttons instead of ENTER button)
It works like a charm.
Paolo