arendst/Tasmota #2182: Serial data feature suggestion

species5618, 2018-03-17:
I have a use case where I would like to replace and old serial data radio with and esp8266 device using mqtt
It is not contious data stream, just a simple 8-16 chars every 60 seconds of so. So mqtt would be spot on

So my thinking is add support for serial comms in the generic device. Or any device where serial port is easily available

So any text reecived form rx port (with cr on end or Confiurable timeout)
Is forwarded over mqtt 
And vice versa.

Baud rate would need to be configurable. 
And probably disable boot messages

Zixim, 2018-03-17:
try google?

"mqtt serial bridge esp8266"

species5618, 2018-03-17:
i have, and built a working protype, however it would be nice for all my iot devices to run the same core software

Zixim, 2018-03-17:
do a pull request ?

arendst, 2018-03-17:
Considering but see opportunities in regards to using the hardware serial for debugging...

I might choose to only support software serial but this needs testing.

OTrumpf, 2018-03-17:
+1 for a serial connection in tasmota!

My use-case: gesture-detection with a dedicate Arduino + several sensors (e.g. IR-/Ultrasonic-sensors) with time critical functions and a serial connection to a tasmota-esp8266 for a bi-dicrectional exchange of MQTT commands.

Perhaps you can define a set of 5 - 10 MQTT-commands in the web interface which are transferred to the Arduino and pass every command from the Arduino to the MQTT-Broker. Example:

Write to Arduino ("IR_SENSOR" is defined in Web interafce):
"IR_SENSOR" --> cmnd/MyDevice/IR_SENSOR:text where "text" ist transferred to the Arduino as "IR_SENSOR:text"

Read from Arduino:
"USM_DISTANCE" --> tele?/MyDevice/USM_DISTANCE:100cm where "USM_DISTANCE:100cm" is transferred from the Arduino.

Software Serial would be fine for me and the text stream of the OP could be integrated via an optional default MQTT-command for received serial text.

arendst, 2018-03-18:
Give it a go!

Use command ``SerialSend`` to send a text string terminated by tasmota with a ``\n`` (linefeed)
Use command ``SerialSend2`` to send a text string without termination
Above command also disables serialloging

Expect responses on mqtt ``SerialReceived`` as a text string

Change baudrate with command ``Baudrate``. This also impacts serialloging if/when enabled.

danielztolnai, 2018-03-18:
I needed a similar solution for some time to communicate with a projector so I gave it a try and created a software serial implementation (#2190). I hope it is relevant here.

species5618, 2018-03-18:
wow,

compiled using platform io into a ESP-01 , device config set to generic, ,with a DS18B20 on GPIO2
and it just works 

still reads temperature and does the Serial port stuff , note the device i am connected to echos a GOOD Message, 

/local/Clock/cmnd/Baudrate : 57600
/local/Clock/cmnd/SerialSend2 :aCLSD-0.8------
/local/Clock/stat/RESULT : {"SerialSend":"Done"}
/local/Clock/tele/RESULT : {"SerialReceived":"aCLSD-0.8---"}
/local/Clock/tele/SENSOR : {"Time":"2018-03-18T15:32:04","DS18B20":{"Temperature":22.1},"TempUnit":"C"}

fantastic ..

danielztolnai, 2018-03-18:
I've added support for the transmission of whitespace characters in #2190 when they are escaped in the MQTT message. Also implemented a "delimiter character" functionality, so a single message will be transmitted through a single MQTT message.

charlietomo, 2018-04-05:
This looks great but I'm struggling to get things to work. Anything obvious wrong?

Wemos D1 running `5.12.0i` with module set as Generic. Configuration:
TX GPIO1 Serial Out = 67 SerBr Tx
RX GPIO3 Serial In = 68 SerBr Rx

TX/RX and 3.3v are connected to a TTL-RS232 adaptor ([like this](https://www.ebay.com/itm/RS232-To-TTL-Converter-Module-Serial-Module-DB9-Connector-3-3V-5-5V-Arduino-nb-/272451026993)) which is connected to my RS232 device.

I want to send command to the device (MQTT > Serial) and I also want the messages from RS232 to be delivered over MQTT (regardless of commands). 

If I plug the RS232 device into a Raspberry pi with USB-RS232 adaptor I can confirm the device is sending output/takes commands.

I'm not seeing anything related to serial on the Tasmota console. Do I have to do anything specific to set things up?

closed 2018-07-02 by arendst