arendst/Tasmota #2184: stat/XX/RESULT ON/OFF different state then stat/XX/POWER

M4RiO32s, 2018-03-17:
Make sure these boxes are checked before submitting your issue - Thank you!

- [x] What hardware you are using: MagicHome / H801
- [x] Provide the output of command ``status 0``: [log.txt](https://github.com/arendst/Sonoff-Tasmota/files/1821512/log.txt)

Instead of using preset profiles i am using Tasmota with generic module setting and manually puting definitions of every gpio and i use SetOption15 1 that lets me use way more options (button at gpio0 ie.) while having color/dimmer settings
Everything is working great and BIG respect to Theo for making that firmware.
I did notice however small issue and would like to ask if it should be working that way:

when i press physical button, use toggle on web interface or publish message ON/OFF all is as it supposed to be however if module is OFF and i use command of dimmer changing its vale module turns on - which is fine publishing its state as ON in result. in my case:
stat/h801_2/RESULT {"POWER":"ON","Dimmer":73,"Color":"BABABA"}

BUT NOT IN 
stat/h801_2/POWER which remains OFF
since this is topic i use to simply monitor ON/OFF state it messes a bit with my configuration.

Is it glitch or done on purpose? Please advice.
M.

Zixim, 2018-03-17:
"and i use command of dimmer changing its vale module turns on"
where are you setting the dimmer value ? with an mqtt command? at web console? in the webif ?

For me, with mqtt (+ Home Assistant) selecting a dimmer value does set stat/h801/POWER to ON
But, I have configured Tasmota as a H801 module.

Jason2866, 2018-03-17:
SetOption20  1 / on     Enable changing Dimmer value without turning power on

M4RiO32s, 2018-03-18:
"where are you setting the dimmer value ? with an mqtt command? at web console? in the webif ?"

web interface of module, web command (GET) and mqtt - doesn't matter which one i use

Result with json says POWER:ON
stat/xx/POWER stays off