arendst/Tasmota #2186: SetOption13 Enable button single and hold only for immediate response

dragouf, 2018-03-17:
I was using SetOption13 and I was wondering why there is not a mode to enable only hold and single press to be able to have immediate response and still have long press.

I mean, I think we can achieve that since a single press is an immediate release and can be detect like that. So if when press there is an immediate release action is triggered.

ionciubotaru, 2018-03-18:
Immediate response is possible for one option only.
For more than one, the software must wait to differentiate between  them.

dragouf, 2018-03-18:
Yes but it is what I’m trying to explain. It’s easy to distinguish between a short press and a long press with immediate response if there is no multi press. And it should be nice to be able’ to have that behavior for example with option13 set to 3. 

For that just need to take immediate action for single press when button is released if time between push down and push up was less than a long press.

Or I missed something ?

Leseratte10, 2018-03-29:
I'd really like having this feature, too. I don't need fancy multi-button press stuff that delays switching, but I would really like an option where the sonoff switches when I let go of the button and the button press time was short enough. If it was longer, it could send a specific MQTT message.

TheCellMC, 2018-04-17:
I would also like this feature to be added. I would prefer to have instant on/off instead of having a delay for this multi-button press functionality.

yanoosh, 2019-03-28:
need +1

emhal, 2019-03-30:
need +1