arendst/Tasmota #2187: HELP - SONOFF Dual R2 restarts and opens rolling shutter

B1G1, 2018-03-18:
hi, a little while ago my object has opened the shutter to which it is connected, without any command or automation. It looks like it has restarted.

This is the log
`00:00:00 Project sonoff tapparelleStudioSU (Topic tapparelleStudio, Fallback DVES_9E8FDF, GroupTopic sonoffs) Version 5.12.0
00:00:00 WIF: Connecting to AP2 indebuurt2 in mode 11N as tapparelleStudio-4063...
00:00:07 WIF: Connect failed as AP cannot be reached
00:00:07 WIF: Connecting to AP1 FRITZ2.4 in mode 11N as tapparelleStudio-4063...
00:00:12 WIF: Connected
00:00:12 DNS: Initialized
00:00:12 HTP: Web server active on tapparelleStudio-4063.local with IP address 192.168.1.49
09:03:18 MQT: Attempting connection...
09:03:18 MQT: Connected
09:03:18 MQT: tele/tapparelleStudio/LWT = Online (retained)
09:03:18 MQT: cmnd/tapparelleStudio/POWER = 
09:03:18 MQT: tele/tapparelleStudio/INFO1 = {"Module":"Sonoff Dual R2","Version":"5.12.0","FallbackTopic":"DVES_9E8FDF","GroupTopic":"sonoffs"}
09:03:18 MQT: tele/tapparelleStudio/INFO2 = {"WebServerMode":"Admin","Hostname":"tapparelleStudio-4063","IPAddress":"192.168.1.49"}
09:03:18 MQT: tele/tapparelleStudio/INFO3 = {"RestartReason":"Software/System restart"}
09:03:18 MQT: homeassistant/switch/tapparelleStudio_1/config =  (retained)
09:03:18 MQT: homeassistant/switch/tapparelleStudio_2/config =  (retained)
09:03:18 MQT: stat/tapparelleStudio/RESULT = {"POWER1":"ON"}
09:03:18 MQT: stat/tapparelleStudio/POWER1 = ON (retained)
09:03:18 MQT: stat/tapparelleStudio/RESULT = {"Command":"Unknown"}
09:03:18 MQT: stat/tapparelleStudio/RESULT = {"POWER1":"OFF"}
09:03:18 MQT: stat/tapparelleStudio/POWER1 = OFF (retained)
09:03:19 MQT: stat/tapparelleStudio/RESULT = {"POWER1":"OFF"}
09:03:19 MQT: stat/tapparelleStudio/POWER1 = OFF (retained)
09:03:19 MQT: stat/tapparelleStudio/RESULT = {"POWER2":"OFF"}
09:03:19 MQT: stat/tapparelleStudio/POWER2 = OFF (retained)
09:03:26 MQT: tele/tapparelleStudio/STATE = {"Time":"2018-03-18T09:03:26","Uptime":"0T00:00:23","Vcc":3.438,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"FRITZ2.4","RSSI":100,"APMac":"C8:0E:14:1E:C4:9C"}}
09:03:26 MQT: tele/tapparelleStudio/SENSOR = {"Time":"2018-03-18T09:03:26","Switch1":"ON","Switch2":"ON"}
09:08:26 MQT: tele/tapparelleStudio/STATE = {"Time":"2018-03-18T09:08:26","Uptime":"0T00:05:23","Vcc":3.434,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"FRITZ2.4","RSSI":100,"APMac":"C8:0E:14:1E:C4:9C"}}
09:08:26 MQT: tele/tapparelleStudio/SENSOR = {"Time":"2018-03-18T09:08:26","Switch1":"ON","Switch2":"ON"}
09:13:26 MQT: tele/tapparelleStudio/STATE = {"Time":"2018-03-18T09:13:26","Uptime":"0T00:10:23","Vcc":3.438,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"FRITZ2.4","RSSI":100,"APMac":"C8:0E:14:1E:C4:9C"}}
09:13:26 MQT: tele/tapparelleStudio/SENSOR = {"Time":"2018-03-18T09:13:26","Switch1":"ON","Switch2":"ON"}
09:18:26 MQT: tele/tapparelleStudio/STATE = {"Time":"2018-03-18T09:18:26","Uptime":"0T00:15:23","Vcc":3.436,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"FRITZ2.4","RSSI":100,"APMac":"C8:0E:14:1E:C4:9C"}}
09:18:26 MQT: tele/tapparelleStudio/SENSOR = {"Time":"2018-03-18T09:18:26","Switch1":"ON","Switch2":"ON"}`

my wifi is scheduled to automatically disable from 2am to 9am on Sunday 
What's the problem? How can I prevent this behavior?

Jason2866, 2018-03-18:
Diasbling wifi can generate unknow side effects. For test leave your wifi always on.
If error is gone, thats the reason....

B1G1, 2018-03-18:
i made some tests and set wificonfig 5 in console, but everytime the sonoff restarts, even if I restart it manually with restart 1 in console, when the sonoff connects to the mqtt server it seems it send a command to open the shutter.

B1G1, 2018-03-18:
restoring default configuration solved the problem.

closed 2018-03-18 by B1G1