arendst/Tasmota #2189: "WeConn" Hardware

n3m3s1s, 2018-03-18:
Hi everyone,
i got some Smart Plugs that look exactly the same but by another vendor. As they come as "M.Way" or "Jinvoo" or "WeConn", all the 3 types seem to have the same hardware and be using the same software. And they look exactly the same as the sonoff S20 (at least from the outside).
The main chip also seems to be an ESP8266 (but some form I have not seen in my projects so far).
Do you think your firmware could run these devices as well? I attached some pictures from outside and inside. Maybe it really is the same hardware.
Thanks
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

arendst, 2018-03-18:
There is so much logic it looks like it has power monitoring also. Without schematics or reverse engineering it will be a challenge.

Perhaps someone is eager to find out.

Jason2866, 2018-03-18:
Do you have a link where you bought it?

n3m3s1s, 2018-03-18:
Not sure, it was a gift. But I am pretty confident its those: [Amazon](https://www.amazon.de/dp/B0792TGLGZ/ref=sspa_dk_detail_5?psc=1)
Anyways I can provide more information from the package if necessary. It shows the model name as "SU02". Other than that it really does not show much info. Not even a vendor.
And @arendst : Yes. Unfortenately it has power monitoring, I am afraid.

Jason2866, 2018-03-18:
If power metering is a feature you dont mind if you loose that you could try tasmota.
If it doesnt work you can go back to stock firmware. With esptool you can save the stock firmware.
Do this before you flash tasmota :-)...

n3m3s1s, 2018-04-09:
Thank you Jason. I have a little bit of a problem now and maybe you can help again:
I meanwhile bought a sonoff device as well. I connected it to my FTDI (ftdi232) and tried using esptool on both my linux and my windows machine. Both failed with this message:
```
Connecting........_____....._____....._____....._____....._____....._____....._____....._____....._____....._____
A fatal error occurred: Failed to connect to Espressif device: Timed out waiting for packet header
```
I got a little impatient and just tried to flash it anyways using the windows box and atom. It worked like charm. I have no idea how.
So now back to the "WeConn" thing. In this case (since nobody knows if it works) I really want to dump the firmware first so I dont brick the device. But I have the same problem. If I change the baudrate to 9600 it still gives me timeouts. With 57600 baud it does:
```
Connecting........_____....._____....._____....._____....._____....._____....._____....._____....._____....._____
A fatal error occurred: Failed to connect to Espressif device: Invalid head of packet (b'\xfc')
```
All different types of baud rates give me gibberish in the arduino serial monitor:
`⸮`
So how could I have gotten it wrong?

Jason2866, 2018-04-09:
For backup under Windows try (you have to change the COM Port):
esptool --port COM5 --baud 115200 read_flash 0x000000 0x100000 test-backup.bin
for flash (COM Port, firmware file has to be in same folder in this example) :
esptool --port COM5 --baud 115200 write_flash -fs 1MB -fm dout 0x000000 firmware.bin

Jason2866, 2018-04-09:
Have you read this
https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting
especially the part over fake FTDI Devices (failing in flashing)

n3m3s1s, 2018-04-09:
I read it but i trief both ways of wiring.
I also noticed that the LED still flashes no matter if I turn it on normally or while pressing the button. So maybe I'm not getting into flash mode. Also as you can see in picture it has another breakout that says "DOWNLD". Is that something i moght need?

n3m3s1s, 2018-04-09:
Oh and btw I made an interesting observation: I wired the FTDI to the correctly flashed sonoff device again and it got me the perfect working logs on the serial monitor. At 115200 baud it is working perfectly. 
But if I call the same esptool command on that port with that baud rate it again tells me there failed to connect.

Jason2866, 2018-04-10:
That is a know behaviour of fake FTDI Adapters. The are not working below a speed of 115200 baud!!!
Try one with a ch340 or a ch341 chip. Or a genuine FTDI (if you get one...)

n3m3s1s, 2018-04-14:
Ok I experimented a lot more and used an Arduino to flash some S20 wich worked well.
Maybe someone could give me some more hints or ideas you have on how to get the device on the pictures into flash modes. Since Everything using esptool failed (but also on other devices including using an ch340 in combination with tasmota on s20) I decided to just flash it and see if it works.
You can see several contact spots on the surface of the chip as well as the "downld" pin. I tried every one of those on GND during chip boot but it keeps not going into flash mode. The button of course does not work either.
Do you have any suggestions?

xMOVx, 2018-05-02:
I have the same device and the same problem as you.

Tried several different wirings and contact spots to get it into flash mode. Maybe someone finds a suitable method...

n3m3s1s, 2018-05-27:
@xMOVx did you find a solution so far?

redhoodie, 2018-07-05:
I'm so keen for this. Let me know if someone is keen to get this done but needs a Weconn device; I can send one.

n3m3s1s, 2018-07-06:
@redhoodie I already tried it. It is a little hard indeed to short that pin properly without unsoldering the whole chip but I could not get it into flash mode. (If it should be in flash mode - and the cause might be my fat fingers - please tell me and I will try again.)
Anyways if someone could find a way to get it into flash mode I will happily try to get it working.

n3m3s1s, 2018-07-07:
Ok I really did fatfinger it. So there is good news and bad news:
Good news: Pull down DWNLD and flash mode gets enabled. 

Now the less good news:
Flashing kind of goes wrong. First try was 115200 baud. It failed to upload with this warning:
```
Auto-detected: COM5
Uploading .pioenvs\sonoff-DE\firmware.bin
warning: espcomm_send_command: wrong direction/command: 0x01 0x00, expected 0x01 0x08
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed
*** [upload] Error -1
```
But it should have written at least something, because the device is not working anymore. So thats on the upside.
Next I checked for the bootlog via serial Monitor:
```
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
vffffffff
~ld
```
Afaik this could be really caused by anything from wrong flashing to code compatibility issues that are caused by a different wiring on this chip (wich is kind of likely because it has a different vendor).
Also possible that my soldering wasn't rock solid but I don't think so as the bootlog is printing readable output and therefor the serial connection should be working quite well.

By The Way: DWNLD needs to stay pulled down for the device to stay in flash mode. Otherwise it will reset and go back to normal operation after few seconds.

Jason2866, 2018-07-07:
It seems you have a probem with your "uploading environment".
This is really hard to squeeze out when not being there...
I dont think the device is dead or unflashable. Probably it does not work with Tasmota but this is a other thing.... In general it should be reachable via wifi after successfull flash with defined wifi settings in firmware file.

Phil1988, 2018-07-09:
Interessting discussion.

The RX&TX goes to a 8bit, 20pin Microcontroller - a Nuvoton N76E003AT20.
Both over a 22Ω resistor.
And they also go directly to U0TXD and U0RXT - so far so good...

But the DOWNLD pin goes to GPIO0 of the ESP8266 over a 200 Ohm resistor.
Its connected to another resisort pulling high over a 4,7 kΩ resistor.

As far as I know - the flashing mode is set by:
GND, MTDO/GPIO15, GPIO0 set to GND
GPIO2, CHPD, VCC set to 3.3V

GND is GND... 
MTDO is always GND over a 4,7 kΩ resistor.
GPIO must be set to GND via the DOWNLD Pin.

VCC is also set to VCC (3.3V)
GPIO2 is also set to VCC via a 4,7kΩ resistor
But I cant find the CHPD pin on the datasheet.

So I should be relatively easy to flash this thing, but I also cant flash it.
Tried the Arduino IDE (pretty familiar with it) and Atom with platformIO (not so familiar with it).
Tried several TLL Adapter: CP2102, CH340G and FTDI FT232RL... none worked.

I tried by pushing the button before, holding, waited a few seconds after pluging in.. 
Sometimes the blue LED shines (with no DOWNLD to GND), after pushing the button by pluging in it goes off... and if you keep pushing it long it blinks..

I also made a drawing so you guys can jump in more easily:
[image]

Jason2866, 2018-07-09:
CHPD is Chip Enable and has always to be high for a working ESP (normal or flash)
I think DOWNLD always connecting to GND during flash is a good idea because the pullup could do
unwanted things if not... Maybe the parallel connected RX TX to the Nuvoton Mikrocontroller disturbs the signals so flashing fails. If you are not a first time solder you could temporary remove the resistors from this lines.

Phil1988, 2018-07-09:
Good idea... desoldering and report back when I have tried this :)

TLaren, 2018-07-09:
What size is the flash chip?

Phil1988, 2018-07-10:
what do you mean with flash chip?
The internal of the ESP8266?
The set value in the Arduino IDE or the FTDI/TLL thing?

Jason2866, 2018-07-10:
Tlaren means the Chip beside the ESP (Flash Memory). Was is printed on it?

Phil1988, 2018-07-10:
My chip next to the ESP8266 is not readable... but probably the same as @n3m3s1s chip.

So you can flash this board with no issues?
I just have one 3.3V Arduino Nano left so I might give it a try

n3m3s1s, 2018-07-10:
> So you can flash this board with no issues?
I just have one 3.3V Arduino Nano left so I might give it a try

Well. I got it into flash mode by pulling DWLNLD down. I then tried to flash and something must have happened as the device does not work anymore ;)

Jason2866, 2018-07-10:
Hi,
before you mabe kill your devices with bad flashing, lets look if there is a chance to get Tasmota running with basic functionality-> relay switching. For this take a look where the "hot" side of the relay goes.
It probably goes to a transistor and from this transistor it goes too???.  If this is the ESP chip there is a good chance to get it running with Tasmota if it goes to somewhere else there is nearly no chance...
In this case it is better to use it as it is.

n3m3s1s, 2018-07-11:
I hijacked @Phil1988 s photo and drawed a little. Fun fact: appearently you have a slightly different HW version as I do (there is something in the middle of your chip that looks like an optocoupler to me, wich I don't have).
I tried to measure both loose ends of the relay and tried to follow them through the chip. Please bear with me if there are mistakes or ask more questions to follow, as I am doing this for the first time.
 
[image]

Phil1988, 2018-07-11:
@Jason2866: Thats the weird thing.... I cant see where the transistor control lines come from.
Relly NOT... I measured everything at least 5times.. the base terminal of these transistors cant be seen to conect anywhere.
Not at the esp8266 neither at the nuvoTon...
At least not that I can see or measure.

@n3m3s1s : no problem for hijacking :D 
.....
But you defenetly have some wrong connections there.
Be careful on charging the transistors befor you measure ;)
It looks like you have activated the transistors and then measured connections, where no are.

I guess that the parts on the left side gives a 5V DC voltage (not measured/tested... just looked at the board and measured the points to confirm my idea).
They are connected to the right side of the relay coil.

Now the "blue" transistor can be "somehow" turned on, so that the current flows threw the coil, activates the relay for a shorting the AC connections (plug is "ON"),

The red transistor can also be turned on "somehow" which activates the HLW8012. Must be a smart engeneer to only set the measuring IC on, when needed - to save some energy.

I also tested all the connections of the ESP8266 to see the connections.
I hope someone can bring us further with this work :)
[image]

gnomosg, 2018-07-12:
Hi everyone,
I have the @n3m3s1s HW version. The base of "blue" npn transistor is connected via 1k ohm resistor to pin 19 of nuvoTon MCU (this is 8051 MCU). Pin 19 is "P0.3/PWM5/IC5/AIN6". 
So relay is not driven directly from esp8266. Probably this is solution like in Sonoff dual: relay is connected to separate MCU, and communication between esp8266 and this MCU is made via TX/RX pins (serial connection).
I checked this:
esp8266/connector/ TX-> pin 3 (RX) of nuvoTon , 
esp8266/connector/ RX-> pin 2 (TX) of nuvoTon

At this moment I also receive: 
_A fatal error occurred: Failed to write to target RAM (result was 0107)_ 
message, and I have no idea how to go further... :(

Jason2866, 2018-07-13:
@gnomosg 
This are bad news. It needs someone who monitors the traffic between the ESP and the MCU.
This traffic needs to be analyzed :-( 
Programming (flashing) the ESP is the smallest problem at all....

Phil1988, 2018-07-16:
Thanks @gnomosg 

my "blue" transistor is also connected like that.

I spend another hour on trying to flash this thing but I think my first attempts on flashing it has somehow destroyed the flash chip or something because I can not flash it.

This time I tried it directly by pulling GPIO0 to GND and went to the pins directly.
I also desoldered the resistors to the nuvoTon but I had no luck.

So it looks like my device is dead.
I hope someone has more skills than I and will push this further because I am done :D

closed 2018-09-14 by stale[bot]

prunck1964, 2021-05-01:
Hello together
Since I accidentally bought 5 of these sockets and even after a long search I did not find a solution to flash Tasmota. I examined the device more closely and found out that the relay  the button and the LEDs are not connected directly to the ESP8266 but to the N76E003AT20. This means that the second processor is monitoring the flash process.
Since I don't know how to program this processor, I decided to replace it completely and solder the necessary wires directly to the ESP8266. 

1.) I cut off pin 9 and 10 of the N76E003AT20 so that the traces remain intact.
2.) Solder pin 15 (GPIO 0) of the ESP8266 to ground.
3.) Flash Tasmota with a USB serial adapter via the known interface.
4.) Remove the wire from pin 15 (GPIO 0) of the ESP8266.
5.) Completely desolder the CPU N76E003AT20 without destroying the tracks.
6.) Solder a bridge from pin 1 N76E003AT20 (red LED) to pin 12 (GPIO 13 ) ESP8266.
7.) Solder a bridge from pin 13 N76E003AT20 (push button) to pin 15 (GPIO 0) ESP8266.
8.) Solder a bridge from pin 10 N76E003AT (blue led) to pin 10 (GPIO 12) ESP8266.
9.) Solder a bridge from pin 19 N76E003AT (relay) to pin 10 (GPIO12) ESP8266.
This way the socket behaves like a Sonoff S20.
I know it is a difficult and radical way but it works. Of course, you should have some experience in soldering and the right tools (magnifying glass soldering station with a very fine tip and a steady hand).
I hope I could help.

prunck1964, 2021-05-01:
[image]